dparson55/NRFLite #4: Compile warnings

killerkalamari, 2017-12-05:
I get a bunch of code warnings trying to compile the Basic_RX example project. These are distracting. Is the library production ready, or am I trying to use it too soon? Thanks!

NRFLite library 2.0.0
Arduino 1.8.4

```.../NRFLite/src/NRFLite.cpp: In member function 'void NRFLite::printDetails()':
.../NRFLite/src/NRFLite.cpp:273:49: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("CONFIG", readRegister(CONFIG));
                                                 ^
.../NRFLite/src/NRFLite.cpp:274:47: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("EN_AA", readRegister(EN_AA));
                                               ^
.../NRFLite/src/NRFLite.cpp:275:55: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("EN_RXADDR", readRegister(EN_RXADDR));
                                                       ^
.../NRFLite/src/NRFLite.cpp:276:53: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("SETUP_AW", readRegister(SETUP_AW));
                                                     ^
.../NRFLite/src/NRFLite.cpp:277:57: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("SETUP_RETR", readRegister(SETUP_RETR));
                                                         ^
.../NRFLite/src/NRFLite.cpp:278:47: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("RF_CH", readRegister(RF_CH));
                                               ^
.../NRFLite/src/NRFLite.cpp:279:53: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("RF_SETUP", readRegister(RF_SETUP));
                                                     ^
.../NRFLite/src/NRFLite.cpp:280:49:** warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("STATUS", readRegister(STATUS));
                                                 ^
.../NRFLite/src/NRFLite.cpp:281:57: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("OBSERVE_TX", readRegister(OBSERVE_TX));
                                                         ^
.../NRFLite/src/NRFLite.cpp:282:53: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("RX_PW_P0", readRegister(RX_PW_P0));
                                                     ^
.../NRFLite/src/NRFLite.cpp:283:53: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("RX_PW_P1", readRegister(RX_PW_P1));
                                                     ^
.../NRFLite/src/NRFLite.cpp:284:59: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("FIFO_STATUS", readRegister(FIFO_STATUS));
                                                           ^
.../NRFLite/src/NRFLite.cpp:285:47: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("DYNPD", readRegister(DYNPD));
                                               ^
.../NRFLite/src/NRFLite.cpp:286:51: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
     printRegister("FEATURE", readRegister(FEATURE));```

dparson55, 2017-12-05:
What version of the Arduino IDE are you using?  I've been developing on 1.8.5 (the latest version).

The code is ready but I have an example that I want to improve.  I'll be making it version 2.0.1 and will just be an improvement to one of the examples.

dparson55, 2017-12-06:
@killerkalamari I enabled verbose compilation and tried 1.6.9, 1.8.4, and 1.8.5, and none of them reported these warnings so I wasn't able to recreate the issue.  I made an update though to see if it will address this problem.  Can you see if NRFLite 2.0.1 reports any more warnings?

killerkalamari, 2017-12-07:
@dparson55 I haven't had time to try your fix at home, but at work, I get no errors in either 2.0.0 or 2.0.1, under Arduino 1.8.4... So, at this point I suspect the problem is not in your library at all. I'll check out 2.0.1 at home as soon as I can. If it doesn't help, I'll probably remove Arduino 1.8.4 and put 1.8.5 on, maybe there's just something corrupted in my install. Thank you for your help!

dparson55, 2017-12-22:
Sounds good, thanks.

commit e1e9bd80 (2017-12-06): Fixes #4
closed 2017-12-22 by dparson55