dparson55/NRFLite #5: correct definition for attinycore

marcolombardi-rm, 2017-12-22:
#if defined( __AVR_ATtiny84__ )
    #define USI_DI 6	// PB6
    #define USI_DO 5	// PB5
    #define SCK 4		// PB4
#elif defined( __AVR_ATtiny85__ )
    #define USI_DI 0	// PB0
    #define USI_DO 1	// PB1
    #define SCK 2		// PB2
#else
    #include <SPI.h> // Use the normal Arduino hardware SPI library if we are not on ATtiny.
#endif

dparson55, 2017-12-23:
Sorry @marcolombardi-rm, this change does not compile with the libraries I'm officially supporting.  I tested with the Arduino IDE 1.8.5 and the **attiny** library by David A. Mellis.  https://github.com/damellis/attiny  What attiny library are you using?

marcolombardi-rm, 2017-12-23:
https://github.com/SpenceKonde/ATTinyCore

commit ccab3502 (2018-01-02): Updated to support ATTinyCore

Closes #5 but I still don't like this library, it's a mess compared to
https://github.com/damellis/attiny
closed 2018-01-02 by dparson55