dparson55/NRFLite #6: Biderectional communication

fugiro, 2017-12-30:
How could be implemented a bidirectional communication protocol? 

When two programs continously send data from one radio to other, both freezes.

dparson55, 2018-01-01:
I believe acknowledgement data packets is how the designers of the NRF24L01 intended bidirectional communication, but if you can provide some code to demonstrate the lock up issue you mentioned, I will try to help.

fugiro, 2018-01-02:
Thanks, but I already solved the problem using a simple protocol.

**Code for Radio 0:**
```
#include <SPI.h>
#include <NRFLite.h>

NRFLite radio;
uint16_t data_tx[10]; // TX Data
uint16_t data_rx[10]; // RX Data
unsigned long timer_loop, radio_timeout;

void setup() {
  radio.init(0, 9, 10); // Set radio Id = 0, along with the CE and CSN pins
  
  Serial.begin(115200);
  
  radio_timeout = millis();
}

void loop() {
  //Timer loop 100Hz
  while (micros() - timer_loop <= 10000);
  timer_loop = micros();

  radio_protocol();

  //Process
  data_tx[1] = random(0,1023); //Sensor 1
  data_tx[2] = 2200; //Sensor 2
}

void radio_protocol() {  
  
  if(data_tx[0] == 1){ //Protocol
    radio.send(1, &data_tx, sizeof(data_tx));  
    data_tx[0] = 0;
    radio_timeout = millis();
  }
  
  if (radio.hasData()) { //Read data
    radio.readData(&data_rx);
    data_tx[0] = data_rx[0];
    print_data_rx();
  }

  if(millis() - radio_timeout >= 500){
    Serial.println("Timeout");
    data_tx[0] = 1;
  }
  
}

void print_data_rx() {
  char str[20];
  sprintf(str, "%04u %04u", data_rx[1], data_rx[2]);
  Serial.println(str);
}
```

**Code for Radio 1:**

```
#include <SPI.h>
#include <NRFLite.h>

NRFLite radio;
uint16_t data_tx[10]; // TX Data
uint16_t data_rx[10]; // RX Data
unsigned long radio_timeout;

void setup() {
  radio.init(1, 9, 10); // Set radio Id = 1, along with the CE and CSN pins
  
  Serial.begin(115200);

  radio_timeout = millis();
}

void loop() {

  radio_protocol();

  //Process
  data_tx[1] = random(0,1023); //Sensor 1 
  data_tx[2] = 1100; //Sensor 2
}

void radio_protocol() {
  
  if(data_tx[0] == 1){ //Protocol
    radio.send(0, &data_tx, sizeof(data_tx));  
    data_tx[0] = 0;
    radio_timeout = millis();
  }
  
  if (radio.hasData()) { //Read data
    radio.readData(&data_rx);
    data_tx[0] = data_rx[0];
    print_data_rx();
  }

  if(millis() - radio_timeout >= 500){
    Serial.println("Timeout");
    data_tx[0] = 1;
  }
}

void print_data_rx() {
  char str[20];
  sprintf(str, "%04u %04u", data_rx[1], data_rx[2]);
  Serial.println(str);
}
```

dparson55, 2018-01-02:
Ok great, thanks for the followup!

closed 2018-01-02 by dparson55