dparson55/NRFLite #7: AutoAck defaults

sjernigan, 2018-01-03:
I new to this so I may be confused.  I have a receiver and sender setup.  I'm trying various combinations of NRF24L01 modules (PCB/Short/Long ant, w and w/o shielding, w & w/o PA/LNA). 
 When I all use the same type of NRF modules, it all works will.  If I use a mixture of the cheap PCB modules with the PA/LNA external antenna modules, it seems to send the message but the ack is not received.  To be more precise, the result of the radio.send method is false when using a mix of modules.  It doesn't seem to matter if the PCB module is used on the sender or receiver.  As such, it doesn't feel like a power issue.  Same result over 1M or 10M.  If I ask for the config details, all of the registers are the same except for the following three: 
OBSERVE_TX 00000000 (PCB) -> 00001001 (Small external ant w/ PA/LNA)
TX_ADDR 231231231231231 -> 12341
RX_ADDR_P0 231231231231231 -> 12341

The observe_tx looks suspicious but I don't see a way to change these nor do I know if the observe_tx is even important.

dparson55, 2018-01-04:
The difference in the observe_tx register is ok, that register simply maintains a count of failed transmissions and retransmit attempts.  The difference in addresses is concerning though and I will do some testing tonight and let you know what I find.

In the meantime, can you try testing the **IRQ_ACK_RX** and **IRQ_ACK_TX** examples with your modules?  These use interrupts for transmission and receiving and since the mechanism they use is different, it will help rule out an issue with the non-interrupt logic.

sjernigan, 2018-01-04:
I can give that a shot tomorrow afternoon.  Thanks

dparson55, 2018-01-04:
Made a mistake thinking about the address length and it is not a problem:  addresses are 5 bytes in length and even though 231231231231231 looks much longer than 12341, the address values are actually 231,231,231,231,231 and 1,2,3,4,1; so both are 5 bytes in length.  I updated the library to show the addresses with commas to make this more clear.  NRFLite version 2.0.6 includes this improvement.

But anyway, you didn't mention RX_ADDR_P1 is different between the receiving and transmitting radios, and they should be.  Let's confirm that if you set the receiver to radio id 0 and the transmitter to radio id 1, you see the following addresses:

**Receiver radio id 0 addresses**
- TX_ADDR 231,231,231,231,231
- RX_ADDR_P0 231,231,231,231,231
- RX_ADDR_P1 1,2,3,4,0

**Transmitter radio id 1 addresses** after removing power from the radio's VCC pin, plugging it back in, and then running radio.init(1, ...) in the setup() function.
- TX_ADDR 231,231,231,231,231
- RX_ADDR_P0 231,231,231,231,231
- RX_ADDR_P1 1,2,3,4,1

**Transmitter radio id 1 addresses** after transmitting data to radio id 0.  Even if you reprogram your microcontroller or reset it, these addresses will remain the same.  It is only if you remove power from the radio that its TX_ADDR and RX_ADDR_PO addresses will reset.
- TX_ADDR 1,2,3,4,0
- RX_ADDR_P0 1,2,3,4,0
- RX_ADDR_P1 1,2,3,4,1

sjernigan, 2018-01-04:
Progress :-)  I have your update and am seeing the commas.  When I swapped the TX to id 1, the ADDRs read as you suggested:
TX_ADDR 1,2,3,4,0
RX_ADDR_P0 1,2,3,4,0
RX_ADDR_P1 1,2,3,4,1
(NB I believe I saw the 231 version before the first transmit)

I don't know if anything else changed but I am now able to use the short antenna versions (with shielding) and get successful ACKs back.   Curiously, I still cannot get a successful ACK back when I use a PCB ant module with a long antenna, no shielding version.  It doesn't matter which is TX and RX.  This is somewhat surprising given that the PCB module and the long antenna module are both from MakerFocus.  The configurations look consistent across all modules.  If I use two of the long ant modules or two PCBs, it works fine so it doesn't seem to be a lack of power issue.  

Anyhow. for my use case, the shorter antenna will actually work better anyhow so I think I'm good.   Thanks for the help!  I'm under a bit of time pressure at the moment so it may be a while before I'll get to the IRQ test.

dparson55, 2018-01-05:
Good to hear.  With all this inconsistency I really hope it is not a problem with the library.  Unfortunately I do not have amplified versions of the modules to test with since for long range work, I decided to use [RFM69](https://github.com/LowPowerLab/RFM69) modules instead.

I would be interested to hear how the IRQ test goes if you ever get to it.  Another thought, you can try a different channel number as well.  By default channel 100 is used but you can override this in the radio.init() method.

closed 2018-01-05 by dparson55

sjernigan, 2018-01-05:
If you are interested in testing further, I would be willing to ship you a version with a long antenna.  Just email your postal address to jernigan_steve at hotmail.com.