dparson55/NRFLite #10: How many nRF24 radio

roboticboyer, 2018-02-03:
Which is the max number of Radio_IDs I can use?
My understanding is that nRF24 has a data pipe so each RX can receive data from 6 different TXs
[see this block diagram](https://4.bp.blogspot.com/-FDWTGC5sKCc/VmLl1ypuy5I/AAAAAAAAAk8/OQqlJmMJi0I/s640/ExampleOfTransmission.jpg)
if confirmed,could you add an example
thank you

dparson55, 2018-02-04:
This library only supports 1 receiving address.  This was intentional to keep the library easy to use.

closed 2018-02-05 by dparson55

adamzxtan, 2019-08-19:
Hi, sorry to reopen back this issue. So based on the examples, I think there will be no issue for me to have like 6 slaves and 1 master doing two way communication? All I have to do is assign different RadioID for the slave devices right? (ex. RadioID 1 to 6)

dparson55, 2019-08-20:
Yes that is correct, you just need to make sure each radio is assigned a different id.  If you watch the 2-pin operation video tutorial, about half way through I show the sensor example sketches which might help get you with your project.  Good luck!

adamzxtan, 2019-08-20:
Thanks @dparson55. Will check out on your second video