dparson55/NRFLite #13: Settings for NRF

LostStudent, 2018-04-16:
Hey thanks for making such a great library. Should it be able to work with a raspberry pi running the RF24 library? I tried matching the setting as much as I could but didn’t get anything come through.

dparson55, 2018-04-17:
Unfortunately no, NRFLite won't work with a RPi.  I skimmed through the RF24 library and they have some specific logic to handle it.  So it would be best to just use RF24 instead.

LostStudent, 2018-04-17:
Hey sorry what I was asking wasn’t clear.

I’m trying to receive packets on a rpi using RF24

And using. An uno with NRFLite to send the packets.

I’ve tried matching the settings as best I can but I don’t sent to be getting any through. 

Do you think there is any reason it wouldn’t work or shall I keep trying to match the settings on each side?

dparson55, 2018-04-17:
Oh I see and in this case, it would be possible, just not very easy.  Basically you'll need to setup the RPI+RF24 to use the settings that NRFLite expects.  The RF24 library has control on what can be adjusted, so you will need to make the adjustments on the RF24 side.

Set PPI+RF24 to:
- Channel 100
- 2MBPS bitrate
- 500 microseconds between retries
- 15 retries
- The receive address for pipe 1 (RX_ADDR_P1) needs to be assigned to the following 5 bytes: { 1, 2, 3, 4, radioId }  You can reference this line for more detail https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L378 Basically you need the address in register TX_ADDR on the NRFLite side to match the address in RX_ADDR_P1 on the RF24 side.
- Enable dynamic packet sizes on pipes 0 and 1.
- Enable dynamically sized payloads, ACK payloads, and TX support with or without an ACK request.

heavyeste, 2020-03-19:
Hello,
I try this configuration:
- Sender : UNO+RF24 
```
CONFIG 00001011    
EN_AA 00111111     
EN_RXADDR 00000011 
SETUP_AW 00000011  
SETUP_RETR 00011111
RF_CH 01001100     
RF_SETUP 00000110  
STATUS 00001110
OBSERVE_TX 00001111
RX_PW_P0 00100000
RX_PW_P1 00000000
FIFO_STATUS 00010001
DYNPD 00000011
FEATURE 00000111
TX_ADDR 1,2,3,4,0
RX_ADDR_P0 1,2,3,4,0
RX_ADDR_P1 1,2,3,4,5
```

- Receiver: RPI+RF24
```
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0x0102030400
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x03
RF_CH		 = 0x4c
RF_SETUP	 = 0x05
CONFIG		 = 0x7e
DYNPD/FEATURE	 = 0x00 0x00
Data Rate	 = 1MBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_HIGH
```

I followed this post but I cannot receive nothing. If I try with RF24 library with same configuration the RPI receive data correctly.

The UNO TX address is correct?
Can  you help me?
thank you very much

dparson55, 2020-03-19:
The two configurations are very hard to compare due to one being binary and the other hex.  You should manually translate them to be the same so a comparision can be made.

Briefly looking I see 2 issues (there may be more).  The RPI+RF24 is set to receive on addresses that do not match the address that UNO+NRFLite will transmit too.  2nd issue:  the RPI+RF24 is set to expect messages transmitting at 1MBPS.

heavyeste, 2020-03-19:
Hello,

> The RPI+RF24 is set to receive on addresses that do not match the address that UNO+NRFLite will transmit to.

UNO+RF24 send to destination ID = 0, so the desination address in HEX format should be `0x0102030400`, it's right?

> The RPI+RF24 is set to expect messages transmitting at 1MBPS.

I have the following code:
UNO+RF24:
```c++
    _radio.init(RADIO_ID, CE_Pin, CSN_Pin, NRFLite::BITRATE1MBPS, 76);
```
RPI+RF24:
```node
    const rf24 = new nrf24.nRF24(22, 8);
    rf24.begin(false);
    // Configure the radio
    rf24.config({
      PALevel: nrf24.RF24_PA_HIGH,
      DataRate: nrf24.RF24_1MBPS,
      AutoAck: true,
      // Channel: 76,
    }, true);
    
    const pipe = rf24.addReadPipe('0x0102030400', true); // listen in pipe "0x0102030400" with AutoACK enabled.
```

dparson55, 2020-03-20:
The two configurations are very hard to compare due to one being binary and the other hex.  I am not sure if the hex address is correct by looking at it.  I'm not sure if the channel of 76 is being setup on the RPI+RF24 since that is commented out.  Again, the thing I would do is print out all the registers on the TX and RX sides, in the same format, so that a comparison can be done.