dparson55/NRFLite #14: Works with ch340g?

jaimeambrosio, 2018-04-29:
Does the library work with the Chinese arduino clone? The clone uses the ch340g driver. It happens that in a normal Arduino it works, but in Chinese it does not.

Thanks!

dparson55, 2018-04-30:
I might not understand the question:  could you explain more about what works with a normal Arduino but not with a clone?

NRFLite works with the ATmega328, ATtiny84, and ATtiny85 microcontrollers.  The communication between the microcontroller and nRF24L01+ transceiver is handled via the SPI or USI buses on these microcontrollers, and the USB to UART integrated circuit (CH340, FTDI, etc) does not relate to this bus.

jaimeambrosio, 2018-04-30:
[image]

I have this Chinese arduino. The method radio.send (radioIdTo, & valuesSend, sizeof (valuesSend)) apparently does not work. In an arduino UNO normal if it works. Has it happened to you?

dparson55, 2018-04-30:
Ok I understand the question and that Arduino Uno clone should definitely work since it uses the ATmega328 microcontroller.  So you will just have to do some debugging.  I can try to help if you explain what troubleshooting steps you have already tried.

One thing to initially do if you have not already:  setup the Basic_RX and Basic_TX examples to make sure they work before using your own code.  Be sure to try using the genuine Arduino Uno as a transmitter and the clone as a receiver.

jaimeambrosio, 2018-05-01:
I already had it working with two normal arduino. Now that I try to receive or transmit between a normal arduino and a Chinese arduino, it is not possible. I think it's because they are different arduinos!

It only shows "INI". It does not show the "PASS".  

[image]

dparson55, 2018-05-01:
Does radio.init succeed?  It will confirm the microcontroller can communicate with the radio.

`void setup()
{
    Serial.begin(115200);
        
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
}`

If the clone is fully functional it will definitely work, so maybe you should verify the microcontroller on the clone is making good connections to the pins used by the radio.
* Verify the 3.3v being used to power the nRF24L01+ is working on the clone.
* Verify the ground is really connected to ground.
* Write a sketch that toggles each pin used by the radio (for example, digital pins 9-13 are used in the Basic_RX and Basic_TX sketches so you would want to verify each of them).  Do each pin one at a time to ensure when you try toggling pin 11, only pin 11 is changed.  Hopefully you get the idea.

jaimeambrosio, 2018-05-02:
- radio.init  returns 1
- I measured the output of 3.3v with a multitester and it is correct.

Solder the condenser. I read that for the module to get its greatest distance the USB source is not enough and requires the capacitor to eliminate the noise.

I will use the previous arduino. The Chinese version did not work. A different question, would the module be able to transmit below the water to the surface?

Very Thanks for your responses.

dparson55, 2018-05-02:
Oh well, it would be interesting to identify the problem but know it can be a little difficult to troubleshoot, especially doing it this way.

I don't know if the radio can transmit through water, don't think I have ever heard of a project doing that.  Good luck, I hope it does!

closed 2018-05-02 by dparson55

jaimeambrosio, 2018-05-02:
After welding the capacitor already works with the Chinese arduino. Thank you for all your replies!!

dparson55, 2018-05-02:
Good news!