dparson55/NRFLite #15: Error in ACK

lucasmaziero, 2018-05-14:
Hello, because even if I can not send the package I get the success message, only after sending four messages that I receive the failure message .. I believe that the failure message should already be on the first unsuccessful sending.. It seems like a HACK error.

lucasmaziero, 2018-05-14:
I would like to receive the unsuccessful submit error in "Basic_TX.ino" already in the first message that hears a send failure, is it possible?

dparson55, 2018-05-14:
I am sure we can figure out a way to do what you need but you will probably need to explain more more detail what you are trying to do.

It is important to understand that what you said is already happening:  the first "send failure" is when packet 4 is sent because **packets 1, 2, and 3 have been successfully acknowledged by the receiving radio**.  It is only the failure of the Arduino on the receiving side to read the data that is currently stored inside the receiving radio.  The Arduino on the receiving side has called **radio.init** so it configured the radio it is plugged into to listen for packets.  However, this Arduino is not doing its job of reading the data when the radio receives packets.  Hopefully that makes sense.

There are many ways to prevent the receiving radio from receiving data packets.  It depends on what you are trying to do so if you can explain more, I might be able to provide better guidance.  But here are a few possibilities:
- Do not call **radio.init** on the receiving side until you are ready to continually monitor the radio for data.
- Call **radio.powerDown** on the receiving side to turn off the radio and put it into a low power state.  When it is powered down it will stop listening for packets and only consume 900 nano-amps.  Calling **radio.hasData** will turn the radio back on.
- Set the CE pin of the radio to LOW.  This will stop the radio from listening to packets but will still consume 26 micro-amps.

lucasmaziero, 2018-05-15:
This was the way I found that everything works fine and is sent 3 times with **for (uint8_t i = 0; i <3; i ++) _radio.send (DESTINATION_RADIO_ID, & _radioData, sizeof (_radioData))** and once again waiting for the I am not aware of the first 3 packets getting the 4 pack, I would like some suggestion of how this can be implemented in a better way .. Thanks

My code TX:
```c++
#include <SPI.h>
#include <NRFLite.h>
#include <SimpleThread.h> //https://github.com/lucasmaziero/SimpleThread

const static uint8_t RADIO_ID = 1;             // Our radio's id.
const static uint8_t DESTINATION_RADIO_ID = 0; // Id of the radio we will transmit to.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

uint32_t time = 0;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
  uint8_t FromRadioId;
  uint32_t OnTimeMillis;
  uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

SimpleThread sendTime(5000);

void setup()
{
  Serial.begin(115200);

  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }

  _radioData.FromRadioId = RADIO_ID;
}

void loop()
{
  if (sendTime.check())
  {
    _radioData.OnTimeMillis = millis();

    Serial.print("Sending ");
    Serial.print(_radioData.OnTimeMillis);
    Serial.print(" ms");

    time = millis();
    for (uint8_t i = 0; i < 3; i++) _radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData));
    if (_radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData))) // Note how '&' must be placed in front of the variable name.
    {
      Serial.println("...Success");
    }
    else
    {
      Serial.println("...Failed");
      _radioData.FailedTxCount++;
    }

    Serial.print("Time: ");
    Serial.print(millis() - time);
    Serial.println(" ms");
  }
}
```
My code  RX:
```c++
#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

uint8_t i = 0;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
  uint8_t FromRadioId;
  uint32_t OnTimeMillis;
  uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
  Serial.begin(115200);

  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }
}

void loop()
{
  while (_radio.hasData())
  {
    _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.

    if (dataIgnore())
    {
      String msg = "Radio ";
      msg += _radioData.FromRadioId;
      msg += ", ";
      msg += _radioData.OnTimeMillis;
      msg += " ms, ";
      msg += _radioData.FailedTxCount;
      msg += " Failed TX";

      Serial.println(msg);
    }
  }
}

bool dataIgnore()
{
  if (i++ >= 3)
  {
    i = 0;
    return true;
  }
  return false;
}
```

dparson55, 2018-05-16:
I still do not understand the problem you are trying to solve.  The TX Arduino will fail on packet 1 if it does not receive an ACK from the RX Arduino, so there is no reason to send 4 packets.

Maybe during your testing you are holding down the RESET button on the RX Arduino in order to verify the TX Arduino fails to transmit, but this can be a problem.  The nRF24L01 will operate independent of the Arduino's microcontroller and will still be able to receive packets if it has been initialized with *radio.init*.  Here are two examples to try explaining the situation:

Assuming you have Basic_TX.ino and Basic_RX.ino programmed into your 2 Arduinos.

Scenerio 1 (TX Arduino is powered on, RX Arduino is powered off):

- To not power the RX Arduino and the RX nRF24L01.  Every time the TX Arduino tries to send a packet, it will fail:  packet 1 will fail, packet 2 will fail, etc.

Scenerio 2 (TX Arduino is powered on, RX Arduino is powered on and then RESET is held down):

- Power on the RX Arduino (so it runs *radio.init*) and then hold the RESET button down on the RX Arduino.  The TX Arduino will successfully send 3 packets because the nRF24L01 on the RX side has been initialized and is receiving power.  The RESET button on the Arduino does not affect the nRF24L01 that it is connected to.  Packet 1, 2, and 3 will be successful and packet 4 will fail because the RX nRF24L01 has a buffer that can hold 3 packets, so on the 4th packet it will be full and it will not send an ACK back to the RX Arduino because it cannot accept the packet.

lucasmaziero, 2018-05-16:
Actually I'm pressing the Arduino reset in my tests, now when I take the power out of the RX the TX gets the ACK error as expected, now everything works 100%. Thank you for your help.

dparson55, 2018-05-16:
Good to hear!

closed 2018-05-16 by dparson55