dparson55/NRFLite #17: Payload bug when sending from atmega328p to NodeMcu(Esp8266-12)

devilsray, 2018-08-12:
Hello,

when sending from an Esp8266 to an Esp8266 with the example sketches (BasicRX/TX), everything works fine. But when sending from plain Atmega328p to Esp8266, the FromRadioId is set, but not the OnTimeMillis.  After resetting the Atmega, than some times there came a value, but this is not changing the whole time.

I think i had the same problem yesterday with an Arduino nano. But not sure.

Is tested multiple Atmegas, NodeMcu's and Nrf24 modules. Seems not to be a hardware problem

dparson55, 2018-08-13:
I wish I had an ESP8266 so I could test the fix for you, but I believe the issue is related to an optimization the ESP compiler is doing.  Can you try adding the **packed** attribute to the code on the ESP side to see if it solves the problem?  So on the ESP side, change the definition of the RadioPacket to be:

```
struct __attribute__((packed)) RadioPacket
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};
```

Telling the compiler to pack the RadioPacket on the ESP will hopefully make the structure the same size and contents as the (unpacked) structure on the Arduino.

devilsray, 2018-08-13:
I tested it right now, it works great. Thank you, great work!

commit 13e6016e (2018-10-13): Added ESP8266 example, fixes #17
closed 2018-10-13 by dparson55