dparson55/NRFLite #18: Clash between devices using SPI bus

AidanTek, 2019-01-25:
Hi Dave - I have been trying to get the NRF24L01 to work in a project with an RF522 RFID reader. Both are sharing the SPI bus on a Teensy 3.5. Without changing anything on the breadboard, I can get both chips to behave independently, but there is a problem when I try to use them together. The MCU freezes when I try to send a packet over the NRF24L01 after reading an RFID tag.

I think that the NRF24L01 is getting a bit scrambled by RF522 activity - I have been trying to enable and disable the two devices at the appropriate points in the sketch. Have I missed something?

```cpp
/* Typical RFID pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS      SDA(SS)      10            53        D10        10               10
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
*/

/*
Tx Rx Connections:

Radio    Arduino
CE    -> 8
CSN   -> 7 (Hardware SPI SS)
MOSI  -> 11 (Hardware SPI MOSI)  
MISO  -> 12 (Hardware SPI MISO)
SCK   -> 13 (Hardware SPI SCK)
IRQ   -> No connection
VCC   -> No more than 3.6 volts
GND   -> GND
*/

#include <SPI.h>
#include <MFRC522.h>
#include <NRFLite.h>

//reader 1
int SS_PIN = 15;
int RST_PIN = 9;

// Radio Variables
byte chan = 1;
byte radio_id = 1; // this radio's ID
byte send_id = 0; // the ID of the receiver radio
byte cepin = 8;
byte csnpin = 7;
byte radio_chan = 0; // please change this number to your group number

NRFLite radio; // this allows us to talk to the library in the sketch

///

MFRC522 rfid(SS_PIN, RST_PIN); // Instance of the class

MFRC522::MIFARE_Key key; 

// Init array that will store new NUID 
byte nuidPICC[4];

void setup() { 
  Serial.begin(115200);
  SPI.begin(); // Init SPI bus
  rfid.PCD_Init(); // Init MFRC522 

  for (byte i = 0; i < 6; i++) {
    key.keyByte[i] = 0xFF;
  }

  Serial.println(F("This code scan the MIFARE Classsic NUID."));
  Serial.print(F("Using the following key:"));
  printHex(key.keyByte, MFRC522::MF_KEY_SIZE);

  // Test the radio is connected properly:
  if (!radio.init(radio_id, cepin, csnpin, NRFLite::BITRATE2MBPS, radio_chan)) {
    Serial.println("Cannot communicate with radio");
    while(true); // Wait here forever.
  }

}
 
void loop() {
  radio.powerDown();
  rfid.PCD_SoftPowerUp();
  
  int user;
  
  // Look for new cards
  if ( ! rfid.PICC_IsNewCardPresent())
    return;

  // Verify if the NUID has been read
  if ( ! rfid.PICC_ReadCardSerial())
    return;

  Serial.print(F("PICC type: "));
  MFRC522::PICC_Type piccType = rfid.PICC_GetType(rfid.uid.sak);
  Serial.println(rfid.PICC_GetTypeName(piccType));

  // Check is the PICC of Classic MIFARE type
  if (piccType != MFRC522::PICC_TYPE_MIFARE_MINI &&  
    piccType != MFRC522::PICC_TYPE_MIFARE_1K &&
    piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
    Serial.println(F("Your tag is not of type MIFARE Classic."));
    return;
  }

  if (rfid.uid.uidByte[0] != nuidPICC[0] || 
    rfid.uid.uidByte[1] != nuidPICC[1] || 
    rfid.uid.uidByte[2] != nuidPICC[2] || 
    rfid.uid.uidByte[3] != nuidPICC[3] ) {
    Serial.println(F("A new card has been detected."));

    // Store NUID into nuidPICC array
    // ** This is where the new UUID is stored in the background array - it will prevent the same card being read twice.
    for (byte i = 0; i < 4; i++) {
      nuidPICC[i] = rfid.uid.uidByte[i];
    }

    // ** This is the user identification that Aidan added
    if (nuidPICC[0] == 0x19 && 
    nuidPICC[1] == 0x5A &&
    nuidPICC[2] == 0xDA && 
    nuidPICC[3] == 0x2A ) {
      user = 1;
      Serial.println("User 1 Start Detected");
    }

    if (nuidPICC[0] == 0x39 && 
    nuidPICC[1] == 0xAF &&
    nuidPICC[2] == 0xD2 && 
    nuidPICC[3] == 0x2A ) {
      user = 2;
      Serial.println("User 2 Start Detected");
    }
   
    Serial.println(F("The NUID tag is:"));
    Serial.print(F("In hex: "));
    printHex(rfid.uid.uidByte, rfid.uid.size);
//    Serial.println();
//    Serial.print(F("In dec: "));
//    printDec(rfid.uid.uidByte, rfid.uid.size);
    Serial.println();

    for (byte i = 0; i < 4; i++) {
      nuidPICC[i] = 0xFF;
    }
  }
  else Serial.println(F("Card read previously."));

  // Halt PICC
  rfid.PICC_HaltA();

  // Stop encryption on PCD
  rfid.PCD_StopCrypto1();

  rfid.PCD_SoftPowerDown();

  delay(10);

  Serial.println("Attempting to send message");
  
  if(radio.send(send_id, &user, sizeof(user))) { // & must be used in front of variable
    Serial.println("...Success");
  }
  else {
    Serial.println("...Failed");
  }
  delay(10);
}

/**
 * Helper routine to dump a byte array as hex values to Serial. 
 */
void printHex(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}
```

dparson55, 2019-01-25:
Hi Aidan,

Your code looks good so I think it must be a problem with the libraries not integrating well.  My NRFLite library is only for AVR and ESP8266 architectures, not the TEENSY.  But maybe we can figure out the problem.  Could you give me a link to the specific RF522 library and version you are using so I can study its code?

I started digging into a library I found (https://github.com/miguelbalboa/rfid<https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp>) and it explicitly assigns SPI settings.  NRFLite does not do this and perhaps the explicitly assigned SPI settings are incompatible with the NRF24L01 chip.  This is all just a guess since I'll need to look at the code your RF522 library is using.

Here's what led me to making this guess:  line 49 from https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp
SPI.beginTransaction(SPISettings(MFRC522_SPICLOCK, MSBFIRST, SPI_MODE0));

I believe the MSBFIRST and SPI_MODE0 are what the NRF24L01 chip expects, but maybe the clock speed setting is too fast.  The maximum speed for the chip is 8MHz.  This could be a quick code change for testing:

  Serial.println("Attempting to send message");
  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));

  if(radio.send(send_id, &user, sizeof(user))) { // & must be used in front of variable
    Serial.println("...Success");
  }
  else {
    Serial.println("...Failed");
  }

   SPI.endTransaction(); // Stop using the SPI bus
  delay(10);

If the SPI settings turn out to be the issue, I'll update my library to explicitly set the necessary SPI settings.

dparson55, 2019-01-25:
Hi Aidan,

Your code looks good so I think it must be a problem with the libraries not integrating well.  NRFLite is only for AVR and ESP8266 architectures, not the TEENSY, but maybe we can figure out the problem.  Could you give me a link to the specific RF522 library and version you are using so I can study its code?

I started digging into a library I found (https://github.com/miguelbalboa/rfid<https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp>) and it explicitly assigns SPI settings.  NRFLite does not do this and perhaps the explicitly assigned SPI settings are incompatible with the NRF24L01 chip.  This is all just a guess since I'll need to look at the RF522 library code being used.

Here's what led me to making this guess:  line 49 from https://github.com/miguelbalboa/rfid/blob/master/src/MFRC522.cpp
SPI.beginTransaction(SPISettings(MFRC522_SPICLOCK, MSBFIRST, SPI_MODE0));

I believe the MSBFIRST and SPI_MODE0 are what the NRF24L01 chip expects, but maybe the clock speed setting is too fast.  The maximum speed for the chip is 8MHz.  This could be a quick code change for testing:

```
Serial.println("Attempting to send message");
SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0));

if (radio.send(send_id, &user, sizeof(user))) { // & must be used in front of variable
  Serial.println("...Success");
}
else {
  Serial.println("...Failed");
}

SPI.endTransaction(); // Stop using the SPI bus
delay(10);
```

If the SPI settings turn out to be the issue, I'll update my library to explicitly set the necessary SPI settings but I probably won't be able to officially add TEENSY to the supported architectures since I don't have one for testing.

AidanTek, 2019-02-11:
Hi Dave - thanks a lot for the reply, I had totally missed it. Yes I am using that specific RF522 library you spotted - I have been using Teensy successfully with your library already without a hitch - I'm using 3.2, 3.5 and 3.6 versions of the Teensy - this particular problem came up with a project that a student has been working on, I *think* they have tried using an Arduino Uno as well. I think you are correct in that the issue is more an incompatibility between the libraries. Thanks for the snippet - I will try it towards the end of the week

dparson55, 2019-02-11:
Ok good deal, I'll await the results of the test.

dparson55, 2019-02-19:
Still no word from you so please let me know by Feb 20 if you need more time.  Since I'm unable to reproduce the issue and already have omitted Teensy as a supported architecture, closing the issue saying Teensy is not supported is all I can do at this point.

dparson55, 2019-02-22:
I made the change to set the SPI transmission speed in the library.  If this doesn't solve the issue AND you have time to help troubleshoot, feel free to open another issue.

closed 2019-02-22 by dparson55

AidanTek, 2019-02-28:
Hi Dave - sorry it took a while to get here but we tested your SPI tweak today and this definitely resolved the problem, we are using a Teensy 3.5. I am confident that your library is compatible with the Teensy 3.x family and the same problem can be reproduced on an Uno. This is the code that is currently running on the Teensy 3.5:

```cpp
/* Typical RFID pin layout used:
 * -----------------------------------------------------------------------------------------
 *             MFRC522      Arduino       Arduino   Arduino    Arduino          Arduino
 *             Reader/PCD   Uno/101       Mega      Nano v3    Leonardo/Micro   Pro Micro
 * Signal      Pin          Pin           Pin       Pin        Pin              Pin
 * -----------------------------------------------------------------------------------------
 * RST/Reset   RST          9             5         D9         RESET/ICSP-5     RST
 * SPI SS      SDA(SS)      10            53        D10        10               10
 * SPI MOSI    MOSI         11 / ICSP-4   51        D11        ICSP-4           16
 * SPI MISO    MISO         12 / ICSP-1   50        D12        ICSP-1           14
 * SPI SCK     SCK          13 / ICSP-3   52        D13        ICSP-3           15
*/

/*
Tx Rx Connections:

Radio    Arduino
CE    -> 8
CSN   -> 7 (Hardware SPI SS)
MOSI  -> 11 (Hardware SPI MOSI)  
MISO  -> 12 (Hardware SPI MISO)
SCK   -> 13 (Hardware SPI SCK)
IRQ   -> No connection
VCC   -> No more than 3.6 volts
GND   -> GND
*/

#include <SPI.h>
#include <MFRC522.h>
#include <NRFLite.h>

//reader 1
int SS_PIN = 15;
int RST_PIN = 9;

// Radio Variables
byte chan = 1;
byte radio_id = 1; // this radio's ID
byte send_id = 0; // the ID of the receiver radio
byte cepin = 8;
byte csnpin = 7;
byte radio_chan = 0; // please change this number to your group number

NRFLite radio; // this allows us to talk to the library in the sketch

///

MFRC522 rfid(SS_PIN, RST_PIN); // Instance of the class

MFRC522::MIFARE_Key key; 

// Init array that will store new NUID 
byte nuidPICC[4];

void setup() { 
  Serial.begin(115200);
  SPI.begin(); // Init SPI bus
  rfid.PCD_Init(); // Init MFRC522 

  for (byte i = 0; i < 6; i++) {
    key.keyByte[i] = 0xFF;
  }

  Serial.println(F("This code scan the MIFARE Classsic NUID."));
  Serial.print(F("Using the following key:"));
  printHex(key.keyByte, MFRC522::MF_KEY_SIZE);

  // Test the radio is connected properly:
  if (!radio.init(radio_id, cepin, csnpin, NRFLite::BITRATE2MBPS, radio_chan)) {
    Serial.println("Cannot communicate with radio");
    while(true); // Wait here forever.
  }

}
 
void loop() {
  radio.powerDown();

  delay(10);
  
  rfid.PCD_SoftPowerUp();
  
  int user;
  
  // Look for new cards
  if ( ! rfid.PICC_IsNewCardPresent())
    return;

  // Verify if the NUID has been read
  if ( ! rfid.PICC_ReadCardSerial())
    return;

  Serial.print(F("PICC type: "));
  MFRC522::PICC_Type piccType = rfid.PICC_GetType(rfid.uid.sak);
  Serial.println(rfid.PICC_GetTypeName(piccType));

  // Check is the PICC of Classic MIFARE type
  if (piccType != MFRC522::PICC_TYPE_MIFARE_MINI &&  
    piccType != MFRC522::PICC_TYPE_MIFARE_1K &&
    piccType != MFRC522::PICC_TYPE_MIFARE_4K) {
    Serial.println(F("Your tag is not of type MIFARE Classic."));
    return;
  }

  if (rfid.uid.uidByte[0] != nuidPICC[0] || 
    rfid.uid.uidByte[1] != nuidPICC[1] || 
    rfid.uid.uidByte[2] != nuidPICC[2] || 
    rfid.uid.uidByte[3] != nuidPICC[3] ) {
    Serial.println(F("A new card has been detected."));

    // Store NUID into nuidPICC array
    // ** This is where the new UUID is stored in the background array - it will prevent the same card being read twice.
    for (byte i = 0; i < 4; i++) {
      nuidPICC[i] = rfid.uid.uidByte[i];
    }

    // ** This is the user identification that Aidan added
    if (nuidPICC[0] == 0x19 && 
    nuidPICC[1] == 0x5A &&
    nuidPICC[2] == 0xDA && 
    nuidPICC[3] == 0x2A ) {
      user = 1;
      Serial.println("User 1 Start Detected");
    }

    if (nuidPICC[0] == 0x39 && 
    nuidPICC[1] == 0xAF &&
    nuidPICC[2] == 0xD2 && 
    nuidPICC[3] == 0x2A ) {
      user = 2;
      Serial.println("User 2 Start Detected");
    }
   
    Serial.println(F("The NUID tag is:"));
    Serial.print(F("In hex: "));
    printHex(rfid.uid.uidByte, rfid.uid.size);
//    Serial.println();
//    Serial.print(F("In dec: "));
//    printDec(rfid.uid.uidByte, rfid.uid.size);
    Serial.println();

    for (byte i = 0; i < 4; i++) {
      nuidPICC[i] = 0xFF;
    }
  }
  else Serial.println(F("Card read previously."));

  // Halt PICC
  rfid.PICC_HaltA();

  // Stop encryption on PCD
  rfid.PCD_StopCrypto1();

  delay(10);

  Serial.println("Attempting to send message"); 
  SPI.beginTransaction(SPISettings(4000000, MSBFIRST, SPI_MODE0)); 
  if (radio.send(send_id, &user, sizeof(user))) { // & must be used in front of variable 
    Serial.println("...Success"); 
  } 
  else { 
    Serial.println("...Failed"); 
  } 
  SPI.endTransaction(); // Stop using the SPI bus 
  delay(10);

}

/**
 * Helper routine to dump a byte array as hex values to Serial. 
 */
void printHex(byte *buffer, byte bufferSize) {
  for (byte i = 0; i < bufferSize; i++) {
    Serial.print(buffer[i] < 0x10 ? " 0" : " ");
    Serial.print(buffer[i], HEX);
  }
}
```

dparson55, 2019-02-28:
Good to hear!  If you update NRFLite to 2.0.10 it has this fix, so you could remove the SPI.beginTransaction and SPI.endTransaction statements.

AidanTek, 2019-02-28:
Super! Thank you :)

commit 5e4128c8 (2019-02-22): Use SPI transactions

Hopefully fixes #18