dparson55/NRFLite #19: Changing SPI speed

offgridsystems, 2019-03-14:
Hello,
I'm finding it impossible to change the SPI speed, which NRFlite wants as a default at 8Mhz .
The nRF24L01+ data sheet calls out a maximum SPI SCK speed of 4M-10Mhz depending on loading on the clock pin. With even 10pf loading, the max. clock speed is down to 5Mhz. I would even recommend going down to 4Mhz to default to, and the user can try to increase from there. 
Second, when I edit the NRFLite.h file:
   const static uint32_t NRF_SPICLOCK = 8000000; // Speed to use for SPI communication with the transceiver.
I don't see any clock change. Is there somewhere else the clock speed is also being set? If I am not finding it!
Thanks...
Tim

dparson55, 2019-03-15:
Well shoot, I made a change to the library recently to fix [Issue 18](https://github.com/dparson55/NRFLite/issues/18) and maybe this caused the problem you are now seeing.  Unfortunately this issue was also related to my attempt to help support the Teensy architecture so maybe I fixed it for one person's situation, but broke it for yours.  I don't have one of these for testing, so adding Teensy support looks to be problematic.

For adjusting the SPI clock speed, NRF_SPICLOCK is used in https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L526 and this is supposed to override any default value assigned in the Arduino SPI library and also override changes other libraries make prior to their usage of the SPI bus (which was issue 18). 

Try commenting out the beginTransaction and endTransaction lines so whatever default values the Arduino SPI library is using for the Teensy are preserved.  You can also try lowering the NRF_SPICLOCK to 1000000 in case 4000000 (or any other values you have tried) are unsupported on the Teensy for some reason.

```
// Transfer with the Arduino SPI library.
// SPI.beginTransaction(SPISettings(NRF_SPICLOCK, MSBFIRST, SPI_MODE0)); // Commented out for Teeny test.
SPI.transfer(regName);
for (uint8_t i = 0; i < length; ++i) {
    uint8_t newData = SPI.transfer(intData[i]);
    if (transferType == READ_OPERATION) { intData[i] = newData; }
}
//SPI.endTransaction(); // Commented out for Teensy test.
```

Thanks for the note about the 8MHz clock speed being too high based on the capacitive load placed on the crystal.  I didn't notice the other tables in the datasheet that showed the max SPI clock being as low as 4MHz, e.g. table 27.  I will get this fixed but will wait to hear back from you about the additional SPI troubleshooting.

dparson55, 2019-03-15:
Unrelated question concerning the code:  was there a reason why you didn't use acknowledgement data packets, was it not working or just easier for you to manually switch the radio from being a transmitter and receiver?

offgridsystems, 2019-03-15:
Thanks for taking the time to look at the problem. I can take a look at problem on my end this weekend and get back to you by Monday latest.
On the ack packets, I'm not a fan, they seem to add a lot of time. My app needs hundreds of nodes (battery blocks) and there is no time for acks, nor do they seem to be very reliable. My application uses a master to send a sync packet, and then all the nodes respond in sequence. Since it all happens so fast, there is time to query the nodes that did not update data in the master. This is much more reliable than the ack packets, in my testing.
t

dparson55, 2019-03-15:
I'll dig into the datasheet of whatever microncontroller the Teensy uses this weekend as well.  Maybe manually controlling one of its registers to set its SPI speed is something else that can be tried.

Great info about your ack packet situation as well.  Your bad experience with the automatic ack data packets makes perfect sense when so many nodes are involved.  Internally the nRF24L01 switches between TX and RX operation each time a REQUIRE_ACK send is performed, and even though it does this much quicker than doing it via the library, having to switch hundreds of times would cause all sorts of packet loss.  The radio will retry each send 16 times as well, each time having to switch between TX and RX operation.  So it is no wonder why the ack data packets did not work for your situation, what you did to manually control the TX and RX mode is perfect, great work!

dparson55, 2019-03-18:
Unfortunately the datasheet for the Teensy is a little beyond me, so I'll just need to depend on the Arduino library honoring the setting NRFLite provides.  I hope the 2 things I mentioned will work, let me know by 3/20 if you can.

dparson55, 2019-03-25:
Closing, not able to reproduce and no response from @offgridsystems.

closed 2019-03-25 by dparson55