dparson55/NRFLite #20: NRF24L01 does not work with Attiny84_Basic_TX

AlvaroGordo, 2019-03-27:
Hi, I'm trying connect NRF24L01 with Attiny84, I've connect ce pin and csn pin in the same Attiny84 pin, PA3, or 3, and it already stays in while(1) in the setup,  it's like nrf is not able to communicate with Attiny84. I upload the program using arduino uno as isp, but I've disconnect arduino uno pins to Attiny84 and it still doesn't work.

dparson55, 2019-03-28:
Are you using the ATtiny library https://github.com/damellis/attiny ?  I've seen issues where the pin numbers for different ATtiny libraries are different between libraries.

AlvaroGordo, 2019-03-28:
Yes, I am using this ATtiny library. I saw that this pin problem existed but I was already using this library. I tried installing the version of David A. Mellis 1.0.0 and 1.0.1, but it still does not work.

AlvaroGordo, 2019-03-28:
Sorry, I mean 1.0.1 and 1.0.2 versions.

AlvaroGordo, 2019-04-01:
It works with 2-pin configuration.

dparson55, 2019-04-01:
Sorry for the delay, I had to borrow an ATtiny84 from a friend and have it in hand now, so I'll test things out later tonight.  I haven't made any recent changes to the library that would effect the ATtiny84 operation, so don't have any idea what the problem would be at the moment.  I'll just run a test tonight and see if I can reproduce the issue.

dparson55, 2019-04-02:
I verified that the **Basic_TX_ATtiny84** example communicates with the **Basic_RX** example in NRFLite 2.0.10.  I used an Ardiuno Uno to run the Basic_RX sketch and an ATtiny84 running at both 1MHz and 8MHz, and each speed worked ok.  I used the ATtiny84 pins documented at the top of the Basic_TX_ATtiny84 sketch, sharing physical pin 10 (Arduino pin 3) with the CE and CSN pins of the nRF24L01 radio module.  I used the same type of radio module on the TX and RX, specifically the module that has the antenna built into the PCB.

Here are the steps:
* Verify you're using 2.0.10 of NRFLite.
* Connect the programming pins of the Arduino ISP to the ATtiny84.
    * Do not connect any nRF24L01 pins to the ATtiny84, if it is powered on it can interfere with programming.
* Set the speed of the ATtiny84 to 1MHz or 8MHz in the Tools menu.
* Go to **Tools > Burn Bootloader** to assign the speed to the ATtiny (it sets the fuse bits).
* Upload the **Basic_TX_ATtiny84** sketch to the ATtiny84, ensuring there are no errors shown in the console window.
* Unplug the ATtiny84 from the Arduino ISP.
* Upload the **Basic_RX** sketch to the Arduino Uno.
* Disconnect power from the Arduino Uno.
* Connect a nRF24L01 radio module to the Arduino Uno using the pins detailed at the top of the **Basic_RX** sketch.
* Connect the Arduino Uno + nRF24L01 circuit to the computer and open the serial monitor.
* Connect the nRF24L01 pins to the ATtiny84 as detailed at the top of the **Basic_TX_ATtiny84** sketch.
* Connect 3.3V and GND to the ATtiny84 + nRF24L01 circuit.  It will hopefully begin transmitting and you will see received data in the serial monitor.

I will have to close this issue since I confirmed the example works ok, the problem is not with the library.

closed 2019-04-02 by dparson55