dparson55/NRFLite #21: Passive receiver

skelstar, 2019-04-11:
Dude, love this library and how simple things are.

One thing I would like to do is have a Tx, Rx, and some sort of passive receiver that (minimum) does not respond. It would be completely passive/quiet.

Looking through your library I have not seen anything obvious that will prevent the acknowledgement.

I don't want anything changed. I can do that locally.

Thanks for your attention! :)

dparson55, 2019-04-11:
Thanks, I'm glad some people like you have found it :)

Would the NO_ACK feature work for your situation?  If you check https://github.com/dparson55/NRFLite/blob/master/examples/Basic_TX/Basic_TX.ino#L72 you'll see there are 2 ways to send a data packet:  REQUIRE_ACK and NO_ACK.  If you do a NO_ACK send, the receiver will not reply with an acknowledgement.

skelstar, 2019-04-12:
Thanks for responding. Cheers!

closed 2019-04-12 by skelstar