dparson55/NRFLite #23: Question on two-way transmission for NRFLite library

dparson55, 2019-04-28:
I was using your library NRFLite, and found it to work much better than the RF24 library, which for me the basic getting started sketch doesn't work. However, for my application I would like to be able to toggle each radio to send and receive on the fly, and I didn't see anything in your examples which allows that.

Is there some easy implementation I could use to accomplish this effect?

I'm building a remote control designed to control an RGBW led strip, but I want to put a button on the base station which can trigger the remote to make a noise when I can't find it.

I was thinking of having the remote periodically ping the base station, to give it an opportunity to return ack data instructing it to beep. However, since I'm trying to make this application ultra power efficient, I would like to keep it in a low power state while not in use.

Let me know if you have any recommendations, and thanks so much for the awesome library!!

dparson55, 2019-04-28:
Glad to hear you've gotten things working with NRFLite, and I'm super interested in the same issue you're facing:  how to have a low power device "receive" a notification.  The approach you described works well and I've implemented it before:  have the battery powered device periodically check to see if it should do something.  I should make a video of this actually because I get asked about it from time to time.

First off though, NRFLite switches the operation mode of the radio module (the nRF24L01+) based on the methods you all.  Whenever you call **init**, **hasData**, or **hasDataISR**, NRFLite puts the radio into RX mode.  Whenever you call **send** or **startSend**, it puts the radio into TX mode.  So you can send a data packet with **send** and afterward call **hasData** to put the radio back into receiving mode.  You don't need to do any of this though, instead you should use ACK data packets as you mentioned.

There is a trick with ACK data packets though, the RX radio has to load the ACK data packet prior to receiving the sent data packet from the TX radio.  So the RX radio doesn't give us an opportunity to process a received packet and then load a customized ACK data packet based upon it.  Here is the flow to better explain this limitation:

- RX radio - Call **addAckData** to load an ACK data packet.
- TX radio - Call **send** to transmit a data packet.  The RX radio will receive the packet and then send back the ACK data packet that was loaded via step 1.
- TX radio - Call **hasAckData** to see if the RX radio provided an ACK data packet.
- TX radio - Call **readData** to place the ACK data packet that was created in step 1 into a variable that can now be processed in the TX code.

To get around this limitation you can send 2 data packets:  the first one indicates that the RX radio should load an ACK data packet, and then the second one will allow the TX radio to receive that ACK data packet.  Unfortunately I don't have an example in the library that shows this approach, but the Sensor example, described here https://youtu.be/URMmgQuPZVc?t=488 has most of the pieces of code that would be needed and would be a good starting point.

If you get stuck just let me know and I can create a more specific example for your situation.

fieryWaters, 2019-05-01:
I got it to work, thank you so much!

fieryWaters, 2019-05-01:
I got it to work in a loop easily, and also with interrupts when I took out the Serial.println() statements. I think it's because I read somewhere else that Serial.println itself uses interrupts, so it doesn't work well in an interrupt.

dparson55, 2019-05-02:
Glad to hear and good point about using Serial in the interrupt handler.  I'll update the examples to do it properly.  I should just set a flag in the interrupt handler and then check for that flag in the main loop.  If it's true, do the Serial prints.

fieryWaters, 2019-05-02:
Sadly, simpler is only simpler if it doesn't introduce weird bugs. I think it's worth updating the tutorial (with a comment about the nuances of interrupts and Serial) because the layman programmer will not be privy to the nature of Serial off hand. I've had several disconcerting instances where using prints causes my code not to work. I would like to note that I believe your examples do work out of the box, but you must be very careful where the print statements are placed. I recommend placing print statements in the loop and calling Serial.flush() before calling startSend() on the radio.

I noticed a couple bugs when trying to do two way communication, I'll describe them in case you ever want to post a tutorial on it. I'll upload my code once it looks pretty and I can simplify it for demonstration purposes.

Because startSend() relies on interrupts, it's possible to start the next packet before the previous one finishes. This was causing my program to crash. The solution to this was to create a flag which showed that the previous packet either failed or succeeded before starting the next send. I only allowed the radio to start the next operation after it has received the interrupt signifying it has finished the current one.

Another issue was that calling hasData() in the main loop successfully switches the radio over to RX mode, yet if startSend() has not finished sending the full packet before switching modes, the packet can get stuck in the buffer (or something) until the next time the radio goes into TX mode. The effect is that the baseStation never received the most recent command from the remote. So if I send packets numbered 1,2,3,4,5, the base station would only ever receive packets 1,2,3. If I sent 1,2,3,4,5,6,7,8, the base station would only receive 1,2,3,4,5,6. It was always two packets behind. Again the solution was to wait for the interrupt to show that the transmission was either a success for failure before allowing the radio to switch into TX mode.

Two-way transmission without interrupts is much simpler, as you don't have the timing issues, but hopefully I can get to ultra-low standby power consumption putting the arduino to sleep until woken by the interrupt. Would you be interested in me posting a sketch demonstrating the use of the IRQ pin to wake an arduino, then sending a response without ackData or polling? I would be happy to help you form a tutorial on low-power two-way transmission, but is that something you're interested in?

Again thanks so much for your help, I'm happy to return the favor!

dparson55, 2019-05-02:
Thanks so much for all this feedback.  It would be great if you could post a snippet of your RX and TX code too, I'm sure it will be of help to someone in the future.

I'll be sure to update the IRQ examples so the Serial prints are outside the interrupt handlers.  If someone is going to attempt using interrupts they are in for a world of complexity anyway, so it makes sense to just show how to use Serial appropriately.

I might be able to shed some light on the behavior you experienced.  The nRF24L01+ has a TX buffer that can hold up to 3 outgoing packets.  When you call startSend, it in turn calls a helper method name prepForTx which [waits until at least 1 free spot is in the TX buffer](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L440).  This wait logic was added when I wrote a [test that calls sendData as fast as possible](https://github.com/dparson55/NRFLite/blob/master/examples/Test_TX/Test_TX.ino#L363) and the test failed with a lot of missing packets because the TX buffer was full.  This logic fixed this particular test and allowed startSend to be called as fast as possible, at least for this particular situation.  I'm not sure what would have caused the lockup in your situation but would be happy to debug it if you'd like to send over code that reproduces the problem.

I'm going to adjust hasData to account for the situation you faced when switching from TX to RX mode.  The hasData method already has a lot of protections inside of it so adding another protection for if it's in the middle of a transmission will not be a problem.  I will update it to see if the radio is in TX mode and if so, wait until the TX buffer is empty before switching into RX mode.

So I've got some work to do tonight on the library and I'll let you know when the new version is ready.

dparson55, 2019-05-03:
Improvements to the examples are complete and I've [started work](https://github.com/dparson55/NRFLite/commit/fe9fe97d859ebff3e2cf58638e2e989483a339bd) on modifying hasData in a new branch.

dparson55, 2019-05-06:
During all the refactoring I was able to reproduce the issue you described with startSend.  I'm still struggling to find a solution.

The issue only occurs about 10% of the time for me so it is hard to debug.  I believe there are 4 interrupts at play that are making this difficult:  the timer 0 interrupt used to increment the millis() and micros() Arduino counters, the SPI interrupts, Serial interrupts, and finally the radio's interrupt.  I spent a few hours Saturday and Sunday trying different approaches and no luck so far, but I have another idea that I'll attempt on Monday.

fieryWaters, 2019-05-06:
Howdy! I thought I'd help you out as I managed to get this error to happen everytime! First, I'll post my two way code which works flawlessly so far as I can tell, then I'll post the one which gives the error. Hope this helps!

`/*

  The circuit for the REMOTE:
   LCD VSS pin to ground
   LCD VDD pin to 5V
   LCD VO pin to series of resisters
   LCD RS pin to digital pin 8
   LCD RW pin to ground
   LCD E pin to digital pin 7
   LCD D4 pin to digital pin 2
   LCD D5 pin to digital pin 3
   LCD D6 pin to digital pin 4
   LCD D7 pin to digital pin 5
   LCD A pin to digital pin 6 (With 220 Ohm in series)
   LCD K pin to ground

  redWiper to A4
  greenWiper to A5
  blueWiper to A6
  whiteWiper to A7

  buzzer+ to A0 (With resister to very volume)

  http://www.arduino.cc/en/Tutorial/LiquidCrystal
*/

//Radio Data members
#include <SPI.h>
#include <NRFLite.h>
const static uint8_t remoteID = 1;
const static uint8_t baseStationID = 0;
const static uint8_t CE = 9;
const static uint8_t CSN = 10;
const static uint8_t radioIRQ = 2;
uint8_t data = 0;
boolean readyToSend = true;
NRFLite radio;
//uint32_t packetCount = 0;//testing
uint8_t txOk, txFail, rxReady;

struct colorPacket
{
  uint8_t radioId = remoteID;
  uint8_t red;
  uint8_t green;
  uint8_t blue;
  uint8_t white;
  //uint32_t count;//testing
};
//LCD code
#include <LiquidCrystal.h>
LiquidCrystal lcd(8, 7, 3, 4, 5, 6  );
const static uint8_t backLightPin = 6;//PWM enabled

const static uint8_t buzzerPin = A0;

/* This values are sent to the light module */
const static uint8_t redWiper = A4;
const static uint8_t greenWiper = A5;
const static uint8_t blueWiper = A6;
const static uint8_t whiteWiper = A7;
const static uint8_t alarmCode = 255;

colorPacket lastValues;

void setup()
{
  // set up the LCD's number of columns and rows:

  Serial.begin(115200);

  if (!radio.init(remoteID, CE, CSN))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }
  else
  {
    Serial.println("Radio working fine");
  }

  lcd.begin(16, 2);
  pinMode(backLightPin, OUTPUT);
  analogWrite(backLightPin, 128);

  lastValues = getKnobValues();

  attachInterrupt(digitalPinToInterrupt(radioIRQ), radioInterrupt, FALLING);

}

void loop() {
  //lcd.clear();
  //lcd.setCursor(0, 0);
  sendIfStateChanges();
  if (readyToSend)
    radio.hasData();//working great

  if (data == 255)
  {
    Serial.println(data);
    buzzForever();
  }

  //Serial.println(data);
}

//int count = 0;
uint32_t timeout = 0;
void sendIfStateChanges()
{
  if (txFail) //ensures every packet gets sent
  {
    Serial.println("FAIL");
    Serial.flush();
    radio.startSend(baseStationID, &lastValues, sizeof(lastValues));
    txFail = 0;
    return;
  }

  colorPacket current = getKnobValues();
  if
  (
    (abs(current.red - lastValues.red) > 1) || /*Checks to see if there is a change greater than the noise of the potentiometers*/
    (abs(current.green - lastValues.green) > 1) ||
    (abs(current.blue - lastValues.blue) > 1) ||
    (abs(current.white - lastValues.white) > 1) ||

    (current.red == 0 && lastValues.red != 0) ||/*Checks to see if any potentiometer has hit zero*/
    (current.green == 0 && lastValues.green != 0) ||
    (current.blue == 0 && lastValues.blue != 0) ||
    (current.white == 0 && lastValues.white != 0)
  )
  {

    if (readyToSend)
    {
      readyToSend = false;
      lastValues = current;
      /*
        Serial.print(" REMOTE Red: "); Serial.print(lastValues.red);
        Serial.print(", green: "); Serial.print(lastValues.green);
        Serial.print(", blue: "); Serial.print(lastValues.blue);
        Serial.print(", white: "); Serial.println(lastValues.white);
        //Serial.print(", count: "); Serial.println(packetCount);
        Serial.flush();
      //*/
      radio.startSend(baseStationID, &lastValues, sizeof(lastValues)); // Note how '&' must be placed in front of the variable name.

    }
    //}

  }
}

colorPacket getKnobValues()
{
  colorPacket colors;
  colors.red = analogRead(redWiper) / 4;
  colors.green = analogRead(greenWiper) / 4;
  colors.blue = analogRead(blueWiper) / 4;
  colors.white = analogRead(whiteWiper) / 4;
  //colors.count = packetCount;
  return colors;
}

void radioInterrupt()
{
  radio.whatHappened(txOk, txFail, rxReady);
  if (rxReady)
  {
    // Use 'hasDataISR' rather than 'hasData' when using interrupts.
    while (radio.hasDataISR())
    {
      radio.readData(&data);
    }
  } else if (txOk)
  {
    readyToSend = true;
    //packetCount++;//counts total number of packets sent
  }
  else if (txFail);
  {
    readyToSend = true;
  }
}

void buzzForever()
{
  while (1)
  {
    tone(buzzerPin, 1000);
    delay(1000);
    noTone(buzzerPin);
    delay(1000);
  }
}`

fieryWaters, 2019-05-06:
/*

BASE Station

  Radio    Arduino
  CE    -> 8
  CSN   -> 10 (Hardware SPI SS)
  MOSI  -> 11 (Hardware SPI MOSI)
  MISO  -> 12 (Hardware SPI MISO)
  SCK   -> 13 (Hardware SPI SCK)
  IRQ   -> No connection
  VCC   -> No more than 3.6 volts
  GND   -> GND

*/

#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 8;//change to 8
const static uint8_t PIN_RADIO_CSN = 10;

#define redPin 3
#define greenPin 5
#define bluePin 6
#define whitePin 9

const static uint8_t buzzerButtonPin = 2;

#define destinationId 1

struct colorPacket
{
  uint8_t radioId;
  uint8_t red;
  uint8_t green;
  uint8_t blue;
  uint8_t white;
  //uint32_t count;//testing
};

NRFLite radio;
colorPacket colors;

void setup()
{
  Serial.begin(115200);

  if (!radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }
  else
  {
    Serial.println("radio works fine");
  }
  pinMode(redPin, OUTPUT);
  pinMode(greenPin, OUTPUT);
  pinMode(bluePin, OUTPUT);
  pinMode(whitePin, OUTPUT);

  pinMode(buzzerButtonPin, INPUT_PULLUP);

  //attachInterrupt(digitalPinToInterrupt(interuptPin), findRemote, FALLING);
}

uint8_t alarmCode = 255;
//int count = 0;
void loop()
{
  if (radio.hasData())
  {
    do
    {
      radio.readData(&colors); // Note how '&' must be placed in front of the variable name.
    } while (radio.hasData());
    analogWrite(redPin, colors.red);
    analogWrite(greenPin, colors.green);
    analogWrite(bluePin, colors.blue);
    analogWrite(whitePin, colors.white);
    /*
        Serial.print("BaseStation Red: "); Serial.print(colors.red);
        Serial.print(", green: "); Serial.print(colors.green);
        Serial.print(", blue: "); Serial.print(colors.blue);
        Serial.print(", white: "); Serial.println(colors.white);
        //Serial.print(", count: "); Serial.print(colors.count);
        //Serial.print(", internalCount: "); Serial.println(count);
        Serial.flush();
        //count++;
      //*/
  }

  if (buttonPressed(buzzerButtonPin, 500))
  {
    if (radio.send(destinationId, &alarmCode, sizeof(alarmCode))) // Note how '&' must be placed in front of the variable name.
    {
      Serial.println("Sent alarmCode sucessfully");
    }
    else
    {
      Serial.println("...FAIL");
    }
  }

}

uint32_t lastTimePressed = 0;
int buttonPressed(byte buttonPin, uint16_t timeout)
{
  if (digitalRead(buttonPin) == 0)
  {
    //delay(1000);
    if ((millis() - lastTimePressed) > timeout)
    {
      /*

        Serial.print("Timeout: ");
        Serial.println(timeout);
        Serial.print("LastTimePressed ");
        Serial.println(lastTimePressed);
        Serial.print("Millis() - lastTimePressed");
        Serial.println((millis() - lastTimePressed));
        Serial.println();
      */
      lastTimePressed = millis();
      //Serial.println(1);
      return 1;
    }
  }
  //Serial.println(0);
  return 0;
}

fieryWaters, 2019-05-06:
The most important factor I found is to ensure that the radio is done with it's operations before starting another task! 

Below is the buggy code. I'm keeping track of how many packets are send vs received. I noticed how the base station always receives a packet that was sent two times ago.

fieryWaters, 2019-05-06:
/*

  The circuit for the REMOTE:
   LCD VSS pin to ground
   LCD VDD pin to 5V
   LCD VO pin to series of resisters
   LCD RS pin to digital pin 8
   LCD RW pin to ground
   LCD E pin to digital pin 7
   LCD D4 pin to digital pin 2
   LCD D5 pin to digital pin 3
   LCD D6 pin to digital pin 4
   LCD D7 pin to digital pin 5
   LCD A pin to digital pin 6 (With 220 Ohm in series)
   LCD K pin to ground

  redWiper to A4
  greenWiper to A5
  blueWiper to A6
  whiteWiper to A7

  buzzer+ to A0 (With resister to very volume)

  http://www.arduino.cc/en/Tutorial/LiquidCrystal
*/

//Radio Data members
#include <SPI.h>
#include <NRFLite.h>
const static uint8_t remoteID = 1;
const static uint8_t baseStationID = 0;
const static uint8_t CE = 9;
const static uint8_t CSN = 10;
const static uint8_t radioIRQ = 2;
uint8_t data = 0;
boolean readyToSend = true;
NRFLite radio;
uint32_t packetCount = 0;//testing
uint8_t txOk, txFail, rxReady;

struct colorPacket
{
  uint8_t radioId = remoteID;
  uint8_t red;
  uint8_t green;
  uint8_t blue;
  uint8_t white;
  uint32_t count;//testing
};
//LCD code
#include <LiquidCrystal.h>
LiquidCrystal lcd(8, 7, 3, 4, 5, 6  );
const static uint8_t backLightPin = 6;//PWM enabled

const static uint8_t buzzerPin = A0;

/* This values are sent to the light module */
const static uint8_t redWiper = A4;
const static uint8_t greenWiper = A5;
const static uint8_t blueWiper = A6;
const static uint8_t whiteWiper = A7;
const static uint8_t alarmCode = 255;

colorPacket lastValues;

void setup()
{
  // set up the LCD's number of columns and rows:

  Serial.begin(115200);

  if (!radio.init(remoteID, CE, CSN))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }
  else
  {
    Serial.println("Radio working fine");
  }

  lcd.begin(16, 2);
  pinMode(backLightPin, OUTPUT);
  analogWrite(backLightPin, 128);

  lastValues = getKnobValues();

  attachInterrupt(digitalPinToInterrupt(radioIRQ), radioInterrupt, FALLING);

}

void loop() {
  //lcd.clear();
  //lcd.setCursor(0, 0);
  sendIfStateChanges();
  //if (readyToSend)
    radio.hasData();//working great

  if (data == 255)
  {
    Serial.println(data);
    buzzForever();
  }

  //Serial.println(data);
}

int count = 0;
uint32_t timeout = 0;
void sendIfStateChanges()
{
  if (txFail) //ensures every packet gets sent
  {
    Serial.println("FAIL");
    Serial.flush();
    radio.startSend(baseStationID, &lastValues, sizeof(lastValues));
    txFail = 0;
    return;
  }

  colorPacket current = getKnobValues();
  if
  (
    (abs(current.red - lastValues.red) > 1) || /*Checks to see if there is a change greater than the noise of the potentiometers*/
    (abs(current.green - lastValues.green) > 1) ||
    (abs(current.blue - lastValues.blue) > 1) ||
    (abs(current.white - lastValues.white) > 1) ||

    (current.red == 0 && lastValues.red != 0) ||/*Checks to see if any potentiometer has hit zero*/
    (current.green == 0 && lastValues.green != 0) ||
    (current.blue == 0 && lastValues.blue != 0) ||
    (current.white == 0 && lastValues.white != 0)
  )
  {

    if (readyToSend)
    {
      readyToSend = false;
      lastValues = current;
      
        Serial.print(" REMOTE Red: "); Serial.print(lastValues.red);
        Serial.print(", green: "); Serial.print(lastValues.green);
        Serial.print(", blue: "); Serial.print(lastValues.blue);
        Serial.print(", white: "); Serial.print(lastValues.white);
        Serial.print(", count: "); Serial.println(packetCount);
        Serial.flush();
      
      radio.startSend(baseStationID, &lastValues, sizeof(lastValues)); // Note how '&' must be placed in front of the variable name.

    }
    //}

  }
}

colorPacket getKnobValues()
{
  colorPacket colors;
  colors.red = analogRead(redWiper) / 4;
  colors.green = analogRead(greenWiper) / 4;
  colors.blue = analogRead(blueWiper) / 4;
  colors.white = analogRead(whiteWiper) / 4;
  colors.count = packetCount;
  return colors;
}

void radioInterrupt()
{
  radio.whatHappened(txOk, txFail, rxReady);
  if (rxReady)
  {
    // Use 'hasDataISR' rather than 'hasData' when using interrupts.
    while (radio.hasDataISR())
    {
      radio.readData(&data);
    }
  } else if (txOk)
  {
    readyToSend = true;
    packetCount++;//counts total number of packets sent
  }
  else if (txFail);
  {
    readyToSend = true;
  }
}

void buzzForever()
{
  while (1)
  {
    tone(buzzerPin, 1000);
    delay(1000);
    noTone(buzzerPin);
    delay(1000);
  }
}

fieryWaters, 2019-05-06:
Terminal output:

Radio working fine
 REMOTE Red: 23, green: 23, blue: 24, white: 23, count: 0
 REMOTE Red: 25, green: 25, blue: 25, white: 25, count: 1
 REMOTE Red: 27, green: 26, blue: 26, white: 26, count: 2
 REMOTE Red: 27, green: 27, blue: 28, white: 27, count: 3
 REMOTE Red: 28, green: 28, blue: 28, white: 29, count: 4
 REMOTE Red: 30, green: 30, blue: 29, white: 29, count: 5
 REMOTE Red: 31, green: 31, blue: 31, white: 31, count: 6
 REMOTE Red: 32, green: 32, blue: 33, white: 33, count: 7
 REMOTE Red: 34, green: 33, blue: 33, white: 33, count: 8

fieryWaters, 2019-05-06:
/*

  Demonstrates simple RX and TX operation.
  Please read the notes in NRFLite.h for a description of all library features.

  Radio    Arduino
  CE    -> 8
  CSN   -> 10 (Hardware SPI SS)
  MOSI  -> 11 (Hardware SPI MOSI)
  MISO  -> 12 (Hardware SPI MISO)
  SCK   -> 13 (Hardware SPI SCK)
  IRQ   -> No connection
  VCC   -> No more than 3.6 volts
  GND   -> GND

*/

#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 8;//change to 8
const static uint8_t PIN_RADIO_CSN = 10;

#define redPin 3
#define greenPin 5
#define bluePin 6
#define whitePin 9

const static uint8_t buzzerButtonPin = 2;

#define destinationId 1

struct colorPacket
{
  uint8_t radioId;
  uint8_t red;
  uint8_t green;
  uint8_t blue;
  uint8_t white;
  uint32_t count;//testing
};

NRFLite radio;
colorPacket colors;

void setup()
{
  Serial.begin(115200);

  if (!radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }
  else
  {
    Serial.println("radio works fine");
  }
  pinMode(redPin, OUTPUT);
  pinMode(greenPin, OUTPUT);
  pinMode(bluePin, OUTPUT);
  pinMode(whitePin, OUTPUT);

  pinMode(buzzerButtonPin, INPUT_PULLUP);

  //attachInterrupt(digitalPinToInterrupt(interuptPin), findRemote, FALLING);
}

uint8_t alarmCode = 255;
int count = 0;
void loop()
{
  if (radio.hasData())
  {
    do
    {
      radio.readData(&colors); // Note how '&' must be placed in front of the variable name.
    } while (radio.hasData());
    analogWrite(redPin, colors.red);
    analogWrite(greenPin, colors.green);
    analogWrite(bluePin, colors.blue);
    analogWrite(whitePin, colors.white);

    Serial.print("BaseStation Red: "); Serial.print(colors.red);
    Serial.print(", green: "); Serial.print(colors.green);
    Serial.print(", blue: "); Serial.print(colors.blue);
    Serial.print(", white: "); Serial.print(colors.white);
    Serial.print(", countFromRemote: "); Serial.print(colors.count);
    Serial.print(", internalCount: "); Serial.println(count);
    Serial.flush();
    count++;

  }

  if (buttonPressed(buzzerButtonPin, 500))
  {
    if (radio.send(destinationId, &alarmCode, sizeof(alarmCode))) // Note how '&' must be placed in front of the variable name.
    {
      Serial.println("Sent alarmCode sucessfully");
    }
    else
    {
      Serial.println("...FAIL");
    }
  }

}

uint32_t lastTimePressed = 0;
int buttonPressed(byte buttonPin, uint16_t timeout)
{
  if (digitalRead(buttonPin) == 0)
  {
    //delay(1000);
    if ((millis() - lastTimePressed) > timeout)
    {
      /*

        Serial.print("Timeout: ");
        Serial.println(timeout);
        Serial.print("LastTimePressed ");
        Serial.println(lastTimePressed);
        Serial.print("Millis() - lastTimePressed");
        Serial.println((millis() - lastTimePressed));
        Serial.println();
      */
      lastTimePressed = millis();
      //Serial.println(1);
      return 1;
    }
  }
  //Serial.println(0);
  return 0;
}

fieryWaters, 2019-05-06:
radio works fine
BaseStation Red: 23, green: 23, blue: 24, white: 23, countFromRemote: 0, internalCount: 0
BaseStation Red: 25, green: 25, blue: 25, white: 25, countFromRemote: 1, internalCount: 1
BaseStation Red: 25, green: 25, blue: 25, white: 25, countFromRemote: 1, internalCount: 2
BaseStation Red: 27, green: 26, blue: 26, white: 26, countFromRemote: 2, internalCount: 3
BaseStation Red: 27, green: 26, blue: 26, white: 26, countFromRemote: 2, internalCount: 4
BaseStation Red: 27, green: 27, blue: 28, white: 27, countFromRemote: 3, internalCount: 5
BaseStation Red: 27, green: 27, blue: 28, white: 27, countFromRemote: 3, internalCount: 6
BaseStation Red: 28, green: 28, blue: 28, white: 29, countFromRemote: 4, internalCount: 7
BaseStation Red: 28, green: 28, blue: 28, white: 29, countFromRemote: 4, internalCount: 8
BaseStation Red: 30, green: 30, blue: 29, white: 29, countFromRemote: 5, internalCount: 9
BaseStation Red: 30, green: 30, blue: 29, white: 29, countFromRemote: 5, internalCount: 10
BaseStation Red: 31, green: 31, blue: 31, white: 31, countFromRemote: 6, internalCount: 11

fieryWaters, 2019-05-06:
There are two strange things going on here, the first is that the baseStation only ever receives packet 6, when the last packet transfered was packet 8. If I twist the brightness knob more, the difference between the packet number received and send is always 2.

Also, the baseStation is receives many copies of the packet from the remote. This suggests to me that the communication is interrupted before the baseStation can send a confirmation that the signal is received.

fieryWaters, 2019-05-06:
I hope this helps your debugging!

dparson55, 2019-05-07:
Thanks for sending the code over.  I let the TX count reach 20,000 a few times and unfortunately it never did cause a lockup.  So I'll just continue debugging the interrupt issue that I see from time to time (which isn't related to your original issue, it's just something I noticed when trying to modify hasData).

The issue you are describing about the base station being behind on packets is that hasData issue (1 of the 3 queued packets makes it out before hasData reconfigures the radio to RX mode).  If you use ACK data packets (the recommended way of doing 2-way communication) everything would be ok.  So that's another strong reason for following the 2-way communication approach the designers of the radio intended:  ACK data packets.

So I'll keep on plugging away on the dang interrupt issue and good luck to you on your work to make a low power design for your remote!

closed 2019-05-07 by dparson55

dparson55, 2019-05-10:
@fieryWaters NRFLite 2.1.0 has been published and includes the fix for hasData.

Rmdhn1, 2022-09-11:
i'm planning to make a 2.4ghz gamepad with RGB, thus it needs to send buttons and potentiometer data almost instantly and receive rgb data, can i just switch between has and send like you've said?