dparson55/NRFLite #24: No return from send() function

k64t34, 2019-05-06:
Hi
I have arduino Nano ATmega328p and NRF24L01+PA+LNA Wireless Module FZ0410.
I try to run example Basic_TX.
When  call send() function inside loop() function,  it dosn't  return to loop() funcion.

I tryed to debug send() function in NRFLite.cpp and saw, that there is  infinite loop inside while(1) code in send function because all times code goto "some else" block.

**NRFLite.cpp**
...
while (1)
    {		
        delayMicroseconds(_transmissionRetryWaitMicros);
        statusReg = readRegister(STATUS_NRF);
        if (statusReg & _BV(TX_DS))
        {
            writeRegister(STATUS_NRF, statusReg | _BV(TX_DS));  // Clear TX success flag.
            return 1;                                           // Return success.
        }
        else if (statusReg & _BV(MAX_RT))
        {
            spiTransfer(WRITE_OPERATION, FLUSH_TX, NULL, 0);    // Clear TX FIFO buffer.
            writeRegister(STATUS_NRF, statusReg | _BV(MAX_RT)); // Clear flag which indicates max retries has been reached.
            return 0;                                           // Return failure.
        }
		else {Serial.println("some else");}
    }
...
**Basic_TX.ino**
/*
Demonstrates simple RX and TX operation.
Please read the notes in NRFLite.h for a description of all library features.
Radio    Arduino
CE    -> 9
CSN   -> 10 (Hardware SPI SS)
MOSI  -> 11 (Hardware SPI MOSI)
MISO  -> 12 (Hardware SPI MISO)
SCK   -> 13 (Hardware SPI SCK)
IRQ   -> No connection
VCC   -> No more than 3.6 volts
GND   -> GND
*/

#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 1;             // Our radio's id.
const static uint8_t DESTINATION_RADIO_ID = 0; // Id of the radio we will transmit to.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
    Serial.begin(9600);
    Serial.println("");
        
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
    Serial.println("Success communicate with radio");
    _radioData.FromRadioId = RADIO_ID;
}

void loop()
{
    _radioData.OnTimeMillis = millis();

    Serial.print("Sending ");
    Serial.print(_radioData.OnTimeMillis);
    Serial.print(" ms");

    if (_radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData))) // Note how '&' must be placed in front of the variable name.
    {
        Serial.println("...Success");
    }
    else
    {
        Serial.println("...Failed");
        _radioData.FailedTxCount++;
    }

    delay(1000);
    
    /*
    By default, 'send' transmits data and waits for an acknowledgement.
    You can also send without requesting an acknowledgement.
    _radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData), NRFLite::NO_ACK)
    _radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData), NRFLite::REQUIRE_ACK) // THE DEFAULT
    */
}

dparson55, 2019-05-06:
Nice work debugging where the lockup is occurring.  Can you answer these questions:
* Are you using a NRF24L01+PA+LNA module to transmit and a second one to receive?
* Is the RX radio getting data that is sent?
* If you unplug the RX radio (so it cannot receive) and run the TX radio again, does it still lock up in the while loop?
* If you change the send mode to ```NRFLite::NO_ACK```, does it still lock up?

The loop is coded this way because only 2 things are supposed to happen:  the send is successful or it fails.  If it's successful the nRF24L01+ will set the TX_DS flag in the status register, or if it fails (it retries up to 15 times) it will set the MAX_RT flag.  There's no sure way know when exactly it will retry or how long it will take, so that's why we loop until it succeeds or fails.  If the radio doesn't succeed or fail, then there is something wrong with the NRF24L01+PA+LNA module, maybe it is not getting enough power, maybe a capacitor is needed between it's VCC and GND pins, etc.

k64t34, 2019-05-07:
It happens without a receiver at all.
I tried to use NO_ACK also.

dparson55, 2019-05-07:
Ok then, it must be some issue with the transmitting radio module since it's not indicating a failure to send, even when there is no receiver.  So there's a hardware issue.

k64t34, 2019-05-08:
Data transfer is working fine using RF24.h.
If you do not add condition processing "some else" , I will not be able to use NRFLite.
It's a pity. The project is good.
Can you still close the leak?

dparson55, 2019-05-09:
Ok that's great, at least you have a solution with another library.

I cannot add a 3rd condition to the loop since NRFLite depends on the STATUS register to be updated with the success bit (TX_DS) or the failure bit (MAX_RT), it is the only way to know when the transmission is successful or not.  The loop needs to occur many times since there is no way to know how long it will take for the radio to send the data.

Since the STATUS register isn't being updated, your radio module is acting like it did not receive the command to send the data.  I looked at how the RF24.h library is coded and it leaves the CE pin HIGH all the time, but NRFLite uses a different approach where it [sets the CE pin HIGH, waits 10 microseconds, and then toggles it back to LOW](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L203).

So can you try changing ```delayMicroseconds(CE_TRANSMISSION_MICROS);``` to ```delayMicroseconds(15);``` so the CE pin stays HIGH for longer?  I currently have ```CE_TRANSMISSION_MICROS``` is [set to 10](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L71) but maybe this is too quick for your radio module.

If this doesn't help, try increasing the delay further, e.g. to 20, 100, 1000.  If that doesn't help, try removing the entire ```digitalWrite(_cePin, LOW);``` command so the CE pin is left HIGH the entire time.

If that still doesn't help, could you update the line ```NRFLite _radio;``` to be ```NRFLite _radio(Serial);``` in **Basic_TX.ino** and then change the _send_ method in **NRFLite.cpp** to read as follows:

```
uint8_t NRFLite::send(uint8_t toRadioId, void *data, uint8_t length, SendType sendType)
{
    prepForTx(toRadioId, sendType);

    debugln("Clearing STATUS register");
    printDetails();

    // Clear any previously asserted TX success or max retries flags.
    uint8_t statusReg = readRegister(STATUS_NRF);
    if (statusReg & _BV(TX_DS) || statusReg & _BV(MAX_RT))
    {
        writeRegister(STATUS_NRF, statusReg | _BV(TX_DS) | _BV(MAX_RT));
    }
    
    debugln("Adding data");
    printDetails();

    // Add data to the TX FIFO buffer, with or without an ACK request.
    if (sendType == NO_ACK) { spiTransfer(WRITE_OPERATION, W_TX_PAYLOAD_NO_ACK, data, length); }
    else                    { spiTransfer(WRITE_OPERATION, W_TX_PAYLOAD       , data, length); }

    // Start transmission.
    // If we have separate pins for CE and CSN, CE will be LOW and we must pulse it to start transmission.
    // If we use the same pin for CE and CSN, CE will already be HIGH and transmission will have started
    // when data was loaded into the TX FIFO.
    uint8_t txPulseIsNeeded = _cePin != _csnPin;
    if (txPulseIsNeeded)
    {
        digitalWrite(_cePin, HIGH);
        delayMicroseconds(CE_TRANSMISSION_MICROS);
        digitalWrite(_cePin, LOW);
    }
    
    // Wait for transmission to succeed or fail.
    while (1)
    {
        delayMicroseconds(_transmissionRetryWaitMicros);
        statusReg = readRegister(STATUS_NRF);
        
        if (statusReg & _BV(TX_DS))
        {
            writeRegister(STATUS_NRF, statusReg | _BV(TX_DS));  // Clear TX success flag.
            return 1;                                           // Return success.
        }
        else if (statusReg & _BV(MAX_RT))
        {
            spiTransfer(WRITE_OPERATION, FLUSH_TX, NULL, 0);    // Clear TX FIFO buffer.
            writeRegister(STATUS_NRF, statusReg | _BV(MAX_RT)); // Clear flag which indicates max retries has been reached.
            return 0;                                           // Return failure.
        }
        else
        {
            debugln("Waiting for success or failure");
            printDetails();
        }
    }
}
```

This will print the values of many registers inside the radio in your Serial monitor, and you can post the results here to help with the debugging.

k64t34, 2019-05-10:
Hi
Thanks for the help.
I reassembled  the project on a different breadboard and now the failure does not occur.

But I continued research.
I replaced.
debugln ("Waiting for success or failure");
on
Serial.print (".");
I also added a time count for sending the package.

The average time to send a packet is 12-15 milliseconds. Is this normal time or should it be less?
`
20:49:18.092 -> Success communicate with radio

20:49:18.126 -> Sending 17 ms Success

20:49:18.126 -> Sending 13 ms Failed

20:49:18.160 -> Sending 14 ms Success

20:49:18.194 -> Sending 13 ms Success

20:49:18.194 -> Sending 15 ms Success

20:49:18.228 -> Sending ...34 ms Success

20:49:18.261 -> Sending 13 ms Success

20:49:18.295 -> Sending 14 ms Success

20:49:18.295 -> Sending 13 ms Success

20:49:18.329 -> Sending 14 ms Success

20:49:18.329 -> Sending ....41 ms Failed

20:49:18.363 -> Sending 14 ms Success

20:49:18.397 -> Sending 13 ms Success

20:49:18.431 -> Sending .21 ms Success

20:49:18.465 -> Sending 14 ms Success

20:49:18.499 -> Sending 13 ms Success

20:49:18.499 -> Sending ..28 ms Failed

20:49:18.533 -> Sending 14 ms Success

20:49:18.567 -> Sending .21 ms Success

20:49:18.601 -> Sending 13 ms Success

20:49:18.635 -> Sending 13 ms Success

20:49:18.635 -> Sending .............104 ms Success

20:49:18.771 -> Sending ..28 ms Failed

20:49:18.805 -> Sending 13 ms Success

20:49:18.839 -> Sending 13 ms Success

20:49:18.839 -> Sending .........77 ms Success

20:49:18.941 -> Sending 14 ms Success

20:49:18.941 -> Sending ..28 ms Success

20:49:19.008 -> Sending 18 ms Success

20:49:19.042 -> Sending 14 ms Success

20:49:19.042 -> Sending ...35 ms Success

20:49:19.110 -> Sending 13 ms Success

20:49:19.110 -> Sending .....48 ms Failed

20:49:19.178 -> Sending .20 ms Success

20:49:19.212 -> Sending 13 ms Success

20:49:19.212 -> Sending ..28 ms Failed

20:49:19.280 -> Sending ....42 ms Failed

20:49:19.314 -> Sending ....41 ms Success

20:49:19.382 -> Sending .25 ms Failed

20:49:19.416 -> Sending ....41 ms Failed

20:49:19.450 -> Sending 13 ms Success

20:49:19.483 -> Sending 13 ms Success

20:49:19.518 -> Sending ..27 ms Success

20:49:19.552 -> Sending 14 ms Success

20:49:19.552 -> Sending ......55 ms Success

20:49:19.621 -> Sending ...........89 ms Success

20:49:19.723 -> Sending 17 ms Success

20:49:19.757 -> Sending .25 ms Failed

20:49:19.791 -> Sending 13 ms Failed

20:49:19.825 -> Sending 14 ms Success

20:49:19.859 -> Sending 13 ms Success

20:49:19.859 -> Sending .............104 ms Success

20:49:19.995 -> Sending 13 ms Success

20:49:20.029 -> Sending 14 ms Success

20:49:20.029 -> Sending 13 ms Success

20:49:20.063 -> Sending .....49 ms Success

20:49:20.097 -> Sending .20 ms Success

20:49:20.165 -> Sending 13 ms Success

20:49:20.165 -> Sending ............98 ms Success

20:49:20.266 -> Sending 13 ms Success

20:49:20.300 -> Sending 14 ms Success

20:49:20.300 -> Sending 14 ms Success

20:49:20.334 -> Sending .........76 ms Success

20:49:20.436 -> Sending .21 ms Success

20:49:20.470 -> Sending 13 ms Success

20:49:20.504 -> Sending ...35 ms Success

20:49:20.538 -> Sending ...35 ms Failed

20:49:20.572 -> Sending 13 ms Success

20:49:20.606 -> Sending ..........83 ms Success

20:49:20.708 -> Sending 18 ms Success

20:49:20.742 -> Sending 14 ms Success

20:49:20.742 -> Sending .........77 ms Success

20:49:20.844 -> Sending 13 ms Success

20:49:20.844 -> Sending ................125 ms Success

20:49:20.980 -> Sending 19 ms Failed

20:49:21.014 -> Sending 14 ms Success

20:49:21.048 -> Sending ....41 ms Failed

20:49:21.082 -> Sending .22 ms Failed

20:49:21.116 -> Sending .20 ms Success

20:49:21.150 -> Sending 14 ms Success

20:49:21.184 -> Sending 13 ms Success

20:49:21.184 -> Sending ....42 ms Failed
`

k64t34, 2019-05-10:
Oh!
There is no printDetails();
Why?

k64t34, 2019-05-10:
I planned to send 25 packets per second.
Sending a packet over 40 milliseconds is not allowed.

Is it possible to add timeout  parameter to the function send ? If the timeout, then return failure.

dparson55, 2019-05-11:
Glad you got it working on another breadboard.

_printDetails_ in [NRFLite.h](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L35) and [NRFLite.cpp](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L241).

Sorry about the slowness.  I just updated the library last night and accidentally introduced a bug which caused the radio to switch into Tx mode every time a packet was set.  It's fixed in 2.1.1 and below are the speed tests using a pair of Arduino Uno's (16MHz microcontroller) and using separate CE and CSN pins (which is faster than using a shared pin).

```
250KBPS
Polling bitrate
  267 packets 68352 bps
Interrupts bitrate
  307 packets 78592 bps

1MBPS
Polling bitrate
  604 packets 154624 bps
Interrupts bitrate
  985 packets 252160 bps

2MBPS
Polling bitrate
  1065 packets 272640 bps
Interrupts bitrate
  1464 packets 374784 bps
```

dparson55, 2019-05-11:
Version 2.1.1 has all the fixes.

dparson55, 2019-05-14:
Haven't heard from you for a few days so I'll assume all is well, so closing the issue.

closed 2019-05-14 by dparson55