dparson55/NRFLite #26: New diagram 2 pin connection

destiny2008git, 2019-05-12:
Hi, 
I tried the 2 pin connection as published and have burned up already 2 NRF24l01 that were perfectly working. R values perhaps causing this?
please advise.

dparson55, 2019-05-12:
Sorry to hear that.  The circuit does work so my only guess is an issue with the connections and maybe a second pair of eyes will help.  So we can try this:

Carefully follow along with the [video at this point](https://www.youtube.com/watch?v=URMmgQuPZVc&t=138) and plug one of your dead radio modules in.  Then take a bunch of pictures where I can see the pins on the radio going into the breadboard.  Measure the resistors you are using and show me pictures of the measured ohm values for confirmation.  Send the pictures to dparson55@hotmail.com and hopefully way we can solve the problem.

Note one change from the video.  You'll see I mention to use a 5K resistor between the MOSI and MISO pins of the radio (in the video I use two 10K's in parallel to achieve this).  But I have since found that just a single 10K between the MOSI and MISO pins works fine.  So all you need is the 0.1uF capacitor, a 1K resistor to separate the CE+CSN pins from the SCK pin, and a 10K resistor between the MOSI and MISO pins.  It's still ok if you use a 5K between MOSI and MISO like I show in the video.  I even tested with just a 1K between these pins and it was ok, but the additional separation with a 5K and eventually 10K didn't cause problems and places less strain on the MISO pin of the radio.

dparson55, 2019-05-12:
Forgot to also mention, include a picture of your VCC voltage for an additional confirmation.

destiny2008git, 2019-05-12:
I rewired the over again and tried the code in the readme.md and I see it send in serial out on both but the once I upload the RX it don't see anything in the console.
I am running one from pro mini and another from Arduino uno. on pins 9,10

dparson55, 2019-05-12:
For 2 pin operation, the code in readme.md will not work.  Instead, use the code in the examples that is for 2 pin operation.  If you just watch the video I made it goes over all of this.

destiny2008git, 2019-05-12:
RX_2pin and TX_2Pin examples correct?

destiny2008git, 2019-05-12:
I tried the above but I keep getting :
23:06:27.511 -> Sending 115 ms...Failed
23:06:28.542 -> Sending 1143 ms...Failed
23:06:29.573 -> Sending 2161 ms...Failed
23:06:30.605 -> Sending 3182 ms...Failed
23:06:31.636 -> Sending 4201 ms...Failed
23:06:32.620 -> Sending 5221 ms...Failed
23:06:33.652 -> Sending 6243 ms...Failed
23:06:34.683 -> Sending 7262 ms...Failed

radio 1,2 on tx and 2 on RX- nothing shows on the terminal of Rx

destiny2008git, 2019-05-12:
I have success now :) was the capacitor on the new nrf I had to add for stability.

dparson55, 2019-05-12:
Good to hear, enjoy!

closed 2019-05-12 by dparson55

destiny2008git, 2019-05-12:
Can I use 2pin on Attiny85? if which pins? thanks a bunch

dparson55, 2019-05-12:
Please read through the documentation, check the examples, and watch the video.  This question is answered by all of them.

destiny2008git, 2019-05-13:
Hi Dave, setup the 2 pin works but i get the following when using the atiny85.
a failed every so often. any insight would be great
23:36:06.093 -> Sending 59561 ms...Success
23:36:07.077 -> Sending 60566 ms...Success
23:36:08.108 -> Sending 61573 ms...Success
23:36:09.093 -> Sending 62579 ms...Success
23:36:10.124 -> Sending 63585 ms...Success
23:36:11.108 -> Sending 64589 ms...Success
23:36:12.140 -> Sending 65595 ms...Success
23:36:13.124 -> Sending 66601 ms...Failed
23:36:14.155 -> Sending 67621 ms...Success
23:36:15.140 -> Sending 68628 ms...Success
23:36:16.171 -> Sending 69635 ms...Success
23:36:17.155 -> Sending 70640 ms...Success
23:36:18.187 -> Sending 71646 ms...Success
23:36:19.171 -> Sending 72650 ms...Success
23:36:20.202 -> Sending 73657 ms...Failed
23:36:21.233 -> Sending 74678 ms...Success
23:36:22.218 -> Sending 75682 ms...Success
23:36:23.249 -> Sending 76689 ms...Success
23:36:24.233 -> Sending 77696 ms...Success
23:36:25.218 -> Sending 78701 ms...Success
23:36:26.249 -> Sending 79706 ms...Success
23:36:27.233 -> Sending 80712 ms...Success
23:36:28.265 -> Sending 81718 ms...Failed
23:36:29.296 -> Sending 82739 ms...Success
23:36:30.280 -> Sending 83745 ms...Success
23:36:31.312 -> Sending 84752 ms...Success
23:36:32.296 -> Sending 85757 ms...Success
23:36:33.327 -> Sending 86762 ms...Success
23:36:34.312 -> Sending 87768 ms...Success
23:36:35.343 -> Sending 88774 ms...Failed
23:36:36.327 -> Sending 89793 ms...Success
23:36:37.358 -> Sending 90799 ms...Success
23:36:38.343 -> Sending 91805 ms...Success
23:36:39.374 -> Sending 92812 ms...Success
23:36:40.358 -> Sending 93817 ms...Success
23:36:41.390 -> Sending 94822 ms...Failed