dparson55/NRFLite #27: failed transmit on Attiny85-2pin

destiny2008git, 2019-05-14:
Hi Dave, setup the 2 pin works but i get the following when using the atiny85.
 a failed every so often. any insight would be great
 23:36:06.093 -> Sending 59561 ms...Success
 23:36:07.077 -> Sending 60566 ms...Success
 23:36:08.108 -> Sending 61573 ms...Success
 23:36:09.093 -> Sending 62579 ms...Success
 23:36:10.124 -> Sending 63585 ms...Success
 23:36:11.108 -> Sending 64589 ms...Success
 23:36:12.140 -> Sending 65595 ms...Success
 23:36:13.124 -> Sending 66601 ms...Failed
 23:36:14.155 -> Sending 67621 ms...Success
 23:36:15.140 -> Sending 68628 ms...Success
 23:36:16.171 -> Sending 69635 ms...Success
 23:36:17.155 -> Sending 70640 ms...Success
 23:36:18.187 -> Sending 71646 ms...Success
 23:36:19.171 -> Sending 72650 ms...Success
 23:36:20.202 -> Sending 73657 ms...Failed
 23:36:21.233 -> Sending 74678 ms...Success
 23:36:22.218 -> Sending 75682 ms...Success
 23:36:23.249 -> Sending 76689 ms...Success
 23:36:24.233 -> Sending 77696 ms...Success
 23:36:25.218 -> Sending 78701 ms...Success
 23:36:26.249 -> Sending 79706 ms...Success
 23:36:27.233 -> Sending 80712 ms...Success
 23:36:28.265 -> Sending 81718 ms...Failed
 23:36:29.296 -> Sending 82739 ms...Success
 23:36:30.280 -> Sending 83745 ms...Success
 23:36:31.312 -> Sending 84752 ms...Success
 23:36:32.296 -> Sending 85757 ms...Success
 23:36:33.327 -> Sending 86762 ms...Success
 23:36:34.312 -> Sending 87768 ms...Success
 23:36:35.343 -> Sending 88774 ms...Failed
 23:36:36.327 -> Sending 89793 ms...Success
 23:36:37.358 -> Sending 90799 ms...Success
 23:36:38.343 -> Sending 91805 ms...Success
 23:36:39.374 -> Sending 92812 ms...Success
 23:36:40.358 -> Sending 93817 ms...Success
 23:36:41.390 -> Sending 94822 ms...Failed

dparson55, 2019-05-14:
I don't know what could be causing that but I'll try recreating the issue.  What hardware are you using on the TX and RX sides?  I assume an ATtiny85 on the TX side, but what are you using on the RX side?  Also, what speed are you running the ATtiny at?

I recently made a lot of updates on the ATtiny85 2pin sensor sketch and tested it with an ATtiny85 @ 8MHz on the TX side, communicating with an Arduino Uno (ATmega328 @ 16MHz), and during all the development and testing, didn't see any issues.

destiny2008git, 2019-05-14:
i am using attiny isp as unio R3 as Rx and pro mini as Tx since i can see the serial out.
unless there is way to see the serial out on attiny.

also uploading 1mhz on attiny85. i have already bricked 3 attiny 84 unrecoverable.

do you have a clear process on uploading @8mhz to attiny 85
tried a few online but they are not clear. using attinycore damelis now.

need to know when to upload and when to use bootloader :)

Sent from A Secure Location on Earth.

destiny2008git, 2019-05-14:
actually my intent is to use the attiny85 with 2 pin to control 3 led on 1 end and buttons on the other end

rf24 led remote scenario example

Sent from A Secure Location on Earth.

dparson55, 2019-05-14:
You can use the SoftwareSerial library with ATtiny's to get serial output.

It's possible to set ATtiny fuses and unbrick them without a dedicated programmer, [here's a guide](http://www.electronics-lab.com/recover-bricked-attiny-using-arduino-as-high-voltage-programmer/).

I use an AVR Dragon and made a [video](https://www.youtube.com/watch?v=ZuCBoI2YL1A) about how it works, which might be a help.

I'll try reproducing the issue and will report back, but will be surprised if there's a problem with NRFLite.

destiny2008git, 2019-05-14:
i tried this guide for unbricking the attiny84 but i see all fuses enabled by default using a programmer and it refuses to write , also all contents are 0000 signature:FFFF and fuses
OOOO.

Sent from A Secure Location on Earth.

dparson55, 2019-05-15:
I'm not able to reproduce the issue you're seeing, I transmitted successfully for 15 minutes without a failure.  Maybe the default channel (100) is noisey in your environment, so you can change that. 
 You can also try changing the bitrate to see if that helps.  Both of these are controlled via the **init** and **initTwoPin** methods as shown in **NRFLite.h**.  Also be sure to add an extra capacitor across VCC and GND for the radio modules.

Below is the exact hardware and code I used.: 
* TX - Arduino Uno, ATmega328 @ 16MHz using the **Basic_TX.ino** code from the examples.
* RX - ATtiny85 @ 1MHz and 8MHz using the 2-pin circuit and led for status, using the code below.

```
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;
const static uint8_t PIN_RADIO_MOMI = 4;
const static uint8_t PIN_RADIO_SCK = 1;

struct RadioPacket
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
    // Setup an led for status.
    pinMode(0, OUTPUT);
    digitalWrite(0, LOW);

    if (!_radio.initTwoPin(RADIO_ID, PIN_RADIO_MOMI, PIN_RADIO_SCK))
    {
        blink();
        while (1); // Wait here forever.
    }
}

void loop()
{
    while (_radio.hasData())
    {
        _radio.readData(&_radioData);
        blink();
    }
}

void blink()
{
    digitalWrite(0, HIGH);
    delay(100);
    digitalWrite(0, LOW);
    delay(100);
}
```

dparson55, 2019-05-19:
Haven't heard from you in a few days so closing the issue.

destiny2008git, 2019-05-21:
HI, sorry was away , i will test these 2 days and revert back just the same.

destiny2008git, 2019-05-21:
Hi dave,
so i tested the code but for some reason the LED connected is very dim.
now i dont think that has anything to do with the code?
below are my both codes tx and rx ....  both running from arduino uno R3 for the moment.
`#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 1;             // Our radio's id.
const static uint8_t DESTINATION_RADIO_ID = 0; // Id of the radio we will transmit to.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

const uint8_t buttonPin[] = {2, 3, 4};
uint8_t myArray[1];

bool buttonState[] = {0, 0, 0};       // current state of the button
bool lastButtonState[] = {0, 0, 0};   // previous state of the button

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
  uint8_t FromRadioId;
  uint32_t OnTimeMillis;
  uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
  Serial.begin(115200);

  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS, 50))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }

  _radioData.FromRadioId = RADIO_ID;
}

void loop()
{
  _radioData.OnTimeMillis = millis();

  for (uint8_t i = 0; i <= 2; i++)
  {
    buttonState[i] = digitalRead(buttonPin[i]);
  }

  for (uint8_t j = 0; j <= 2; j++)
  {
    if (buttonState[j] == HIGH)
    {
      myArray[0] = (j + 1);
      //radio.write(myArray, 1);
      _radio.send(DESTINATION_RADIO_ID, &myArray, sizeof(myArray));
    }
    if (buttonState[j] != lastButtonState[j])
    {
      if (buttonState[j] == LOW)
      {
        myArray[0] = 0;
        //radio.write(myArray, 1);
        _radio.send(DESTINATION_RADIO_ID, &myArray, sizeof(myArray));
      }
      delay(5);
    }
    lastButtonState[j] = buttonState[j];
  }

  //_radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData))) // Note how '&' must be placed in front of the variable name.

  /*
    By default, 'send' transmits data and waits for an acknowledgement.
    You can also send without requesting an acknowledgement.
    _radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData), NRFLite::NO_ACK)
    _radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData), NRFLite::REQUIRE_ACK) // THE DEFAULT
  */
}
`

`#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

const uint8_t LED[] = {5, 6, 7};
uint8_t myArray[1];

struct RadioPacket // Any packet up to 32 bytes can be sent.
{

};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
  Serial.begin(115200);

  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS, 50))
  {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }

  /*
    By default, 'init' configures the radio to use a 2MBPS bitrate on channel 100 (channels 0-125 are valid).
    Both the RX and TX radios must have the same bitrate and channel to communicate with each other.
    You can assign a different bitrate and channel as shown below.
    _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS, 0)
    _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE1MBPS, 75)
    _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE2MBPS, 100) // THE DEFAULT
  */
}

void loop()
{
  while (_radio.hasData())
  {
    _radio.readData(&myArray); // Note how '&' must be placed in front of the variable name.

    code();
    Serial.println("Code in Action..");

  }
}

void code()
{
  //Light up the correct LEDs
  for (uint8_t j = 1; j <= 3; j++)
  {
    if (myArray[0] == j) {
      digitalWrite(LED[j - 1], HIGH);
    }
    if (myArray[0] == 0) {
      digitalWrite(LED[j - 1], LOW);

    }
  }
}`

dparson55, 2019-05-22:
You opened this issue about the ATtiny85 and 2 pin operation, and neither your RX or TX code above is related to the ATtiny85.  I provided the ATtiny85 code and Arduino Uno code that I used for testing since I could not reproduce the issue you described.  If you can please try the code I provided we can continue to troubleshoot the issue you initially had.

This NRFLite issue list is not the place to ask about lighting led's but I looked through your code and on the RX side, you need set the **pinMode** of your led pins to **OUTPUT**.  All pins default to INPUT and when you set an input pin HIGH, it simply engages the pull-up resistor on the pin.

destiny2008git, 2019-05-22:
> 
> 
> You opened this issue about the ATtiny85 and 2 pin operation, and neither your RX or TX code above is related to the ATtiny85. I provided the ATtiny85 code and Arduino Uno code that I used for testing since I could not reproduce the issue you described. If you can please try the code I provided we can continue to troubleshoot the issue you initially had.
> 
> This NRFLite issue list is not the place to ask about lighting led's but I looked through your code and on the RX side, you need to set the **pinMode** of your led pins to **OUTPUT**. All pins default to INPUT and when you set an input pin HIGH, it simply engages the pull-up resistor on the pin.

you are right and i am sorry for this, it was a miss on my part i will have the test made on 2 pin shortly, hope all goes well ..

destiny2008git, 2019-05-23:
is it normal to get this message?

C:\Users\Arduino\libraries\NRFLite\src\NRFLite.cpp:299:51: note: containing loop

     for (uint8_t channelNumber = 0; channelNumber <= MAX_NRF_CHANNEL; channelNumber++)

destiny2008git, 2019-05-23:
C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp: In member function 'void NRFLite::printDetails()':

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:254:49: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("CONFIG", readRegister(CONFIG));

                                                 ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:255:47: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("EN_AA", readRegister(EN_AA));

                                               ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:256:55: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("EN_RXADDR", readRegister(EN_RXADDR));

                                                       ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:257:53: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("SETUP_AW", readRegister(SETUP_AW));

                                                     ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:258:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("SETUP_RETR", readRegister(SETUP_RETR));

                                                         ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:259:47: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("RF_CH", readRegister(RF_CH));

                                               ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:260:53: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("RF_SETUP", readRegister(RF_SETUP));

                                                     ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:261:53: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("STATUS", readRegister(STATUS_NRF));

                                                     ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:262:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("OBSERVE_TX", readRegister(OBSERVE_TX));

                                                         ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:263:53: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("RX_PW_P0", readRegister(RX_PW_P0));

                                                     ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:264:53: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("RX_PW_P1", readRegister(RX_PW_P1));

                                                     ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:265:59: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("FIFO_STATUS", readRegister(FIFO_STATUS));

                                                           ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:266:47: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("DYNPD", readRegister(DYNPD));

                                               ^

C:\Users\phalaby\Documents\Arduino\libraries\NRFLite\src\NRFLite.cpp:267:51: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

     printRegister("FEATURE", readRegister(FEATURE));

destiny2008git, 2019-05-24:
tried your code with uno basic_tx (no Ack) I get success , but attiny 85 2 pin with below code I get 
only initializing.. nothing more connections is as diagram- using latest library with compile messages
1-4 momi/sck
0 led blink
'#include <NRFLite.h>
#include <SoftwareSerial.h>

const static uint8_t RADIO_ID = 0;
const static uint8_t PIN_RADIO_MOMI = 4;
const static uint8_t PIN_RADIO_SCK = 1;

#define RX    2   // *** D3, Pin 2
#define TX    3   // *** D4, Pin 3

SoftwareSerial Seria(RX, TX);

struct RadioPacket
{
  uint8_t FromRadioId;
  uint32_t OnTimeMillis;
  uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
  Seria.begin(9600);
  Seria.println("Initializing...");
  // Setup an led for status.
  pinMode(0, OUTPUT);
  digitalWrite(0, LOW);
  // initialize digital pin LED_BUILTIN as an output.

  if (!_radio.initTwoPin(RADIO_ID, PIN_RADIO_MOMI, PIN_RADIO_SCK))
  {
    blink();
    while (1); // Wait here forever.
  }
}

void loop()
{
  while (_radio.hasData())
  {
    _radio.readData(&_radioData);
    blink();
    Seria.println("InCode...");
  }
}

void blink()
{
  digitalWrite(0, HIGH);
  delay(100);
  digitalWrite(0, LOW);
  delay(500);
}'

dparson55, 2019-05-24:
So it looks like you are not getting as far as you were before.  After you program the ATtiny, be sure to remove the connection between the programmer and ATtiny.  That's about all I can think of at this point.

destiny2008git, 2019-05-24:
so it works but intermittent, i tried different channels but it seems when i touch the nrf antenna or wave my hand closeby i see the blink work a bit then goes offline again.

i am using 250kbps/0

dparson55, 2019-05-24:
The last code you posted for the ATtiny85, where you added SoftwareSerial to the example I provided, has the bitrate set to the default which is 2mbps, so maybe that is the issue.  The bitrate on the TX and RX side must be the same.

dparson55, 2019-05-24:
Also worth checking each of the breadboard jumper wires you are using, maybe one of them is not making a good connection.

destiny2008git, 2019-05-24:
> 
> 
> The last code you posted for the ATtiny85, where you added SoftwareSerial to the example I provided, has the bitrate set to the default which is 2mbps, so maybe that is the issue. The bitrate on the TX and RX side must be the same.

latest code both side match - its only mis-bhaving on attiny
i even replaced 3 different nrf24 to rule out the transmitter

`#include <NRFLite.h>
#include <SoftwareSerial.h>

const static uint8_t RADIO_ID = 0;
const static uint8_t PIN_RADIO_MOMI = 4;
const static uint8_t PIN_RADIO_SCK = 1;

#define RX    2   // *** D3, Pin 2
#define TX    3   // *** D4, Pin 3

SoftwareSerial Seria(RX, TX);

struct RadioPacket
{
  uint8_t FromRadioId;
  uint32_t OnTimeMillis;
  uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
  Seria.begin(9600);
  Seria.println("Initializing...");
  // Setup an led for status.
  pinMode(0, OUTPUT);
  digitalWrite(0, LOW);
  // initialize digital pin LED_BUILTIN as an output.

  if (!_radio.initTwoPin(RADIO_ID, PIN_RADIO_MOMI, PIN_RADIO_SCK, NRFLite::BITRATE250KBPS, 0))
  {
    blink();
    while (1); // Wait here forever.
  }
}

void loop()
{
  while (_radio.hasData())
  {
    _radio.readData(&_radioData);
    blink();
    Seria.println("InCode...");
  }
}

void blink()
{
  digitalWrite(0, HIGH);
  delay(100);
  digitalWrite(0, LOW);
  delay(500);
}

`

destiny2008git, 2019-05-24:
OK.. so i tried the basic_TX with no Ack removed it works now , library issue ?

if (_radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData)))

22:22:34.663 -> Sending 83756 ms...Success
22:22:35.647 -> Sending 84758 ms...Success
22:22:36.678 -> Sending 85775 ms...Success
22:22:37.663 -> Sending 86777 ms...Success
22:22:38.694 -> Sending 87780 ms...Success
22:22:39.725 -> Sending 88812 ms...Success
22:22:40.710 -> Sending 89823 ms...Success
22:22:41.741 -> Sending 90849 ms...Success
22:22:42.772 -> Sending 91851 ms...Success
22:22:43.756 -> Sending 92855 ms...Success
22:22:44.741 -> Sending 93857 ms...Success
22:22:45.772 -> Sending 94860 ms...Success
22:22:46.756 -> Sending 95862 ms...Success
22:22:47.788 -> Sending 96869 ms...Success

are you releasing a new version for the compile errors?

dparson55, 2019-05-24:
When a REQUIRE_ACK (the default) send is performed, the radio will send the packet and quickly switch into RX mode, and wait for an acknowledgement from the receiver.  If it doesn't receive the acknowledgement, it will switch back into TX mode and try sending the packet again.  It will do this 16 times and only report a failure if none succeed.

When a NO_ACK send is performed, the radio sends it out blindly hoping a receiver will see it, and there is no guarantee the transmission will be successful.  The send attempt will only happen 1 time.  That single packet can be lost due to EMF, waving your hand in the right way, etc.  So NO_ACK sends are great for when you're streaming data while REQUIRE_ACK are best for when you need to guarantee packet delivery.

You can try the new [ChannelScanner](https://github.com/dparson55/NRFLite/tree/master/examples/ChannelScanner) example that shows which channels are the most noisy in your environment.  It can help identity the cleanest channel to use to help improve the success of transmissions.

I'm going to release the new version of the library now that this issue is closed.  It will include the fix for the compiler warnings that you discovered.

dparson55, 2019-05-24:
I don't think I can help you any more but you can always post questions in the official [Arduino Forums](https://forum.arduino.cc/).  Your constant set of changes, inconsistent messages, and lack of detail is making it too hard for me to understand what you are doing.  So good luck and I hope you are able to figure out the delay issue.

closed 2019-05-24 by dparson55

destiny2008git, 2019-05-25:
Dave, just a quick question would the 2 pin setup of the nrf cause a delay in transmittion to attiny I/O compared to using all 5 pins of the nrf on an arduino ?

dparson55, 2019-05-25:
2-pin hookups are slower than using the USI hardware of the ATtiny's.  See the ATtiny USI hookup guides in the readme to do some testing.  Also transferring at 250Kbps is theoretically 8x slower than transferring at 2Mbps.

destiny2008git, 2019-05-25:
> 
> 
> 2-pin hookups are slower than using the USI hardware of the ATtiny's. See the ATtiny USI hookup guides in the readme to do some testing. Also transferring at 250Kbps is theoretically 8x slower than transferring at 2Mbps.

tried all speeds , wasn't related but using other than 1 - 4 on tiny got the speed back!
thanks, all working now.