dparson55/NRFLite #28: How to check that the radio is in receive mode?

k64t34, 2019-05-19:
How to check that the radio is in receive mode?
Or force it to  receive mode?
After sending is complete, packets are no longer receiving.

dparson55, 2019-05-19:
NRFLite switches the operation mode of the radio module based on the methods you call. Whenever you call **init**, **initTwoPin**, **hasData**, or **hasDataISR**, NRFLite puts the radio into RX mode. Whenever you call **send** or **startSend**, it puts the radio into TX mode. So you can send a data packet with **send** and afterward call **hasData** to put the radio back into receiving mode.

If you have a situation where you just want the transmitter to be able to get data from a receiver, a better approach is to use _acknowledgement data packets_.  That's how the designers of the radio intended 2-way communication to be implemented and the hardware has special features to make this work well, and fast.  See the **IRQ_ACK_RX** and **IRQ_ACK_TX** examples on how to send data from a receiver to a transmitter with ACK data packets.

k64t34, 2019-05-19:
ACK work perfect.
But I no need to confirm every radio packet.
In addition to the useful data, I need to ensure the exchange of control packets between nodes.
For example, when an executor node does not receive heartbeat packet for more than 1 second, the executor must turn off the motor.
In other side, when the operator node does not receive heartbeat packet from the executive, an alarm should be triggered.

Is the decision to use the function prepForRx() necessary and sufficient to switch to RX mode?  

I propose to  debug a template for this mode, and to add to examples of the NRFLite library. Perhaps it will be interesting to some developers.

And one more quesion.
Is it good idea to use **_radio.readData(&data)** inside **radioInterrupt()** in **IRQ_RX mode** ?

dparson55, 2019-05-19:
Rather than changing the library to expose prepForRx, you could also call init to switch back into RX mode after you've waited for all outgoing packets to be sent.  hasData has logic to wait for prior transmissions to complete, init and prepForRx do not have this logic.

It's ok to do the data reading in radioInterrupt if you like, but the best practice is to set a flag that an interrupt has occurred and then read the data outside the interrupt handler.  See [IRQ_RX.ino](https://github.com/dparson55/NRFLite/blob/master/examples/IRQ_RX/IRQ_RX.ino) for an example.

dparson55, 2019-05-19:
Here's a thought on the example you mentioned:
* Executor node sends a heartbeat packet to the operator node every second.  If this packet is not acknowledged, the executor node turns off its motor.  The operator node stores the heartbeat received time.
* Operator node checks its heartbeat received times to see if it needs to raise an alarm.

k64t34, 2019-05-20:
Thanks!
I study examples from library. It like bible for novice.

Is the time spent on the execution of functions of init and pk different?
I see that the pk function has a minimum number of commands.

dparson55, 2019-05-21:
Yep, init will take a bit longer.  Good luck with your project.

closed 2019-05-21 by dparson55

dparson55, 2019-05-22:
@k64t34 Just a heads up, I've added a [startRx](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L56) method and also [examples](https://github.com/dparson55/NRFLite/tree/master/examples) showing two way communication.
* TwoWayCom_HardwareBased
* TwoWayCom_SoftwareBased

I haven't released a new version of the library because I'm going to add a channel scanner next, so once that's done all of these improvements will be in 2.2.0.