dparson55/NRFLite #29: nrf24l01+ and arduino uno twowayCommunicationSoft

Azsxzsxa, 2019-06-27:
I have this problem with nrf24l01+ and Arduino UNO. I've checked the wires many many times , tried other wires , tried other nrf24l01+s . But the problem is the same . It always sends a couple of successful messages and then it stops 
[image]

dparson55, 2019-06-27:
It looks like this is the serial output from [TwoWayCom_SoftwareBased_RX](https://github.com/dparson55/NRFLite/blob/master/examples/TwoWayCom_SoftwareBased_RX/TwoWayCom_SoftwareBased_RX.ino) and the Arduino running [TwoWayCom_SoftwareBased_TX](https://github.com/dparson55/NRFLite/blob/master/examples/TwoWayCom_SoftwareBased_TX/TwoWayCom_SoftwareBased_TX.ino) appears to have stopped transmitting the heartbeat.  Please confirm if this guess is correct and if not, let me know what code you are using in each Arduino.

My first idea is the 3.3V power going to the Arduinos is insufficient.  You can try adding a 10uF capacitor between the nRF24L01+ VCC and GND pins to solve this problem.

If that doesn't help, do this test and report the results:
1. Program Arduino 1 with [Basic_RX](https://github.com/dparson55/NRFLite/blob/master/examples/Basic_RX/Basic_RX.ino) and Arduino 2 with [Basic_TX](https://github.com/dparson55/NRFLite/blob/master/examples/Basic_TX/Basic_TX.ino) and confirm they communicate successfully.
2. Program Arduino 1 with [Basic_TX](https://github.com/dparson55/NRFLite/blob/master/examples/Basic_TX/Basic_TX.ino) and Arduino 2 with [Basic_RX](https://github.com/dparson55/NRFLite/blob/master/examples/Basic_RX/Basic_RX.ino) and confirm they communicate successfully.

Azsxzsxa, 2019-06-27:
Yes , it is the output from TwoWayCom_SoftwareBased RX and TX. I havent changed anything inside the code , its the bare library example .
 I have tried using the 10uF capacitor but nothing changed. 
 Regarding BasicRX and TX. It acts weird I would say because : Arduino1 Basic Rx , Arduino2 Basic TX works perfectly but
Arduino1 Basic TX , Arduino2 Basic RX doesnt work. It says "success" but on the other arduino nothing appears. 
 I have tried plenty of nrf24l01+ modules but there is no difference

dparson55, 2019-06-28:
Are you using nRF24L01 PN-LNA radio modules?  I've seen a situation where such a radio did not support auto-acks which is something built into the nRF24L01 chip itself.  The radio sent data but could never receive the ACK that the receiver would transmit back to confirm it received the data.  There was nothing I could do in the library to resolve this because it was something wrong with the hardware of the radio module.

One work-around is to do NO_ACK sends so that an ACK is never requested.  You can also try using the RadioHead library because it does not use the auto-ack feature of the nRF24L01 chip for its 2-way communication.

Azsxzsxa, 2019-06-28:
It is not the PN-LNA module. I will try without the ACK and see how it goes

Azsxzsxa, 2019-06-30:
It worked when I didn't use ACK . I have another problem now:
I have this radio module on a "car" , 2 motors with a L293D shield . The motors don't work if radio.init() is in the setup. Once I comment the radio.init() , the motors work 
`
#include <SPI.h>
#include <NRFLite.h>
#include <AFMotor.h>

AF_DCMotor MotorOne(1);
AF_DCMotor MotorTwo(2);

const static uint8_t RADIO_ID = 1;             // Our radio's id.
const static uint8_t DESTINATION_RADIO_ID = 0; // Id of the radio we will transmit to.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

bool vehicleArrivedCheck=false;
bool devicesAckCheck=false;
int readRadioMsg;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup() {
  Serial.begin(115200);
  MotorOne.setSpeed(200);
  MotorTwo.setSpeed(250);
  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
    
    _radioData.FromRadioId = RADIO_ID;
    
}
int ReadDistance(){
  int distanceValue = analogRead(A0);
  Serial.println(distanceValue);
  return distanceValue;
}

int ReadPercentage(){
  int sensorValue = analogRead(A3);
  float voltage = sensorValue * (5.0 / 1023.0)*2;
  Serial.print(voltage);
  Serial.print("   ");
  int percentage=(voltage - 6) * (100 - 1) / (11 - 6) + 1;
  Serial.print(percentage);
  Serial.println("%");
  return percentage;
  delay(50);
}

void SendRadioData(int dataToSend)
    {
     Serial.print("Sending ");
     Serial.print(dataToSend);
    
    if (_radio.send(DESTINATION_RADIO_ID, &dataToSend, sizeof(dataToSend),NRFLite::NO_ACK)) // Note how '&' must be placed in front of the variable name.
    {
        Serial.println("...Success");
    }
    else
    {
        Serial.println("...Failed");
    }
  delay(100);
}

String ReadRadioData()
    {
      String msg="0";
      while (_radio.hasData())
    {
        _radio.readData(&readRadioMsg); // Note how '&' must be placed in front of the variable name.
        msg = readRadioMsg;
        Serial.println(msg);
    }
    return msg;
    }

void ForwardDrive()
{
  MotorOne.run(FORWARD);
  MotorTwo.run(FORWARD);
}

void StopDrive(){
  MotorOne.run(RELEASE);
  MotorTwo.run(RELEASE);
}

void loop() {
  
 //ForwardDrive();
  
  
  if(!vehicleArrivedCheck && ReadDistance()<400){
    ForwardDrive();
  }else{
    vehicleArrivedCheck=true;
    StopDrive(); 
    SendRadioData(ReadPercentage());
   // if(ReadRadioData()!="0"){
 
    //} 
  }  
  
}`

dparson55, 2019-07-01:
The readme for the [AFMotor](https://github.com/adafruit/Adafruit-Motor-Shield-library) library mentioned it did not share the SPI bus very well, so I looked through its code and sure enough, it doesn't follow best practices for SPI bus sharing.  The readme mentions they changed their hardware and made a new library that uses the I2C bus rather than fixing their SPI code, so without major modifications to AFMotor, it's just not going to be possible to share the SPI bus with the radio.

One work around is to use the NRFLite 2-pin support where any two GPIO pins can be used to control the radio.

Azsxzsxa, 2019-07-01:
I've made it work by not calling the _radio.init() in setup, I'm calling it in the Loop() after I am done withing the motors and this worked . The problem now is that I would like to somehow "de-init" the radio so I can use the motors again after a while. Any suggestion for this. Also thank you so much for the research you are doing , truly a life-saver

Azsxzsxa, 2019-07-01:
I can't do the 2-pin because I have already glued everything

dparson55, 2019-07-01:
Sure thing.  Try calling setSpeed to re-init the motor library.

```
MotorOne.setSpeed(200);
MotorTwo.setSpeed(250);
```

If that doesn't work you can look at the init method in NRFLite to see if undoing the [pin modes it sets up](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L30) to try de-initing.

Azsxzsxa, 2019-07-01:
So from these
```
    pinMode(_cePin, OUTPUT);

    pinMode(_csnPin, OUTPUT);

    digitalWrite(_csnPin, HIGH);
```
I should make them INPUT , INPUT , LOW?

Azsxzsxa, 2019-07-01:
re-init-ing the Motor library didn't work so I want to try to undo the pin modes but im not sure into what states

dparson55, 2019-07-01:
Yep pretty much that, when the microcontroller first starts up its pins are INPUT and LOW.  So something like this:

```
pinMode(PIN_RADIO_CE, INPUT); digitalWrite(PIN_RADIO_CE, LOW);
pinMode(PIN_RADIO_CSN, INPUT); digitalWrite(PIN_RADIO_CSN, LOW);
```

dparson55, 2019-07-01:
One other thing, you should connect the nRF24L01's CE and CSN pins to pins that is not used by the motor shield.

dparson55, 2019-07-09:
Any updates, should I just close the issue?

closed 2019-07-13 by dparson55