dparson55/NRFLite #30: radio.send() returns success but slave device receives no message

adamzxtan, 2019-08-20:
I'm trying to build a sensor network project that consist of 1 master and 6 slave devices. I would need to do two-way communication in between the slave and master, reliability of the two way communication is the priority for me. I've based my code on the software-based two way communication example, because I require the flexibility of letting the slave device to initiate transmission after receiving from the master. 

However, I ran into the problem of sometimes the master will indicate success in sending to slave device, but the targeted slave device doesn't receive anything. I'm not quite sure what's causing this, I've even tried the radio.addAckData() method, but its just pure disaster for me.

Anyway, this is the code base that I'm using for master device:
```
#include <NRFLite.h>
#include <SPI.h>

int ledPin[] = {9, 10, 4, A0, 5, A1};
bool isReturnMessageReceived = false;

const static uint8_t RADIO_ID = 0;
const static uint8_t DESTINATION_RADIO_ID[] = {1, 2, 3, 4, 5, 6};
const static uint8_t PIN_RADIO_CE = 7;
const static uint8_t PIN_RADIO_CSN = 8;

enum RadioPacketType {MODE_1, MODE_2};

struct RadioPacket {
  RadioPacketType PacketType;
  uint8_t FromRadioID;
};

NRFLite radio;

void setup() {
  Serial.begin(115200);

  for (int i = 0; i < 6; i++) {
    pinMode(ledPin[i], OUTPUT);
  }

  if (!radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, 100)) {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }

  randomSeed(analogRead(A6));
}

void loop() {
  byte n = random(6);   //generate a random number between 1-6
  Serial.print("n = ");
  Serial.println(n);
  RadioPacket radioData;
  radioData.FromRadioID = RADIO_ID; //to let reeiver verify the message comes from master
  radioData.PacketType = MODE_1;    //let the receiver know we want to enter MODE 1

  if (radio.send(DESTINATION_RADIO_ID[n], &radioData, sizeof(radioData))) {
    Serial.println("send success!");
    digitalWrite(ledPin[n], HIGH);

    Serial.print("Awaiting reply from slave node: ");
    Serial.println(n + 1);
    while (isReturnMessageReceived != true) {
      if (radio.hasData()) {
        RadioPacket radioData;
        radio.readData(&radioData);

        if (radioData.FromRadioID == DESTINATION_RADIO_ID[n]) {
          isReturnMessageReceived = true;
          Serial.print("Received message from slave node: ");
          Serial.println(radioData.FromRadioID);
          Serial.println("");
        }
      }
    }
    isReturnMessageReceived = false;
    digitalWrite(ledPin[n], LOW);
    delay(1000);
  }
  else {
    Serial.println("send fail...");
    Serial.println("");
  }
}
```
Code base for the slave devices (Radio_ID: 1 to 6):
```
#include <NRFLite.h>
#include <SPI.h>

int led = 10, IR_PIN = 2;
bool isReturnMessageReceived = false;

const static uint8_t MASTER_RADIO_ID = 0;    //MASTER DEVICE
const static uint8_t PIN_RADIO_CE = 7;
const static uint8_t PIN_RADIO_CSN = 8;

//change this when uploading to different slave devices, RADIO_ID: from 1-6
const static uint8_t RADIO_ID = 6;

enum RadioPacketType {MODE_1, MODE_2};

struct RadioPacket {
  RadioPacketType PacketType;
  uint8_t FromRadioID;
};

NRFLite radio;

void setup() {
  Serial.begin(115200);

  pinMode(led, OUTPUT);
  pinMode(IR_PIN, INPUT);

  if (!radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, 100)) {
    Serial.println("Cannot communicate with radio");
    while (1); // Wait here forever.
  }

  //blink led to signal user that setup is ready
  digitalWrite(led, HIGH);
  delay(1000);
  digitalWrite(led, LOW);
  delay(1000);
}

void loop() {
  while (radio.hasData()) {
    Serial.println("Received message!");
    RadioPacket radioData;
    radio.readData(&radioData);

    if (radioData.FromRadioID == MASTER_RADIO_ID) {
      Serial.println("Identified packet from master");
      switch (radioData.PacketType) {
        case MODE_1: handle_MODE_1(); break;
        case MODE_2: handle_MODE_2(); break;
      }
    }
    else {
      Serial.println("Unknown sender, message ignored...");
      Serial.print("Signal source: ");
      Serial.print(radioData.FromRadioID);
      Serial.println("");
    }
  }
}

/************************************************************************************/
void handle_MODE_1() {
  digitalWrite(led, HIGH);
  while (digitalRead(IR_PIN) != LOW) {}   //stay in this loop until infrared beam is broken

  digitalWrite(led, LOW);
  RadioPacket radioData;
  radioData.FromRadioID = RADIO_ID;   //to let master know the message comes from the right source

  if (radio.send(MASTER_RADIO_ID, &radioData, sizeof(radioData))) {
    Serial.println("send success!");
  }
  else {
    Serial.println("send fail...");
  }
}

void handle_MODE_2() {
  Serial.println("Do something else here...");
}
```

Anyone has any idea on what's causing the issue?

dparson55, 2019-08-21:
The logic looks ok so it makes me wonder, if you simply run the software-based TX and RX examples, does it work ok?  This would help to rule out a hardware issue.

Also, could you setup a master and then one slave using your code, and then provide the serial output from both?  Unfortunately I'm on a trip and won't be back for a couple of weeks, so I can't troubleshoot this directly.

I can confirm your experience that you should not use the addAckData method when using the software-based two way communication, it will just confuse things and make the logic even harder.

adamzxtan, 2019-08-21:
Hi @dparson55. Yes, I've tried and modified the software-based TX, RX example for my application, it works just fine. The master and slave devices are communicating actively. Here's part of the serial output from the modified version of the software-based TX:
```
....
Received data from 4, 44000 ms
Sending heartbeat...Success
Received data from 5, 44000 ms
Received data from 1, 48000 ms
Sending heartbeat...Success
Received data from 3, 44000 ms
Received data from 6, 52000 ms
Received data from 2, 48000 ms
Sending heartbeat...Success
Sending heartbeat...Success
Received data from 4, 48000 ms
Sending heartbeat...Success
Received data from 5, 48000 ms
Received data from 1, 52000 ms
Sending heartbeat...Success
Received data from 3, 48000 ms
Received data from 6, 56000 ms
Received data from 2, 52000 ms
Sending heartbeat...Success
Sending heartbeat...Success
Received data from 4, 52000 ms
Sending heartbeat...Success
Received data from 5, 52000 ms
Received data from 1, 56000 ms
....
```

I've tried with you suggestion with only 1 active slave on my code, here's the serial output from the master:
```
n = 0
send fail...
n = 0
send fail...
n = 1
send success!
Awaiting reply from slave node: 2
Received message from slave node: 2
n = 0
send fail...
n = 0
send fail...
n = 1
send success!
Awaiting reply from slave node: 2  ---->>> at this point the led light on master is switched on
```
And the output on the slave device just shows:
```
Received message!
Identified packet from master
send success! 
```

I've been testing my code since last night to see the behavior of it. I found that there's one time the slave will actually have fail send back to master, although it was receiving fine before. So, its just really weird. 

But considering that the modified version of the software-based TX, RX works fine, could it be that there can't be any long delay in the code? For example, in my handle_MODE_1() function, its waiting for the IR sensor signal before it starts sending back a signal to master again.

Not sure if that caused the code locking up issue, just a thought.

dparson55, 2019-08-22:
I would keep adding more serial output to help with the debugging.  For instance, you could update the while loop that locks the slave as follows:

```
while (digitalRead(IR_PIN) != LOW)
{
    Serial.print("Waiting for IR signal ");
    Serial.println(millis());
    delay(500);
}
```

This while loop shouldn't be the issue.  But just keep adding more output so you can study the serial monitors of all the radios and pinpoint the problem.

Also try adding a 1uF, or higher, capacitor to the VCC and GND pins of the radio.

adamzxtan, 2019-08-22:
Hmm, ok. A 10uF cap is already attached to VCC and GND pin of the radio when I'm doing the test, the VCC has a dedicated 3.3V supply from LM1117 regulator, so it shouldn't be the power issue. 

Anyway, I won't be able to do anything for the next few days as I'll be travelling. Will continue to debug the setup and report back when I got home. Maybe the radio just need a few days of solitary to "behave" properly, who knows?

dparson55, 2019-08-23:
Ok that's good news on the power front, no need to add any additional capacitors.  I'll await your next reply.

adamzxtan, 2019-08-30:
Hey @dparson55, just got back today from my trip. Just wanna ask you, in the NRFLite.h, I see the radio.send() function says:

>  // send = Sends a data packet and waits for success or failure.  The default REQUIRE_ACK sendType causes the radio to attempt sending the packet up to 16 times.  If successful a 1 is returned.  Optionally the NO_ACK sendType can be used to transmit the packet a single time without any acknowledgement.

So, if the master returns true under this condition, it means the host has received ACK packet from the slave node, right? 
```
....
if (radio.send(DESTINATION_RADIO_ID[n], &radioData, sizeof(radioData))) {
    Serial.println("send success!");
    digitalWrite(ledPin[n], HIGH);
....
```
But from the testing I'm running with 1 master, 6 slave nodes, it looks like the radio.send() is returning true when it successfully sends out the message, not because it receives ack from the target node. 

Also, is there any reason why radio.send() is declared as uint8_t instead of bool?

dparson55, 2019-08-31:
If you provide power to the master and do not provide power to any slaves, see if **send** is still indicating success.  If it does then my next guess would be there is a problem with the nRF24L01 module, perhaps it is a fake or just broken in some way.  Behind the scenes, this is what should be happening:  the nRF24L01 chip sends the data to another nRF24L01 chip that has the radio id assigned to it.  Right after it transmits the data, it temporarily switches into RX mode and waits.  If all goes well the nRF24L01 that is listening for data will receive the packet and temporarily switch into TX mode to send a single acknowledgement packet back to the radio id who sent the data.  If the master never receives the ack packet it will switch back into TX mode and try again, up to 15 more times.  The nRF24L01 has all this logic built into it and it's called auto-acknowledgement.  Since it's built into the hardware it's very fast and doesn't require any software-based acknowledgement logic.

I've chosen to use uint8_t since it is the native data type used by the AVR hardware, bool is a higher level abstraction of what is happening at a lower level.  Behind the scenes a false is a byte containing eight zero's while a true is a byte with one or more bits that have a 1.  When I was learning to interact with the nRF24L01 chip and having to determine if something occurred, I was having to look for a 1 in certain bit positions of the nRF24L01's registers, so thinking in terms of bits making up bytes was necessary and I decided to keep everything consistent with native data types rather than higher level abstractions.

adamzxtan, 2019-09-21:
Hi, so sorry for lack of update. Been quite busy this recently. So I've tried out your suggestion, by only switching on the master, without any active slave nodes. The **send** result in fail, which is kind of expected. When I switch back on all the slave node, the sending indicates success but after a few send and receive I end up with the same situation again which I've described in the beginning. 

I really have no idea what's causing this, so I've switched to using another library, which is the RF24Mesh by TMRh20. That library works great for me, although now I'm having issue with the interference from environment that causes my ack from slave node will not be received by master node sometimes. 

Anyhow, thanks for your help along the way. Really appreciate it!

closed 2019-09-21 by adamzxtan