dparson55/NRFLite #32: How to determine if the receiver lost connection with radio station/s

clabnet, 2019-09-27:
This is not an Issue. The library working fine for my use.

I have a TX and RX running fine with this library, based on TwoWayCom_HardwareBased samples.
I would know how to determine when the receiver not receive yet any data AFTER a regular  transmission from the station/s.

This is my use case. 
The RX receive regular heartbeat from the TX, and the ACK is received correctly from TX.
After a few of time, the TX has a power shutdown.
After 10 seconds from the shutdown, I would like to signal on RX that the receiver no longer receives anything.  How to ?

Thank's

dparson55, 2019-09-27:
Hi @clabnet, maybe one of these ideas would work for your situation:

Idea 1:  On the RX side, save the time, in milliseconds, of the last received transmission and check it every 10 seconds to see if too much time has passed since the last transmission.

Idea 2:  on the TX side, detect when the primary power source has been lost and install a temporary power source, like a large capacitor, that gives the TX radio enough power to send a transmission to the RX radio that the TX radio is about to shut down.

dparson55, 2020-01-14:
Closing since there's been no response.

closed 2020-01-14 by dparson55