dparson55/NRFLite #36: ATTiny85 Sensor with 2 Pin

AlshainAvior, 2020-03-09:
Hi Dave,

This library (ATTiny85 with 2 Pin) is really help me a lot since tiny85 have limited IO, but is it possible to modify the library so the TX Mode can run in a two way communication (TX and RX using two pin for Attiny 85) ?

Thank you in advance

dparson55, 2020-03-10:
The 2-pin operation supports two way communication, you can use the TwoWayCom_Software or TwoWayCom_Hardware examples as a starting point and just tweak them slightly to use the 2-pin init and rather than using Serial.print to see status, instead blink an led (for example).

AlshainAvior, 2020-03-11:
Hi Dave,
It is run properly for the 2-Pin communication betwen Tiny85 and NRF, but for two way communication (**Parsing/Sending String from TX <--> RX**) still not work, currently I usen two unit arduino (uno) with NRF in each side and run deploy your Two way communication SW Based (TX and RX) it running properly, but once i tried to send the string it's not work, you can find the code in below 

[https://github.com/AlshainAvior/NRFLITE-TEST/blob/master/Arduino_SWBased_TX/Arduino_SWBased_TX.ino](url)

and

[https://github.com/AlshainAvior/NRFLITE-TEST/blob/master/Arduino_SWBased_RX/Arduino_SWBased_RX.ino](url)

Please help to take a look the code, I also mention about the packet data type (enum).
Thank you in advance

dparson55, 2020-03-12:
I forked your example and got it working.  I created a [pull request](https://github.com/AlshainAvior/NRFLITE-TEST/pull/1) with the changes and put notes about the necessary changes.  Let me know if you have any questions.

[image]

AlshainAvior, 2020-03-12:
Hi Dave,

Big Thanks to you, it's work also in 1mbps, really appreciate your response and amazing library.

again, Thank you

closed 2020-03-12 by AlshainAvior

dparson55, 2020-03-12:
Good to hear it's working, best wishes!