dparson55/NRFLite #37: In Basic_RX, hasData() returns 0 always

yvesgrignon, 2020-04-04:
Hi Dave
Before using the nRF24 and NRFLite in a real project, I tested it by connecting 2 radios to 2 Arduino Nano 328.  I had a little hardware problem which I fixed.  I then ran the scanner program from the examples and it worked fine on both setups and showed me that channel 100 would be fine.
I then ran Basic_TX on one and Basic_RX on the other.  Basic_TX was straight from github and produced:
Sending 104 ms...Failed
Sending 1170 ms...Failed
Sending 2232 ms...Failed
Sending 3293 ms...Failed
Sending 4414 ms...Failed
At first I assumed that the reason was that the receiver was not acknowledging because I had not yet started it.  I first ran a unmodified version of Basic_RX which did not produce any output and did not change the output from the receiver.  I then added a few Serial.print to see where the problem was.  The modifications were:
    ret = _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN);
    Serial.print("Init returned: ");
    Serial.println(ret);
    if (!ret)
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
and
void loop()
{
  Serial.println("Inside loop");
    if ((ret=_radio.hasData())>0)
    {
      Serial.print("Radio has data, length: ");
      Serial.println(ret);
      _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.
When I ran the modified Basic_RX, I got:
Init returned: 1
Inside loop
Inside loop
Inside loop
Somehow radio.hasData() always returns 0.  Anything obvious to you that I have missed?
Yves

dparson55, 2020-04-05:
Hi Yves,

Since you're working with 2 Arduino Nanos I would not expect any strange software issue, so I would keep focusing on the hardware.  Make sure the 3.3V you're supplying to the radio is good by adding a 10uF capacitor between it's VCC and GND pins.  Use a multimeter to check each of your connections to the radio modules on both sides:  make sure you don't have a bad jumper wire, a bad breadboard connection, and you don't have a connection to the wrong pin.

I know that some of the high powered nRF24L01+PA+LNA radio modules do not support the automatic acknowledgement feature of the radio, but the symptom is the TX side always fails while the RX side successfully receives the data.  So this doesn't match what you are seeing.

dparson55, 2020-04-05:
One other concern.  It's hard for me to understand what this does just by looking at it.
```
if ((ret=_radio.hasData())>0)
```

You can rewrite this line to be more clear like this.
```
uint8_t receivedDataLength = _radio.hasData();
if (receivedDataLength > 0) 
```

yvesgrignon, 2020-04-05:
Thanks a  lot for such a quick answer.  I was not expecting it and it was a very pleasant surprise when I read my emails this morning.
And your suggestion did the trick!  Thanks even more for that.  The 10 uF capacitor did the trick even though it was far from the receiving nRF24.  I did not get to install a capacitor on the transmitter and the system worked flawlessly for an hour.  Final design will have the capacitor on both of course.

closed 2020-04-05 by yvesgrignon

dparson55, 2020-04-05:
Great to hear, best of luck with the project!