dparson55/NRFLite #44: Tr. is not receiving acknowledgement. Interrupt stopped working in Transmitter side.

Dream-Project, 2020-06-09:
Transmitter is not receiving acknowledgement. Interrupt stopped working in Transmitter side. 

I tried IRQ sample. Receiver is receiving data but the acknowledge is not received by Transmitter. This issue started after taking the release on 8th June 2020

dparson55, 2020-06-09:
Try adding a 0.1uF capacitor between Vcc and Gnd on both your TX and RX radios.  Also, there are some nRF24L01+ PA/LNA high-powered radios that do not work correctly with acknowledgement packets.  Someone actually sent me such a radio and sure enough, it did not work with the hardware-based acknowledgement system that these radios support.

I confirmed all is still working fine with two ATmega328's running the IRQ_TX and IRQ_RX examples.  The transmitter resports success and if I unplug the receiver, the transmitter begins reporting a failure as expected.  If I plug in the receiver again, the transmitter starts reporting success.

The release a couple days ago only included a new method called `discardData` for #39, so it did not affect any of the sending or receiving logic in NRFLite.  If you did another library update or other hardware change, look at those changes instead.

Note the `hasAckData` method is only used when sending and receiving acknowledgement **data** packets, not plain acknowledgement packets that contain no data.  In order for a transmitter to get data from a receiver, the receiver needs to add an acknowledgement data packet to itself via `addAckData`.  See [TwoWayCom_HardwareBased_RX](https://github.com/dparson55/NRFLite/blob/master/examples/TwoWayCom_HardwareBased_RX/TwoWayCom_HardwareBased_RX.ino) for an example.
* RX side:  call `addAckData` to add an acknowledgement data packet
* TX side:  call `sendData` and then check `hasAckData` to see if the receiver provided an acknowledgement data packet

Dream-Project, 2020-06-17:
I was using 10uf.   Do you recommend to change to 0.1uf?

dparson55, 2020-06-17:
10uF is fine as well.  Since you already have a capacitor on there, hopefully it's not a power problem.

Dream-Project, 2020-06-18:
i am trying to build a RC car. For receiving part, instead of loop, i want to use interrupt. is it good approach? does it consume less current than looping?

dparson55, 2020-06-18:
Using the interrupt will not help with battery life unless you also sleep the microcontroller, which I'm sure you would not want to do.  Still though, using the interrupt should work perfectly well, you just need to make sure to use the interrupt examples and slowly make changes to not break anything.  I've found interrupt programming to be harder to avoid bugs, but that is just me.  Were you able to get the interrupt examples working?

dparson55, 2020-06-23:
Closing the issue, sounds like you're figuring things out.

closed 2020-06-23 by dparson55