dparson55/NRFLite #47: [Question] Send data without triggering IRQ

bart, 2020-07-01:
Hey @dparson55, I couldn't figure it out so I decided to ask you here: Is it possible to send data without triggering an interrupt? I have an NRF24 module that sends data continuously and listens on IRQ to stop. In the interrupt function I can't use whatHappened() method to check reason but it's triggered after every startSend() method call. So my idea is to find a way sending data without triggering the interrupt. Is that possible? IRQ should only get triggered when it received data.

Thanks a lot!

dparson55, 2020-07-01:
You sure can but NRFLite doesn't have a native way of letting you change it programatically.  The radio's CONFIG register (see page 57 of the [nRF24L01+ datasheet](https://infocenter.nordicsemi.com/pdf/nRF24L01P_PS_v1.0.pdf)) lets you configure which interrupts are enabled.  If you change [this line](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L72) in NRFLite.h as shown below, it will disable the TX_DS interrupt that occurs when a transmission is successful, and disable the MAX_RT interrupt that occurs when a transmission fails.  See if this works ok.

Old
`const static uint8_t CONFIG_REG_SETTINGS_FOR_RX_MODE = _BV(PWR_UP) | _BV(PRIM_RX) | _BV(EN_CRC);`

New
`const static uint8_t CONFIG_REG_SETTINGS_FOR_RX_MODE = _BV(PWR_UP) | _BV(PRIM_RX) | _BV(EN_CRC) | _BV(MASK_TX_DS) | _BV(MASK_MAX_RT);`

bart, 2020-07-01:
Thanks a lot for your answer! I think I won't change config but try to find a way to use whatHappened() method instead. I'm having radio processing in a class (h / cpp file) and attaching the interrupt here means a static interrupt method. Struggling with that that in combination with whatHappened() method but will hopefully solve it. Currently having a different problem. When using TX/RX mode on both receivers my Arduino Nano seems to restart over and over again as soon as it receives data but might be a code problem and not a library issue. Will keep you updated 👍

bart, 2020-07-02:
All right I found a solution. Reason for my two issues was different type size of data transferred from ESP32 to Arduino. Hint: char is not char. Setting volatile variable on interrupt in my radio class now and handling it in the loop function of my main class so that I don't have any problem with a non-static object I try to call a method on in from a static (interrupt) method. Again thanks @dparson55 for this great library with a bunch of amazing examples!

closed 2020-07-02 by bart

dparson55, 2020-07-02:
Awesome, good to hear!