dparson55/NRFLite #51: Packets printing weirdly

RCarran, 2020-08-15:
Hey Dave, I have been trying to transmit GPSRMC sentences from one Arduino to another. Since the sentence is about ~71 bytes in length I divided the string into three different char arrays and each char array is in its own struct to account for the 32 byte packet limit. This results in two 32 byte packets and one 14 byte packet. The transmitting Arduino says the data was sent successfully, but when I read and only print the first packet, it will print packets 2 and 3 aswell in addition to a duplicate packet 3 that is newlined. If I attempt to read and print all three packets, the sentence is split up and newlined in a very messy manner. I'm not sure what's going on. Could it be that packets 1 and 2 are the same size but not packet 3? Does that even matter?

This is how I want it to print:
[image]

This is how it prints if I only read and print packet 1 (weird how it prints data from packet 2 and 3):
[image]

And this is how it prints if try to print all three packets:
[image]

Any suggestions or advice? Maybe what I'm trying to do just isn't possible?

dparson55, 2020-08-15:
A couple of initial thoughts that I had that might be causing the challenge:
1. Strings in C are terminated with a null terminator character and I found it easiest to simply account for this and limit the length of strings to be 1 character shorter than the available space in the data packet, e.g. https://github.com/dparson55/NRFLite/blob/master/examples/Sensor_TX_ATtiny85_2Pin/Sensor_TX_ATtiny85_2Pin.ino#L76
2. The hardware of the radio modules includes a buffer that stores up to 3 packets 

```

RCarran, 2020-08-16:
Okay, I understood your first point and made sure to leave space for the null terminator. I also tried printing the data the same way you showed me with the second link, though that didn't help either. Your idea of structuring the data packets in your example left me a little confused. I think I understand that "SentencePart" numbers which part of the GPRMC sentence "Data[30]" is from but I'm not sure how to use it to validate the packet, nor do I understand what exactly SentenceId is for. One thing I noticed is that if I only send the first packet, I am able to print out that packet flawlessly. If I then change the code to send the first two packets, and still ONLY print the first packet, the second packet still prints out in a new line. I'm not sure what this behavior means but maybe I'm messing up in my transmitting instead of receiving? I just don't know how as I was using the Basic_TX.ino as my reference(the library examples work flawlessly btw).

dparson55, 2020-08-16:
The idea of the SentenceId is something that seems like it would be helpful for the logic on the receiver.  Imagine the receiver receives SentenceParts 2, 1, 2, 0, 1.  It could look at the SentenceId for each packet to determine which 3 it should join together to form the complete sentence.  Without the SentenceId the receiver would have to make assumptions about the order in which it receives packets or assume only 1 transmitter is sending packets.  Maybe this is ok for your situation though and I was just throwing a suggestion out to hopefully spark an idea for the situation you are facing.

When you use `Serial.print` it will not append a new line character onto the end of the object that you pass into it, like an Arduino String object or a char array.  But if your Arduino String object or char array already contains a new line character, the Serial Monitor will respect it and start placing the characters after the new line character onto a new line in the window.  So maybe the packets you are receiving contain the ASCII character code for a new line character embedded within them.

Splitting data that won't fit into a single packet is an interesting challenge and I started work on an example.  I have most of the transmitter code done and hope to finish the receiver code tomorrow.  To keep the example as simple as possible, I think it will assume there is only 1 transmitter.  Below is what I have so far...

```
#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 1;
const static uint8_t DESTINATION_RADIO_ID = 0;
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct GpsSentence
{
    uint8_t Id;
    String Data; // Contains a string too long to fit into a single packet.
};

const static uint8_t MAX_GPS_PACKET_DATA_LENGTH = 30;

struct GpsPacket
{
    uint8_t SentenceId;
    uint8_t PartId;                        // We will assume the sentence will be broken up into 3 parts:  0, 1, and 2.
    char Data[MAX_GPS_PACKET_DATA_LENGTH]; // 30 since the radio's max packet size = 32.
};

NRFLite _radio;
uint32_t _lastSendTime, _sentenceId;

GpsSentence getGpsSentence();              // Need to pre-declare these functions since they use structs.
void sendGpsSentence(GpsSentence sentence);
bool transmitPacket(GpsPacket packet);

void setup()
{
    Serial.begin(115200);

    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
}

void loop()
{
    // Send data once per second.
    if (millis() - _lastSendTime > 999)
    {
        _lastSendTime = millis();

        GpsSentence sentence = getGpsSentence();
        sendGpsSentence(sentence);
    }
}

GpsSentence getGpsSentence()
{
    GpsSentence sentence;

    _sentenceId++;
    sentence.Id = _sentenceId;
    sentence.Data = "$GPGGA,123456.00,789.00,S,12345.00,E,6,78,9.00,12356.00,M,7.00,M,*8F," + String(millis());

    return sentence;
}

void sendGpsSentence(GpsSentence sentence)
{
    Serial.println("Sending Sentence " + String(sentence.Id));
    Serial.println("  " + sentence.Data);

    GpsPacket packet;
    packet.SentenceId = sentence.Id;
    packet.PartId = 0;
    uint8_t packetDataIndex = 0;
    uint8_t sentenceLengthWithNullTerminator = sentence.Data.length() + 1;

    for (uint8_t i = 0; i < sentenceLengthWithNullTerminator; i++)
    {
        // Copy character from the sentence into the packet.
        packet.Data[packetDataIndex] = sentence.Data[i];

        // Determine if the packet is ready for sending.
        bool notFirstTimeCopyingData = i > 0;
        bool noMoreDataCanBeAddedToPacket = i % MAX_GPS_PACKET_DATA_LENGTH == 0;
        bool sendPacket = notFirstTimeCopyingData && noMoreDataCanBeAddedToPacket;

        if (sendPacket)
        {
            bool transmissionSucceeded = transmitPacket(packet);

            if (transmissionSucceeded)
            {
                // Prepare to build the next packet.
                packet.PartId++;
                packetDataIndex = 0;
            }
            else
            {
                Serial.println("  Failed to send sentence");
                break; // Abort transmission of the sentence since we failed to send this packet.
                       // The receiver will not be directly notified that we failed to send the sentence,
                       // but it will only assume a sentence is complete when it receives a packet with
                       // a PartId = 2.
            }
        }
        else
        {
            // We are still be building the current packet so increment the location where the next
            // character will be stored within the packet.
            packetDataIndex++;
        }        
    }

    // Send the last packet that we were building.
    transmitPacket(packet);
}

bool transmitPacket(GpsPacket packet)
{
    Serial.print("  Sending Part " + String(packet.PartId) + "...");

    if (_radio.send(DESTINATION_RADIO_ID, &packet, sizeof(packet)))
    {
        Serial.println("Success");
        Serial.println("  " + String(packet.Data));
        return true;
    }
    else
    {
        Serial.println("Failure");
        return false;
    }
}
```

RCarran, 2020-08-16:
Yes, I'm working off of one transmitter and wow your approach to splitting up the RMC sentence is very different to mine! This is how I originally constructed my TX code:
```#include <SPI.h>
#include <NRFLite.h>
#include <Adafruit_GPS.h>
#include <Chrono.h>

Adafruit_GPS GPS(&Serial1);

const static uint8_t RADIO_ID = 1;             // Our radio's id.
const static uint8_t DESTINATION_RADIO_ID = 0; // Id of the radio we will transmit to.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;
char *GPSRMC;
int sizee;

Chrono timer;

struct RadioPacket { // Any packet up to 32 bytes can be sent.
    uint8_t FromRadioId;
};
struct GpsPacket1 {
    char GPSRMC1[32];
};
struct GpsPacket2 {
    char GPSRMC2[32];
};
struct GpsPacket3 {
    char GPSRMC3[16];
};

NRFLite _radio;
RadioPacket _radioData; 
GpsPacket1 _radioData1; 
GpsPacket2 _radioData2; 
GpsPacket3 _radioData3;
//====================================================================================================
void setup(){
    Serial.begin(115200);
    _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE1MBPS, 125);

    
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN)){
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
    _radioData.FromRadioId = RADIO_ID;
    GPS.begin(9600);
    GPS.sendCommand(PMTK_SET_NMEA_OUTPUT_RMCONLY);
    GPS.sendCommand(PMTK_SET_NMEA_UPDATE_10HZ); 
}
//=====================================================================================================
void loop(){
  gpsInfo();
  if (timer.hasPassed(100)) {
    //if (_radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData))} {
        _radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData));
        _radio.send(DESTINATION_RADIO_ID, &_radioData1, sizeof(_radioData1));
        _radio.send(DESTINATION_RADIO_ID, &_radioData2, sizeof(_radioData2));
        _radio.send(DESTINATION_RADIO_ID, &_radioData3, sizeof(_radioData3));
        //Serial.println("...Success");
    //}
    //else{
        //Serial.println("...Failed");
    //}
    timer.restart(0);
  }  
}
//======================================================================================================
void gpsInfo() {
  char c = GPS.read(); // read data from the GPS in the 'main loop'
  if (GPS.newNMEAreceived()) {
    GPSRMC = GPS.lastNMEA();
    //Serial.print(GPSRMC);
    RMCsplit();
    if (!GPS.parse(GPSRMC)) 
      return; 
  }
}
//======================================================================================================
void RMCsplit () {
  sizee = strlen(GPSRMC);
  if (sizee >= 60) {
    int i = 0; int j = 0; int k = 0;
    while (i < 29 && j+29 < 59) {
      _radioData1.GPSRMC1[i] = GPSRMC[i]; _radioData2.GPSRMC2[j] = GPSRMC[j+29]; 
      i++; j++;
    }
    while (k+58 < sizee) {
      _radioData3.GPSRMC3[k] = GPSRMC[k+58];
      k++;
    }
    _radioData1.GPSRMC1[30] = '\0';
    _radioData2.GPSRMC2[30] = '\0';
    _radioData3.GPSRMC3[14] = '\0';
    //Serial.print(_radioData.FromRadioId);
    Serial.print(_radioData1.GPSRMC1);
    Serial.print(_radioData2.GPSRMC2);
    Serial.print(_radioData3.GPSRMC3);
  } else {
     int i = 0; int j = 0;
     while (i < 29) {
        _radioData1.GPSRMC1[i] = GPSRMC[i];
        i++; 
      }
      while (j+29 < sizee) {
        _radioData2.GPSRMC2[j] = GPSRMC[j+29];
        j++;
      }
  _radioData1.GPSRMC1[30] = '\0';
  _radioData2.GPSRMC2[30] = '\0';
  Serial.print(_radioData1.GPSRMC1);
  Serial.print(_radioData2.GPSRMC2);
  }
}
```
And this is how I constructed the RX code (where only printing the first array still prints the other arrays):
```
#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket { // Any packet up to 32 bytes can be sent.
    uint8_t FromRadioId;
};
struct GpsPacket1 {
    char GPSRMC1[32];
};
struct GpsPacket2 {
    char GPSRMC2[32];
};
struct GpsPacket3 {
    char GPSRMC3[16];
};
NRFLite _radio;
RadioPacket _radioData; 
GpsPacket1 _radioData1; 
GpsPacket2 _radioData2; 
GpsPacket3 _radioData3;
//===============================================================================================
void setup(){
    Serial.begin(115200);
       _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE1MBPS, 125);
    
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN)){
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
}
//==================================================================================================
void loop(){

    if (_radio.hasData()){
        _radio.readData(&_radioData1);
        String msg = _radioData1.GPSRMC1; 
        Serial.print(msg);
    }
}
```

Maybe there's a simple mistake in my code that I don't see or it's just not as robust as your idea of the SentenceId? I look forward to your RX example and comparing it to mine!

dparson55, 2020-08-17:
I see the problem with the RX code, it will print all 3 packets the way it is written here.  After the 3 packets are sent, the first loop iteration where `hasData` is true will result in `Serial.print` starting to print the first packet read using `_radio.readData`.  During the second loop iteration the second packet will be read and then added to the Serial object's printing buffer.  It doesn't matter that you named the variable `radioData1`, the radio indicated that is has another packet (packet #2 that you sent) and the radio will simply write this second packet to the variable you provided.  The same goes for the third loop iteration where the third packet will be read into the `radioData1` variable and then queued for printing.  In order to only print the first packet, you would need to read it into a different variable than the other two packets and only call `Serial.print` on the first packet.

The second problem is packets are never recombined into a single variable for printing.  The current logic would print out all 3 packets on the same line, and then when another set of three packets is received they will also be added to the same line. 
 In order to solve this problem, read and `Serial.print` the first packet, read and `Serial.print` the second packet, and then do a `Serial.println` on the third packet to start a new line for the next set of packets.

I didn't finish my example yet but made more progress.  I should be done tomorrow or the day after.

dparson55, 2020-08-17:
Here is the output from the code above and it looks to be working.  So you can use it as a starting point if you cannot get your current implementation to work.

**TX Output**
```
Sending Sentence 1
    $GPGGA,123456.00,789.00,S,12345.00,E,6,78,9.00,12356.00,M,7.00,M,*8F,4000
  Sending Packet 0...Success
      $GPGGA,123456.00,789.00,S,123
  Sending Packet 1...Success
      45.00,E,6,78,9.00,12356.00,M,
  Sending Packet 2...Success
      7.00,M,*8F,4000
```
**RX Output**
```
Receiving Sentence 1
  Packet 0
      $GPGGA,123456.00,789.00,S,123
  Packet 1
      45.00,E,6,78,9.00,12356.00,M,
  Packet 2
      7.00,M,*8F,4000
  Complete
    $GPGGA,123456.00,789.00,S,12345.00,E,6,78,9.00,12356.00,M,7.00,M,*8F,4000
```

RCarran, 2020-08-17:
I get the same results you do with your example. Next up on my part is to adapt your `sentence.Data` to equal the GPRMC data from my GPS module! I'm still curious and confused though regarding my RX code and I really want to learn. How should I go about having each packet read into their respective `_radioData` variable? I saw you mentioned reading each packet into one big array and print that out, and I've also been trying to implement the `sentence.Id` idea, but I was planning on adding four temperature sensors into my project that would read the temp data into its own struct and packet (I'm not the best with saving memory haha but I hope to get to that level of programming soon) and thought if I could control the `_radioData` variable then I'd be fine with managing the GPS data and temperature data. Though I might me biting off more than I can chew with this project now. By the way thank you for the suggestions and examples you have provided, they really are insightful even if I find myself having to reread the comments multiple times to understand what is going on haha.

dparson55, 2020-08-17:
To read the packets into 3 separate variables here are 2 options:
* Do as I did and add some type of Id to the packet so you can determine which packet it is.
* Assume that the first packet the receiver gets should be put into `_radioData1`, then the next packet that is received should be put into `_radioData2`.  That sort of thing.

RCarran, 2020-08-17:
Thank you so much! The `currentPacketId` example worked! I even tried it with my temperature sensors and the GPS data at the same time and it is printing the data exactly the way I want it to! I don't know why I wasn't able to wrap my head around the `packetId` variable but yes I see your edits now and I understand! Thank you soooo much for your help and patience! I get scared asking for help on my code online but you really did help me out. I'll keep looking at the examples you provided to try and further optimize my code but I consider this issue closed!

closed 2020-08-17 by RCarran

dparson55, 2020-08-17:
Good to hear.  Just keep in mind how fragile using the 2nd approach is, the moment the transmitter sends a packet but the receiver does not receive it, the packets will become out of sync.  So that's a huge reason why approach 1 should be used.