dparson55/NRFLite #56: HW issue

sweshn, 2021-01-10:
Hi
I have a project that works fine with a NRF24L01+ module in different speeds. But when using a NRF24L01+PA+LNA module everything works except using ackdata. It goes wrong in TX end. 
Regards
Staffan

dparson55, 2021-01-10:
Thanks for the report, is this similar to issue #44?  There is a known problem with some PA+LNA versions of the radio that do not receive ACK packets after successfully sending normal data packets.  I have one of these PA+LNA radios and it is some hardware problem.

Here are some options:
- Use only ```NRFLite::NO_ACK``` sends and implement sending and receiving data similar to the software-based two-way communication example.
- Use the [Radiohead](https://www.airspayce.com/mikem/arduino/RadioHead) library.  It implements ACK data via software rather than using the hardware-based ACK support built into the nRF24L01+ chip.
- Try a different PA+LNA radio module made by a different manufacturer.

sweshn, 2021-01-11:
Hi

Thanks for the response. I have missed #44. It is the same problem.

I will try your options.

Regards

Staffan

Skickades från E-post för Windows 10

Från: Dave Parson
Skickat: den 10 januari 2021 23:43
Till: dparson55/NRFLite
Kopia: sweshn; Author
Ämne: Re: [dparson55/NRFLite] HW issue (#56)

Thanks for the report, is this similar to issue #44? There is a known problem with some PA+LNA versions of the radio that do not receive ACK packets after successfully sending normal data packets. I have one of these PA+LNA radios and it is some hardware problem.
Here are some options:
• Use only NRFLite::NO_ACK sends and implement sending and receiving data similar to the software-based two-way communication example.
• Use the Radiohead library. It implements ACK data via software rather than using the hardware-based ACK support built into the nRF24L01+ chip.
• Try a different PA+LNA radio module made by a different manufacturer.

closed 2021-01-20 by dparson55