dparson55/NRFLite #57: stuck on _radio.init

superhacker35, 2021-01-12:
Hi! Thank you for sharing your library! I just started with arduino a year ago and wanted to try NRF24L01. I decided to try it with your library. I initially tried your sample TX and TX codes but I'm getting TRUE value for     if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))  so I'm not sure if this can be a board problem or NRF problem. I'm using UNO for TX and LEONARDO for RX. I've also tried switching both TX and RX with nano but still stucked with radio.init.

dparson55, 2021-01-12:
When ```init``` fails it means the microcontroller tried to adjust one of the configuration registers within the nRF24L01 chip and then read it back to make sure it changed, but no change was made.  So this means the Uno was not able to interact with the radio module via its SPI or 2-wire connection, depending on which one you are using.

Make sure each of your breadboard wires has a very low resistance (some have bad connections) and double check the SPI pins or 2-wire connection pins that you are using between the Uno and radio module are correct.  Also make sure you are powering the radio with the 3.3V output from the Uno and have connected the Uno's GND to the radio.

You can then try running this code which calls ```printDetails``` to print out the values of all the nRF24L01 registers.

```
#include <SPI.h>
#include <NRFLite.h>

const static uint8_t RADIO_ID = 0;
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

NRFLite _radio(Serial);

void setup()
{
    Serial.begin(115200);
   
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
    }

    _radio.printDetails();
}

void loop()
{
  
}
```

You should see something like this in the serial monitor.

```
CONFIG 00001011 
EN_AA 00111111 
EN_RXADDR 00000011 
SETUP_AW 00000011 
SETUP_RETR 00011111 
RF_CH 01100100 
RF_SETUP 00001110 
STATUS 00001110 
OBSERVE_TX 00000000 
RX_PW_P0 00000000 
RX_PW_P1 00000000 
FIFO_STATUS 00010001 
DYNPD 00000011 
FEATURE 00000111 
TX_ADDR 231,231,231,231,231 
RX_ADDR_P0 231,231,231,231,231 
RX_ADDR_P1 1,2,3,4,0
```

superhacker35, 2021-01-13:
Sir! Thank you so much!! I really appreciate your answer. The code you gave help me identify the error. All a long i have switched CSN and CE pin. Looking forward  to using this library more! Will explore on my own for now :)

closed 2021-01-13 by superhacker35

dparson55, 2021-01-13:
Glad to hear, good luck with the radio!