dparson55/NRFLite #58: NRFlie using second SPI hardware bus

freetoair, 2021-02-23:
Hi Dave ! Let me explain my problem first, I must have to use 2` SPI on ESP32 because TFT ST7789 has CE pin hardwired on GND, so i can't use shared SPI with nRF24L01. According to the example for ESP32 SPI_Multiple_Buses it is possible to define two objects that use different hardware SPI interfaces (HSPI, VSPI). Is it possible to somehow use a new SPI object in NRFlite, which is previously defined in the void setup (). Obviously I'm not some C/C++ expert, but I tried everything I could. Even the version with SoftSPI would be helpful.

dparson55, 2021-02-24:
NRFLite doesn't support SPI objects and just uses the default SPI pins setup by the SPI library, but I think you can use the radio and display on the same SPI bus.  The CE pin is not used for SPI communication, it is the _chip select/slave select_ pin that controls which device interacts with the SPI bus.  On the nRF24L01+ this pin is labeled CSN while on the TFT ST7789 it is labeled CS.  On the ESP32 it [seems like](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPI/src/SPI.h#L50) the HSPI pins are used as the default SPI pins, so try connecting the ESP32 HSPI pins for MISO, MOSI, and SCK to the associated pins on the radio and display, then use separate pins for the CSN and CS pins on the devices.

I just threw this together from some examples I found, maybe it will help.

```
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ST7789.h>
#include <NRFLite.h>

// Display
#define TFT_CS     15 // Using 15 because it is the default Slave Select pin for HSPI
#define TFT_DC     9
#define TFT_RST    8  // Reset pin, or set to -1 and connect to Arduino RESET pin

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
SPIClass * hspi = NULL;

// Radio
const static uint8_t RADIO_ID = 0;
const static uint8_t PIN_RADIO_CE = 4;
const static uint8_t PIN_RADIO_CSN = 5;

NRFLite _radio;

void setup()
{
    Serial.begin(115200);
    
    // Setup the SPI bus.
    hspi = new SPIClass(HSPI); // MISO = 12, MOSI = 13, SCLK = 14, SS = 15
    hspi->begin(); 
    
    tft.init(240, 240, SPI_MODE0); // The radio uses mode 0 so hopefully this mode works with the tft.
    tft.fillScreen(ST77XX_BLACK); // See if the display actually works.
    
    // Indicate to NRFLite that it should not call SPI.begin() during initialization since it has already been done.
    uint8_t callSpiBegin = 0;
    
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE2MBPS, 100, callSpiBegin))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
}

void loop() { }
```

Lukas1337, 2021-02-24:
I have a similar issue with STM32F103C8 Bluepill, although I'm not using a display and just want to utilize the second SPI.

freetoair, 2021-02-24:
Hi Dave ! Thanks  for your prompt response. Looks like we didn't understand , I don't have access to the CE line. It is simply not connected to the display port. Someone who designed the printed  board thought it was not necessary and connect it permanently to the GND. That's why I can't use a shared SPI bus, Once you turn on the ST7789 it doesn't release the SPI bus  so that some other peripherals can use the same SPI lines. In the meantime, I was able to customize another library (RF24.h) that has the ability for SoftSPI to work with ESP32. There were small problems with setting the parameters for receiving NRFlite format, but I succeeded. Otherwise for those who want to use this library (RF24.h) on ESP32, they should know that it does not work in the original with ESP32 in SoftSPI mode. Since the ESP32 is a very fast controller SoftSPI works satisfactorily fast. Anyway thanks for the effort to help me !

@Lukas1337 NRFlite calls the SPI.xxx functions directly from the SPI driver for a particular controller, and the other connected SPI peripherals do the same, causing a collision on the bus. Although you created a new SPI object for SPI_2 it is never called from NRFlite.

regards, freetoair !

Lukas1337, 2021-02-24:
Thank you for the reply, how do I set up NRFlite so that it calls the new object? Or is there a better way?

freetoair, 2021-02-24:
@Lukas1337 That was my question for @ dparson55. Unfortunately, changing the library seems either impossible or difficult. I'm not good enough at C/C++ to be able to do it myself. Unfortunately !!!

freetoair, 2021-02-24:
@Lukas1337 If you use a fast enough controller on the receiving side, then the variant with RF24 and SoftSPI will be enough in most cases.

dparson55, 2021-02-25:
@freetoair I think we are still not on the same page.  You mentioned again that you do not have access to the CE pin.  This is ok, it is not used for SPI communication.  You do not need access to the CE pin for what I was suggesting.

CE = chip enable = not used for SPI
CS/CSN/SS = chip select/chip select not/slave select = used for SPI communication

Are you actually saying the display you are using does not expose the CS pin?  If you have a picture of the display and the pins it exposes, I could double check for you.

dparson55, 2021-02-25:
@freetoair and @Lukas1337 I started looking into if I could easily update NRFLite to allow providing an SPIClass object in the constructor.  I think it will be straightforward but I don't have an ESP32 to test with.  If I make the change would you be able to test it for me?

Lukas1337, 2021-02-25:
Wow, thanks for your work. I'll try and test it this weekend!

dparson55, 2021-02-25:
Ok thanks!

freetoair, 2021-02-25:
First I apologize for possible spelling mistakes, English is not my native language so I use Google Translate. I have now sent a picture of the display. Yes, that's right, the CE line on the printed  board, that holds the display, is not connected.
[image]

I have ESP32, BluePill (STM32F103C8T6) and DIY MORE (STM32F407VGT6) available. I'll try it on all three boards now, and I'll send you how things are.

regards.

freetoair, 2021-02-26:
Yes, I forgot to mention, my first choice for a graphics library is TFT_eSPI. This is because, as @bodmer claims, this library is better suited for 32 bit controllers, and I guess faster.  In the meantime, I  did it to connect the CE/CS pin on the ST7789 display instead of BLK, and the variant that Dave suggested  with Adafruit GFX works great.  But the TFT_eSPI/NRFlite (shared SPI) variant didn't work for me, probably due to the complicated TFT_eSPI setup.  Now I need time to connect all this and try it.

dparson55, 2021-03-11:
@freetoair just checking in to see if you can try out the new feature.

closed 2021-06-16 by dparson55