dparson55/NRFLite #60: Seeeduino Xiao compatibility?  Not an issue, but didn't know where else to submit a question

HR19, 2021-04-15:
I'm wondering if it's possible/feasible to add support for the Seeduino Xiao?  It's one of my favorite MCUs and I have a couple of them, and it would be really cool to have this feature.  I have a cheap oscilloscope and I'm not afraid to manipulate libraries, but I have no clue where to even start.  Note: I tried it already, but the program wouldn't compile, said that there's no initTwoPin function, although the regular, full wiring compiled and worked just fine.

HR19, 2021-04-15:
Would the place to start would be to try copying the initTwoPin function from another device and just see if it works?

dparson55, 2021-04-16:
What IDE or compiler are you using with this board?  The initTwoPin method is wrapped by an `if defined` block to prevent non-AVR architectures from including it during compilation with the Arduino IDE, so it seems like the compiler being used isn't fully respecting this logic.

Since this method can only be used with AVR architectures, you can manually do what the compiler should be doing and simply delete the method header from [NRFLite.h line 38](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L38) and the associated method implementation from [NRFLite.cpp starting at line 57](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L57).

HR19, 2021-04-16:
I'm using Arduino IDE.  It is correctly stopping me from using the SAMD21 on the Xiao.  I'm just curious if it's reasonable to try and see if it can be applied to that one.  If there's something fundamental in the architecture that makes it hard or impossible on SAMD21, then it's not a really big deal, I can use the full pinout instead.

dparson55, 2021-04-16:
Oh I misunderstood your question, you are asking if it is possible to update the 2 pin feature of NRFLite to support the SAMD21 architecture.  Absolutely it can be done, it involves toggling the pins as described in the article [nRF24l01 control with 2 MCU pins using time-division duplexed SPI](http://nerdralph.blogspot.com/2015/05/nrf24l01-control-with-2-mcu-pins-using.html).  Because timing is critical, interacting with pins using pinMode, digitalRead, and digitalWrite functions is not possible, so it is necessary to directly interact with the pins using so called direct port manipulation.  I did a quick search and there are [discussions](https://forum.arduino.cc/t/faster-pin-i-o-on-zero/322111) on how to do this on the SAMD21.

So if you read through that article and study the [initTwoPin](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L57) and [twoPinTransfer](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L612) methods, you can see exactly how it is implemented within AVR and then use those methods as a starting point for implementing the same in SAMD21.  It is really complex stuff though and I don't have a SAMD21 to even experiment with, but I am happy to answer any specific questions you have if you decide to undertake the challenge.

HR19, 2021-04-16:
Thanks!  It honestly doesn't like a terrible complex idea to adapt it since you and the other person have dinner most of the leg work, but it's gonna have to be a back burner thing for me.  For the moment, I'm just too frustrated in general with these nRF24 modules I have (maybe because I got cheap ones and I'm not using breakouts), I'm just done.  I'm probably just going to use an ESP8266, or take apart an RC car since all I need is simple on/off, one way communication.

dparson55, 2021-04-16:
Ok sounds good.  Cheap nRF24 modules are all I have and common issues are problems with the 3.3V going to them.  If the 3.3V is shared between the SAMD21 and the nRF24 maybe the power supply is to blame.  Try giving the nRF24 a dedicated 3.3V supply and add a 10uF capacitor between its VCC and GND.  These are very common solutions when you are experiencing inconsistent performance.  If you are instead having trouble with communication over a long distance you can also consider RFM69 modules which are more suited for that type of application and they will go much further than ESP modules.

closed 2021-04-16 by dparson55