dparson55/NRFLite #63: Mega 2560 compatibility

RMW946, 2021-05-18:
Hello. I've made a couple of projects with nrflite. Really a great library! The issue I have is using the 2 pin control on an Arduino Mega. The SPI pins (50-53) are used for an TFT display and SD card reader.

 This handset works fine with a generic nrf24l01. However it locks up when used with a longer range module with antenna.
 I understand there are differences between radio modules due to different manufacturers. 
The project has a keypad and 2 pots. Transmit is only on change of state. Pressing the keypad is an auto ack send, and the pots are programmed as No Ack.
The pots seem to cause the lock so the assumption is the more fancy radio module cannot cope.

What can I adjust in the library which might help?
I have been reluctant to edit libraries until I get familiar with an IDE which can keep different versions, also, I'm more of a hardware person.
Any suggestions? Thanks!

[image]

dparson55, 2021-05-19:
Here are 2 quick things to try:
1. Change the pots to use a REQ_ACK send rather than NO_ACK.
2. Limit how often changing the pots is allowed to perform the NO_ACK send, e.g. if a user turns the pot consistently for 1 second, only allow a data transmission once every 10 milliseconds.  So in this case 100 packets would be sent over the course of that 1 second.

The 2-pin control is quite a bit slower than using SPI and when doing REQ_ACK sends, the library can regulate how quickly a program will ask the radio to send data.  However with NO_ACK sends, NRFLite is limited to checking to see if the radio's TX buffer is full before enqueuing a new packet (see [line 463](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L463)).  Maybe there is a bug and this check is not working if a huge number of packets are attempted to be sent by the ATmega, but I have not seen this bug in action before so it is only a guess.

Just for reference, there is a 500 microsecond delay before and after every command issued to the nRF24L01+ to allow the 0.1uF capacitor to charge and discharge for use with the 2-pin control.  So that really slows down how quickly the microcontroller can interact with the radio.

RMW946, 2021-05-19:
Option 1, use REQUIRE_ACK solved the problem. Thanks very much! I finished this handset last year and was disappointed the long range radio didn't work. All good now.
I will review your other recommendations when I have more time.

Yes, I have looked at networking the SPI with the same TFT_SD module, I will try it on the next version.

The library notes state that all send transmissions are REQ_ACK by default, is that correct? 
I will review my program and get back with a few more questions if that's ok.

[image]

dparson55, 2021-05-19:
Good news, well glad that fixed it. 

Yes the default send type is defined [here](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L50) so if you do not specify one in the send method, it will default to REQUIRE_ACK.

closed 2021-05-19 by dparson55