dparson55/NRFLite #64: Receiver responds with 0 Failed TX

sjgcornish, 2021-05-18:
Hi 

Using the Basic TX and Basic RX with two NRF24L01+ PA/LNA.   I set the TX to NRFLite::NO_ACK  and hence get a "success" at the TX but the  RX prints for example "Radio 1, 86750ms,  0 failed TX"

The ms count is going up which suggests that the transmission is OK, so I am puzzled by the "0 failed TX"  does this mean 0 retries hence success..
Thank you

dparson55, 2021-05-19:
The receiver simply prints out the _radioData packet when it receives one, it doesn't perform any logic on the _radioData.FailedTxCount variable.  So you need to look into the transmitter code to determine what sets its value.

When doing NO_ACK sends there is no way for the transmitter to know if it successfully sent the packet, so NO_ACK sends are considered a success by the radio hardware itself.  Because of this, the logic in the transmitter code to increment the _radioData.FailedTxCount field will never execute if there is actually a failure.  So every time the _radioData packet is sent, it will contain a 0 in the FailedTxCount.

sjgcornish, 2021-05-19:
Thank you for clarifying this. If I send TX on one device will running a channel scan on the other would I expect to see a signal on the default channel of 100 because I do not, if so what could be preventing transmission since the scanner is only looking at signal strength.  Both devices run the channel scanner Ok which suggests they are both working and I have swapped devices over. I am using a Arduino Uno and Arduino mini Pro (3.3v 8MHz)

dparson55, 2021-05-19:
From what you mentioned in your first question, it seems to me like everything is working.  You are successfully transmitting a packet to the receiver and it is printing out the contents of that data packet.  If the receiver does not receive a data packet, it will not print out any messages.

Are you asking a new, separate question about the channel scanner?  If so, it is an interesting question and something I have not yet tried.  You'll need to update the Basic_TX code to perform NO_ACK sends as fast as possible to give the scanner a chance to detect the signal.  The Basic_TX code only sends 1 packet per second and the scanner only monitors a channel for about 100ms, so it would be easy for the scanner to miss the 1 packet that is sent.  So just update Basic_TX by removing the delay(1000); in the loop so it constantly sends data and then the scanner will hopefully detect the channel being used.

sjgcornish, 2021-05-19:
OK the scanner did detect the transmit on channel 100. I went back to the very basic example with the incrementing data and it is working. I just need to find out why the Basic example did not.

dparson55, 2021-05-19:
Ok good to hear, I'll have to try this out one day just to see it.

closed 2021-05-19 by dparson55