dparson55/NRFLite #65: ESP8266 RX Example boot mode:(3:6)

sjgcornish, 2021-05-19:
Hello
After some success using Arduino's I decided to give the ESP8266 ago.  I am using a NodeMCU -12E  V2.  With or with out the NRF connected I get an error...

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 

I tested the SPI on its own NodeMCU to UNO and it works.  The channel scanner gives the same error. Any ideas
Thank you
Steve

dparson55, 2021-05-19:
Looks like the same issue as #31.  Unfortunately the person who created that issue never provided a solution so I don't have any suggestions.  There are people using NRFLite with esp8266's though so it is possible to make it work.

sjgcornish, 2021-05-19:
It is caused by the statement
_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN) which points to either the CE or CSN pin.

If I remove this line and enable the slave with digitalWrite(CSN, LOW);  then no error
Changing the CE pin has no effect i.e. still the watchdog timer reset. My guess is that something in the radio initialization code is taking to long which triggers the watchdog timer and then the error Cause 4 is the timer)

sjgcornish, 2021-05-19:
My guess is that the..

_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN)

Is taking to long to respond with either "connected to radio" or "cannot connect to radio" which causes the watchdog timer to trigger and hence a following reset. It may be that the radio initialization code includes try events or there is a wait period  which means the watchdog timer kicks in. i.e. nothing to do with the SPI. I would also guess that NRFLite has similar functions to the wifi.h to initialise and connect to the wireless in the ESP8266 module

dparson55, 2021-05-19:
Ok very interesting, there is a 100 ms delay when the library waits for the radio to power up.  As a test, can you change the line

https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L373

```
delay(OFF_TO_POWERDOWN_MILLIS);
```

to this instead

```
  uint8_t totalWaitTime = 0;
  
  while (totalWaitTime < OFF_TO_POWERDOWN_MILLIS)
  {
    delay(1);
    totalWaitTime++;

    #if defined(ESP8266) || defined(ESP32)
      yield();
    #endif
  }
```

This will do the same thing, basically waiting 100 ms to allow the radio to transition from its off state to its power down state, but it will call `yield` every millisecond so the ESP8266 can do its thing.

sjgcornish, 2021-05-20:
No Go!. I am very puzzled, I replaced your _radio.init with

uint8_t NRFLite::initRadio(uint8_t radioId, Bitrates bitrate, uint8_t channel)
{
   uint8_t success = 0;
    return success;
}

and the arduino setup code with...

    uint8_t success = 0;
    uint8_t success = _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN);
    Serial.println("success: "); Serial.println(success);

This still results in the same hardware watchdog error (cause 4) but when I comment out the _radio.init line I do not get the error. This suggests that there is something else happening when _radio.init is called but ihave no idea what.

This link explains the watchdogs

https://techtutorialsx.com/2017/01/21/esp8266-watchdog-functions/

The 3,6 is the boot mode in this case Flash = 3 and 6 and think UART.

dparson55, 2021-05-20:
The only other suspect code that I can think of is the call to SPI.begin within `init` [here](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L45).  Try commenting that out next to see if it prevents the crash from occurring.

If that works, maybe you can try calling SPI.begin within your setup function as shown in the [ESP32 example](https://github.com/dparson55/NRFLite/blob/master/examples/Basic_RX_ESP32/Basic_RX_ESP32.ino).  I'm not sure if this works for ESP8266's but it's worth a shot.

sjgcornish, 2021-05-20:
No difference however I wonder if I have been looking at the same place and I wonder if it is a pin assignment problem that is causing the boot to fail. i.e. one of the pins controlling booting is being changed during boot and hence the boot failure ad hardware reset 

1. I am using NodeMCU, Your pin out for the ESP2866 module in the basic example is
2866 Module                                                                           NodeMCU
CE    -> 4     GPIO16                                                                    ??                                                                         
CSN   -> 8 (and connect 8 to GND via a 3-5K resistor)               ??               
MOSI  -> 7  GPIO13                                                                    D7
MISO  -> 6  GPIO12                                                                    D6
SCK   -> 5  GPIO14                                                                      D5

Pin 8 is actually VCC  so I think you mean pin 16 which is GPIO15 or the CS pin which on the NodeMCU is pin 8 So in
const static uint8_t PIN_RADIO_CE = 4;
const static uint8_t PIN_RADIO_CSN = 8;

Are you referring to D4 and D8 ?

dparson55, 2021-05-20:
Unfortunately I don't have an ESP8266 so don't know, the example came from another person who provided it after confirming it worked with their ESP8266 ESP-12E module.

The only things left in the init method are pinMode and digitalWrite calls, so hopefully you can debug that on your own.  Just remove all the code in init and add one line at a time until it fails, then you know what is causing the problem.

I read that the watchdog expires after 1.5-2 seconds, so the 100 millisecond wait that is in initRadio is probably not the problem.

sjgcornish, 2021-05-20:
Hi Done

I have found the problem it is line 32 and line 33

pinMode(_csnPin, OUTPUT);
digitalWrite(_csnPin, HIGH);

If I comment these two lines out, I think the reason is that during flash this pin needs to be an input. Even if I add these to lines in the Arduino setup after the _radio.int call I get the error. What do you use this pin for. I understood the CSN (i.e. the SS pin) is the slave select (select when low) . It looks as if it does not like the pinMode(PIN_RADIO_CSN, OUTPUT);

if (_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN)){
  Serial.println("Cannot communicate with radio");
}else{
  Serial.println("Connected with radio");
 }
This does not cause any errors and reports connected to radio although as yet I have not actually tested to see if I can receive data

sjgcornish, 2021-05-20:
D8 | GPIO15 | pulled to GND |(CS)Boot fails if pulled HIGH

sjgcornish, 2021-05-20:
//const static uint8_t PIN_RADIO_CSN = 8;

#define PIN_RADIO_CSN D0

No errors

sjgcornish, 2021-05-20:
This is not as easy as I thought although I got rid of the boot problem. If I connect the CSN pin to ground I can connect to the radio  but if I try connecting it to a pin such as D1, etc then I cannot connect with the radio even if I comment out lines 32 and 33 or set the pin to LOW in line 33. I also tried a 3.3k resistor from pin D1 to GND. 

As soon as I define the CSN pin as an output on the NodeMCU it does not connect. so what is different between connecting CSN to ground and a NodeMCU pin defined as an output

dparson55, 2021-05-20:
Maybe these issues are related to an external flash chip on your ESP8266 board.  According to https://nodemcu.readthedocs.io/en/release/modules/spi/ there are 2 SPI busses on the ESP8266, one of which might be used by onboard flash.  If you are using SPI bus ID 0, maybe you should try switching to SPI bus ID 1...or the other way around.

> The ESP hardware provides two SPI busses, with IDs 0, and 1, which map to pins generally labelled SPI and HSPI. If you are using any kind of development board which provides flash, then bus ID 0 (SPI) is almost certainly used for communicating with the flash chip. You probably want to choose bus ID 1 (HSPI) for your communication, as you will have uncontended use of it.

sjgcornish, 2021-05-21:
Working sending from UNO receiving on NodeMCU. I am afraid I have been a bit of a "woolly" I assumed that the "cannot connect to radio" was referring to the radio connected to the NodeMCU not that it is actually testing that it has a connection to the sender radio. I have been testing with out the TX connected and on.

But saying that if you add the while(1) and the TX is not on you get a watchdog error and a reboot so just leave this out.

dparson55, 2021-05-21:
Well unfortunately things are still not working as designed.

>  I assumed that the "cannot connect to radio" was referring to the radio connected to the NodeMCU

Your assumption is correct:  the `init` method configures registers within the nRF24L01 chip that control the bitrate, channel, and other settings; `init` does not ask the nRF24L01 chip to perform packet sends or receives, it only configures the nRF24L01 chip.

One of the last steps init does is read back one of the registers it modifies within the nRF24L01 and if the value of the register equals the value that was assigned to it, it means that init was successful and was able to successfully communicate with the nRF24L01.  When this occurs init will return `1` to indicate it was successful, otherwise a `0` is returned.

So here is what the status messages should read in the example you provided along with a potential fix for the while loop:

```
if (_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN)){
  Serial.println("The ESP8266 successfully configured the registers within the nRF24L01");
}else{
  Serial.println("The ESP8266 could not validate the register assignments within the nRF24L01");
  while (1) { // Wait here forever since there is some problem with the ESP8266 and nRF24L01 communication.
    yield();  // Allow the ESP8266 to boot successfully by allowing its background tasks to execute.
  }
}
```

To troubleshoot the validation logic it would be helpful to see what the logic is doing inside the NRFLite library.  If you could make these modifications and give me the serial output, that would be a good first step.

Update the constructor for the NRFLite object in your main code like this:

```
NRFLite _radio(Serial); // Pass a Serial object into NRFLite so it can print messages.
```

And then replace [startRx](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L188) within NRFLite.cpp:

```
uint8_t NRFLite::startRx()
{
    waitForTxToComplete();

    // Put radio into Standby-I mode in order to transition into RX mode.
    digitalWrite(_cePin, LOW);

    // Configure the radio for receiving.
    writeRegister(CONFIG, CONFIG_REG_SETTINGS_FOR_RX_MODE);

    // Put radio into RX mode.
    digitalWrite(_cePin, HIGH);

    // Wait for the transition into RX mode.
    delay(POWERDOWN_TO_RXTX_MODE_MILLIS);

    // Validate the CONFIG register was assigned to the correct value.
    uint8_t wasConfigRegSet = readRegister(CONFIG) == CONFIG_REG_SETTINGS_FOR_RX_MODE;
    debug("Checking CONFIG register using method 1: ");
    debugln(wasConfigRegSet);

    debug("Attempted to set CONFIG register to: ");
    debugln(CONFIG_REG_SETTINGS_FOR_RX_MODE);

    debug("Read the following value for the CONFIG register: ");
    debugln(readRegister(CONFIG));

    // Just as a sanity check, try validating the register using another method.
    uint8_t readConfigRegValue = readRegister(CONFIG);
    if (readConfigRegValue == CONFIG_REG_SETTINGS_FOR_RX_MODE)
    {
        wasConfigRegSet = 1;
    }
    else
    {
        wasConfigRegSet = 0;
    }
    debug("Checking CONFIG register using method 2: ");
    debugln(wasConfigRegSet);

    return wasConfigRegSet;
}

dparson55, 2021-05-28:
Closing due to lack of response. Maybe we can reopen when you have more time.

closed 2021-05-28 by dparson55

sjgcornish, 2021-05-28:
Changing the radio.int success/unsuccess code as per you code above solved the problem and I have not had time to investigate why by adding your new StartRX code. I would also suggest that the ESP8266 SPISlave.h code be used rather than the standard SPI.h