dparson55/NRFLite #66: Transmission fails at 1MB bit rate

sjgcornish, 2021-06-22:
Hi
Using Arduino nano as receiver and and pro-mini as the transmitter and two nrf24l01+pa+lna and the basic RX /TX examples. At at bit rate of 2MB all is Ok but at 1MB all transmission fails. I tried adding different delays in the transmitter loop 100ms, 200ms etc and it makes no difference, not that i would expect it to.

On a side issue for which advice would be welcome at 2MB transmission is successful but if I move the transmitter then I get a one or two failures and then stabilises when I stop moving it around. When the transmitter is stationary I can prod, shake the transmitter even touch it with my finger and there is no error. I have also used the nrf24l01 (with out the pa+lna) as the receiver and it is more sensitive to the transmitter being moved around.  Note we are talking about distances of a 1meter to 4 meters

Has any one else experienced this.?

dparson55, 2021-06-26:
I did a quick test and 1Mbps worked ok, so it might be an incompatibility with the particular PA+LNA modules that you have.  For my test I used two Arduino Uno's that each use an ATmega @ 16MHz.  Other people have reported issues with some PA+LNA modules (#44, #56, #63), and I personally have one that will not work with the auto-acknowledgement features of the nRF24L01+ chip, so there can definitely be issues with these modules.  I've never received a report of the PCB antenna version of the nRF24L01+ acting strange so they seem to be the most compatible.

[image]

For troubleshooting, try using one PA+LNA module and another PCB antenna module and switch them between TX and RX operation to see if you can identify if one of them will not work.  It would also be interesting to hear if 250KBPS works.

sjgcornish, 2021-06-30:
Hi Dave

I retried with two nrf24l01 and it worked. Note I was using a 8MHz pro mini for the transmitter and a 16MHz Nano for the receiver so this may have has an impact I will investigate further. If it works with some of the  PA+LNA and not others this suggests to me that it is an internal timing issue in the modules themselves, as a result of manufacturing tolerances, due to the addition of the PA.

dparson55, 2021-06-30:
Ok good deal, it's a shame the PA+LNA modules are so inconsistent.  I never had a good application for them in my projects, so I haven't found a good manufacturer to recommend.

closed 2021-07-06 by dparson55