dparson55/NRFLite #68: Data not transfering between two Attiny85s

jjderooy, 2021-08-11:
Hello,

I have set up two identical breadboards for wireless transmission. Both use an Attiny85 microcontroller, with the core recommended in the NRFLite readme. Both Tinys are set up with NRF modules using the two pin guide. The only difference between the two setups are that one is running as a transmitter with a button connected to a digital pin for testing, the other is set up as a receiver.

The Tinys initialize fine with the radios, but when data (a byte representing whether the button is pressed or not) is sent by the transmitter, the receiver does not receive the message. I've been trying to figure out whats going on for that last few days but have not had success yet.

I read through the .h and .cpp files but am struggling to come up with other ways to debug things. If you have any thoughts let me know. Thanks!

By the way, nice job on the repo, the guides were super helpful, and the code is really well documented and formatted!

Here is the code I am running on the Tinys. I'm using SoftwareSerial and an Arduino Nano for debugging purposes.

Transmitter
```
#include <NRFLite.h>
#include <SoftwareSerial.h>

// For serial debugging of the attiny use this article:
// https://jloh02.github.io/projects/connecting-attiny85-serial-monitor/
// Another weird quirk is that the reset pin needs to be pulled low
// if you want to see serial messages in setup()

// Uncomment the serial stuff to messages, although PIN_BTN will interfere
// with the serial comms

const static uint8_t RADIO_ID = 2;
const static uint8_t PIN_RADIO_MOMI = 1;
const static uint8_t PIN_RADIO_SCK = 0;
const static uint8_t DESTINATION_RADIO_ID = 0;

const static uint8_t PIN_BTN = 3;
byte msg;

// Serial pins. RX -> RX and TX -> TX on arduino nano loaded with a blank sketch
#define RX  3
#define TX  4

NRFLite _radio;
SoftwareSerial ser(RX, TX);

void setup(){
    ser.begin(9600);
    pinMode(PIN_BTN, INPUT_PULLUP);

    // Check if radio turned on. 
    // Serial is passed to a Nano which relays it to the computer
    ser.print("Radio init ");
    if(_radio.initTwoPin(RADIO_ID, PIN_RADIO_MOMI, PIN_RADIO_SCK, NRFLite::BITRATE250KBPS))
        ser.println("success");
    else
        ser.println("fail");
}

void loop(){
    // Send the state of the button at 10Hz
    msg = !digitalRead(PIN_BTN); // Internal pullup enabled so invert

    // Don't care if the RX actually gets the data so require no acknowledgment
	_radio.send(DESTINATION_RADIO_ID, &msg, sizeof(msg), NRFLite::NO_ACK);
    ser.println("Sent " + String(msg));
	delay(100);
}
```

Receiver
```
// Radio circuit
// Follow the 2-Pin Hookup Guide on https://github.com/dparson55/NRFLite

#include <NRFLite.h>
#include <SoftwareSerial.h>

const static uint8_t RADIO_ID = 0;
const static uint8_t PIN_RADIO_MOMI = 1;
const static uint8_t PIN_RADIO_SCK = 0;

const static uint8_t PIN_USB = 3;

// Serial pins. RX -> RX and TX -> TX on arduino nano loaded with a blank sketch
#define RX  3
#define TX  4

NRFLite _radio;
SoftwareSerial ser(RX, TX);

byte msg;

void setup(){
    ser.begin(9600);
    pinMode(PIN_USB, OUTPUT);

    // Check if radio turned on. Serial is passed to a Nano which relays it to the computer
    ser.print("Radio init ");
    if(_radio.initTwoPin(RADIO_ID, PIN_RADIO_MOMI, PIN_RADIO_SCK, NRFLite::BITRATE250KBPS))
        ser.println("success");
    else
        ser.println("fail");
}

void loop(){

    ser.print("Received ");
    if(_radio.hasData())
        _radio.readData(&msg);
    else
        msg = 0;
    ser.println(String(msg));

    // msg is either 0 or 1 which evaluates to LOW or HIGH
    digitalWrite(PIN_USB, msg);

    delay(100);
}
```

dparson55, 2021-08-12:
Thanks, I read through the code and it looks ok, so it might be a power or other hardware issue.

* Add a 10uF capacitor between the 3.3 volt pin and ground on the nRF24L01+ on the transmitter and receiver to help resolve a power issue.
* Confirm the MOMI and SCK pins are what you are expecting by hooking an led to each and toggling them on and off in code.
* Run the [ChannelScanner](https://github.com/dparson55/NRFLite/blob/master/examples/ChannelScanner/ChannelScanner.ino) example on each ATtiny, modifying it as needed for your serial setup, to make sure each shows a graph with received radio signals.  So long as you have 2.4 GHz devices in your home, signals will be received on multiple channels.
* Try checking the resistance of each breadboard wire to make sure none are faulty.

jjderooy, 2021-08-16:
Sorry for the long delay. Those are great suggestions! I'll have another crack at debugging things when I get some spare time.

If I end up resolving the issue I'll update you on what worked so hopefully it helps someone else in the future.

Cheers Dave

dparson55, 2021-08-16:
Ok thanks, good luck!

dparson55, 2021-09-15:
Closing since there's been no further response.

closed 2021-09-15 by dparson55