dparson55/NRFLite #74: RP2040 support

sweshn, 2022-01-15:
Today I use NRFlite for a project with 4 nodes  with Arduino Mega that have worked fine for long period now.

I have tried to use PR2040 instead. I have changed definition for SPI pins still no sucess.

Regards
Staffan

dparson55, 2022-01-15:
What pin numbers did you try that did not work?

I looked around and discovered you can set your own SPI pins as described here https://arduino-pico.readthedocs.io/en/latest/spi.html  Maybe doing that will help.

Also during the search it seems there are 2 sets of hardware SPI pins for the RP2040, maybe try using the other set if you haven't already.

dparson55, 2022-01-22:
@sweshn Any luck?

dparson55, 2022-01-23:
Closing due to lack of response.

closed 2022-01-23 by dparson55

sweshn, 2022-01-23:
Sorry for bad response !! I have problem with uploading sketches to RP2040 . Will solve that and after that continue with NRF lite.