dparson55/NRFLite #75: Question: Can you send a message without a receiver address?

mvoss96, 2022-04-16:
Hi,
Is it possible to send some kind of broadcast message without defining a receiver address?

I would need this functionality to implement dynamic binding of a sensor node and a receiver. So on first boot the sensor would just broadcast a greeting message  and the receiver node would then answer with its own address

dparson55, 2022-04-16:
The nRF24L01 doesn't have such a feature, part of its internal workings is to discard any packets it receives that do not match its assigned RX address(es).  Also on the TX side the nRF24L01 needs an address to send to; it is required since it places this address in the header of the packet it generates and sends.

It's easy to  work around this limitation though by assuming the receiver is one of a set of possible addresses and attempting to send a packet to each address until an appropriate response is received.  NRFLite only supports radio id's (addresses) from 0-255, so it wouldn't take long to send to every single one of them if you didn't want to limit the receiver to a smaller subset.

mvoss96, 2022-04-16:
thank you for the response

closed 2022-04-16 by dparson55