dparson55/NRFLite #76: This library collide with servo.

orangefabri, 2022-05-13:
If I use this library I experience servo to stop after boot on 1 minutes.

I tested my code and confirmed only using servo is okay, but with NRFLite library servo stops moving after 1 minutes.

I think this is interrupt problem.

orangefabri, 2022-05-13:
https://github.com/dparson55/NRFLite/issues/71

I referred this instructions 

but this is not the way of solving problem..

my application need exact timing of servo.

so solution #1 doesn't look good to me.

and if I use solution #3, other applications use SPI protocol gets effect.

dparson55, 2022-05-13:
What you are describing probably isn't related to the #71 interrupt issue because it would only cause a servo to jitter, not cause a full break of the servo control.  The interrupt issue will not get worse and worse over time, it will just cause the timing of the servo control pulses to be slightly off every now and then.  What is it that makes you think it is an interrupt issue?

Since your code is working okay for 1 minute it sounds like a logic problem.  If you can provide a basic tx and rx example to recreate the problem, I would be happy to look them over to see if I can help with a solution.

One quick thing you can do is put some print statements in to determine if a variable you are using is being changed correctly.  If you can identify what is different about your variables before and after the servo stops responding, I would guess you will be able to solve the problem.

orangefabri, 2022-05-13:
this is not a logic problem because I used LED to check my logic.

I tested like this code(simplified) :
[code]
if(~~~~)
{
servo.write(1500);
digitalWrite(RED_LED,HIGH);
}
[/code]

LED still turned on by my logic but only servo don't move after certain times. time randomly changes.

I suspect something close global interrupt and not set again correctly, but it is just my thinking.

if you want, I will show you my code, but it is little bit complicated. and I changed my code to use timer register to manipulate servo. so I have to recover previous code in my brain.

but, it is very simple code
there are only 3 components : servo, 2 color LED, NRF24L01

also, this is not hardware or power problem. I tested with code that eliminated NRF24L01 code. it worked more than 5 mins, more than 3 times.

this is the reason why I suspect interrupt problem.

orangefabri, 2022-05-13:
Here I post my code :

`//#define USE_TIMER_REGISTER
#define USE_SERVO_LIBRARY

#ifdef USE_TIMER_REGISTER
  volatile uint8_t activated_servo = 0;
  volatile uint16_t pulse_width = 2777; //2999
  volatile uint16_t period = 39999; //uint8_t 255
#endif

#ifdef USE_SERVO_LIBRARY
  #define HOOK_CLOSE_VALUE 1389
  #define HOOK_OPEN_VALUE 1722
  #include <Servo.h>
  Servo hook;
#endif

#define DELAY_TIME_HOOK_CLOSE 1000
#define DELAY_TIME_HOOK_OPEN 1000 //unit : milliseconds.

#include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 3;
const static uint8_t PIN_RADIO_CSN = 4;

//숫자로 보내자.
struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    //uint32_t OnTimeMillis;
    uint32_t hook_cmd_code;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

#define USE_SS_LIGHT //SS_LIGHT KRGB3
//#define USE_5V_LED 

#ifdef USE_SS_LIGHT
  #define RED_PIN 7
  #define GREEN_PIN 6
#endif

void setup()
{
  //servo begin
  #ifdef USE_TIMER_REGISTER
  DDRB |= B00000010;
  cli();
  //TCCRXA,B -> X번 타이머에 대한 설정.
  TCCR1A = 0; //normal counting mode. 
  //TCCR1B = _BV(CS11); //set prescaler of 8. 16000000/8 = 2000000. 
  TCCR1B = 0;
  TIMSK1 = 0;
  TCCR1B |= (1 << CS11);
  TCNT1 = 0;//clear the timer count
  OCR1A = 2999;
  //TIMSK1 = (1 << OCIE1A) | (1 << OCIE1B); 
  TIMSK1 = (1 << OCIE1A);
  sei();
  #endif

  
  /*
    By default, 'init' configures the radio to use a 2MBPS bitrate on channel 100 (channels 0-125 are valid).
[... 216 lines cut ...]
   #endif
} //end of void loop

//39999 : 20msec
//2999 : 1.5msec
#ifdef USE_TIMER_REGISTER
ISR(TIMER1_COMPA_vect)
{
  if(activated_servo == 0)
  {
    TCNT1 = 0;
    PORTB |= B00000010; //9 HIGH
    OCR1A = pulse_width; 
    //OCR1A = 2999;
    activated_servo = 1;
  }
  else if(activated_servo == 1)
  {
    PORTB &= ~B00000010; //9 LOW
    //OCR1A = period;
    OCR1A = 39999;
    activated_servo = 0;
  }
}
#endif`

orangefabri, 2022-05-13:
Here is video link :

https://youtu.be/lT5ocwBBQX4

you can see servo stops to moving after 5 minutes.(usually it stops after 1 minutes.)

this is not the problem of servo because I changed servo already and if I use simple code just move the servo, it works after very long time.

orangefabri, 2022-05-13:
also I tested with this code for check power issue :

#include <Servo.h>
Servo hook;

void setup() {
  // put your setup code here, to run once:
  hook.attach(9);
  pinMode(6,OUTPUT);
  pinMode(7,OUTPUT); //LED
  digitalWrite(6,HIGH);
  digitalWrite(7,LOW);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(1000);
  digitalWrite(6,HIGH);
  digitalWrite(7,LOW);
  hook.writeMicroseconds(1500);
  delay(1000);
  digitalWrite(6,LOW);
  digitalWrite(7,HIGH);
  hook.writeMicroseconds(1500);
}

dparson55, 2022-05-13:
Thanks for the videos and code.  To help rule out some of the logic challenges can you try the simplified code below?  Note I've used `writeMicroseconds` for consistency and moved the commands to change the servo position to the same `if` statements as the LED change statements.  I also briefly looked through https://github.com/arduino-libraries/Servo and just wanted to confirm, are you using an ATmega328 microcontroller and is this the library you are using for the servo control?

```
    while (_radio.hasData())
    {
        is_no_signal = false;
        _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.
        if (_radioData.FromRadioId == 1)
        {
            if (is_receiving_close_code == false && _radioData.hook_cmd_code == 1111) //1111 = close
            {
                time_started_receiving_close_cmd = millis();
                is_receiving_close_code = true;
            }
            else if (is_receiving_close_code == true && _radioData.hook_cmd_code != 1111)
            {
                is_receiving_close_code = false;
            }

            if (is_receiving_close_code == true)
            {
                digitalWrite(RED_PIN, LOW);
                digitalWrite(GREEN_PIN, HIGH);
                hook.writeMicroseconds(HOOK_CLOSE_VALUE);
            }

            if (is_receiving_open_code == false && _radioData.hook_cmd_code == 2222) //2222 = open
            {
                time_started_receiving_open_cmd = millis();
                is_receiving_open_code = true;
            }
            else if (is_receiving_open_code == true && _radioData.hook_cmd_code != 2222)
            {
                is_receiving_open_code = false;
            }

            if (is_receiving_open_code == true)
            {
                digitalWrite(RED_PIN, HIGH);
                digitalWrite(GREEN_PIN, LOW);
                hook.writeMicroseconds(HOOK_OPEN_VALUE);
            }
        }
    }
```

orangefabri, 2022-05-13:
[1]"Note I've used writeMicroseconds for consistency and moved the commands to change the servo position"

<- Oh, How could I was stupid! I can't believe... sorry for your inconvinience.
I Imeedietely changed my code from "hook.write" to "hook.writeMicroseconds"

but there was no change, what interesting thing is, the moving angle is same even after changed the code!

but... it didn't helped. the symtom was same.

[2]"same if statements as the LED change statements. "
<- I tried this before and it didn't helped. there is no fault on "if" statement... but I will try soon.

[3]"I also briefly looked through https://github.com/arduino-libraries/Servo and just wanted to confirm, are you using an ATmega328 microcontroller and is this the library you are using for the servo control?"
<- I remember I fiddled with servo library before. for some reason I reinstalled it very many time. as soon as you say about servo library I barely rememberd it.
I will dig into my servo library status and let you know soon

orangefabri, 2022-05-13:
this is the error message I got serveral months ago(I backed up the error message)
I think this is not the cause of problem, but for your reference I post this.

sorry for non-english characters 
:

아두이노:1.8.15 (Windows 10), 보드:"Arduino Duemilanove or Diecimila, ATmega328P"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\hlfan\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\hlfan\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\hlfan\Documents\Arduino\libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -ide-version=10815 -build-path C:\Users\hlfan\AppData\Local\Temp\arduino_build_86611 -warnings=none -build-cache C:\Users\hlfan\AppData\Local\Temp\arduino_cache_352007 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\hlfan\Desktop\아두이노 나노\Transform_Winch_Stable_Server_VeryFirstVersion\Transform_Winch_Stable_Server_VeryFirstVersion.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\hlfan\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\hlfan\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\hlfan\Documents\Arduino\libraries -fqbn=arduino:avr:diecimila:cpu=atmega328 -ide-version=10815 -build-path C:\Users\hlfan\AppData\Local\Temp\arduino_build_86611 -warnings=none -build-cache C:\Users\hlfan\AppData\Local\Temp\arduino_cache_352007 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -verbose C:\Users\hlfan\Desktop\아두이노 나노\Transform_Winch_Stable_Server_VeryFirstVersion\Transform_Winch_Stable_Server_VeryFirstVersion.ino

Using board 'diecimila' from platform in folder: C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4

Using core 'arduino' from platform in folder: C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4

Detecting libraries used...

"C:\\Users\\hlfan\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR "-IC:\\Users\\hlfan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\cores\\arduino" "-IC:\\Users\\hlfan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\variants\\standard" "C:\\Users\\hlfan\\AppData\\Local\\Temp\\arduino_build_86611\\sketch\\Transform_Winch_Stable_Server_VeryFirstVersion.ino.cpp" -o nul

Alternatives for HX711_ADC.h: [HX711_ADC@1.2.8]

ResolveLibrary(HX711_ADC.h)

  -> candidates: [HX711_ADC@1.2.8]

"C:\\Users\\hlfan\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10815 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR "-IC:\\Users\\hlfan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\cores\\arduino" "-IC:\\Users\\hlfan\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.4\\variants\\standard" "-IC:\\Users\\hlfan\\Documents\\Arduino\\libraries\\HX711_ADC\\src" "C:\\Users\\hlfan\\AppData\\Local\\Temp\\arduino_build_86611\\sketch\\Transform_Winch_Stable_Server_VeryFirstVersion.ino.cpp" -o nul

Alternatives for EEPROM.h: [EEPROM@2.0]

ResolveLibrary(EEPROM.h)

[... cut ...]
라이브러리 HX711_ADC를 버전 1.2.8 폴더: C:\Users\hlfan\Documents\Arduino\libraries\HX711_ADC  에서 사용

라이브러리 EEPROM를 버전 2.0 폴더: C:\Users\hlfan\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.4\libraries\EEPROM  에서 사용

폴더:C:\Users\hlfan\Documents\Arduino\libraries\ServoTimer2-master (legacy)의 라이브러리 ServoTimer2-master 사용

라이브러리 Servo를 버전 1.1.8 폴더: C:\Users\hlfan\Documents\Arduino\libraries\Servo  에서 사용

폴더:C:\Users\hlfan\Documents\Arduino\libraries\pollserial (legacy)의 라이브러리 pollserial 사용

exit status 1

보드 Arduino Duemilanove or Diecimila 컴파일 에러.

orangefabri, 2022-05-14:
if you want, I will post code I've tested.

but, as I said before, I tested this code and checked it is not logic issue :

if(~~~~)
{
servo.write(1500);
digitalWrite(RED_LED,HIGH);
}

orangefabri, 2022-05-14:
if I use servotimer2(it uses timer2 instead of timer1.)library to manipulate servo, it works well most of the time. but sometimes NRFLite connection losts. 

if I use timer register(OCR1A, TCCR1A, TCCR1B, etc) to directly manipulate servo, with sei() and cli() not to be interrupted by other process, It works well, but I didn't tested thoroughly.

and, this is the final code I successed to work for your reference(this works very well for more than 10 minutes, so I suspect the problem is colliding with servo library) :

#define USE_TIMER_REGISTER
//#define USE_SERVO_LIBRARY

#ifdef USE_TIMER_REGISTER
  volatile uint8_t activated_servo = 0;
  volatile uint16_t pulse_width = 2777; 
  volatile uint16_t period = 39999; 
  volatile boolean time_to_radio_receive = false;
#endif

#ifdef USE_SERVO_LIBRARY
  #define HOOK_CLOSE_VALUE 1389
  #define HOOK_OPEN_VALUE 1722
  #include <Servo.h>
  Servo hook;
#endif

#define DELAY_TIME_HOOK_CLOSE 1000
#define DELAY_TIME_HOOK_OPEN 1000 //unit : milliseconds.

#include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 3;
const static uint8_t PIN_RADIO_CSN = 4;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    //uint32_t OnTimeMillis;
    uint32_t hook_cmd_code;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

#define USE_SS_LIGHT
//#define USE_5V_LED 

#ifdef USE_SS_LIGHT
  #define RED_PIN 7
  #define GREEN_PIN 6
#endif

void setup()
{
  //servo begin
  #ifdef USE_TIMER_REGISTER
  DDRB |= B00000010;
  //this code should be read with ATMEGA328P manual. with that you can understand the code's meaning.
  cli();
  //TCCRXA,B -> for Xth Timer
  TCCR1A = 0; //normal counting mode
  TCCR1B = 0;
  TIMSK1 = 0;
  TCCR1B |= (1 << CS11);
  TCNT1 = 0;//clear the timer count
  OCR1A = 2999;
  //TIMSK1 = (1 << OCIE1A) | (1 << OCIE1B);
  TIMSK1 = (1 << OCIE1A);
  sei();
  #endif
[... 230 lines cut ...]
} //end of void loop

//39999 : 20msec
//2999 : 1.5msec
#ifdef USE_TIMER_REGISTER
ISR(TIMER1_COMPA_vect)
{
  if(activated_servo == 0)
  {
    TCNT1 = 0;
    PORTB |= B00000010; //9 HIGH
    OCR1A = pulse_width; 
    //OCR1A = 2999;
    activated_servo = 1;
  }
  else if(activated_servo == 1)
  {
    PORTB &= ~B00000010; //9 LOW
    //OCR1A = period;
    OCR1A = 39999;
    activated_servo = 0;
    time_to_radio_receive = true;
  }
}
#endif

dparson55, 2022-05-14:
> if you want, I will post code I've tested.
> but, as I said before, I tested this code and checked it is not logic issue

Your original code was tough (for me at least) to go through since it wasn't formatted and contained a lot of extra logic that isn't related to the radio and servo control.  Still though, I did spend the time to format and go through it and remove excess code, and am just trying to help with the code you provided, so let's continue to use it.

If the suggestion I made provides no value, another troubleshooting step would be to run `detach` and `attach` before attempting to control the servo again.  This will help identify if the initial setup steps performed by the servo library are being lost.

```
    while (_radio.hasData())
    {
        is_no_signal = false;
        _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.
        if (_radioData.FromRadioId == 1)
        {
            if (is_receiving_close_code == false && _radioData.hook_cmd_code == 1111) //1111 = close
            {
                time_started_receiving_close_cmd = millis();
                is_receiving_close_code = true;
            }
            else if (is_receiving_close_code == true && _radioData.hook_cmd_code != 1111)
            {
                is_receiving_close_code = false;
            }

            if (is_receiving_close_code == true)
            {
                digitalWrite(RED_PIN, LOW);
                digitalWrite(GREEN_PIN, HIGH);
                hook.detach();
                hook.attach(9);
                hook.writeMicroseconds(HOOK_CLOSE_VALUE);
            }

            if (is_receiving_open_code == false && _radioData.hook_cmd_code == 2222) //2222 = open
            {
                time_started_receiving_open_cmd = millis();
                is_receiving_open_code = true;
            }
            else if (is_receiving_open_code == true && _radioData.hook_cmd_code != 2222)
            {
                is_receiving_open_code = false;
            }

            if (is_receiving_open_code == true)
            {
                digitalWrite(RED_PIN, HIGH);
                digitalWrite(GREEN_PIN, LOW);
                hook.detach();
                hook.attach(9);
                hook.writeMicroseconds(HOOK_OPEN_VALUE);
            }
        }
    }
```

I spent some time going through the servo library and unfortunately I don't see where it might be conflicting with the SPI and NRFLite libraries.  Even if the global interrupt flag is disabled when the servo library saves it (https://github.com/arduino-libraries/Servo/blob/master/src/avr/Servo.cpp#L289), NRFLite will re-enable it every time it communicates with the radio (https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.cpp#L592).

Maybe worse case, you can use a different radio library and I will update the NRFLite readme with a note that isn't not compatible with the servo library.  I don't understand what the servo library is doing and how NRFLite would be impacting it.  Let me know.

orangefabri, 2022-05-14:
anyway I solved my problem as I mentioned above.

direct manipulating servo with timer register(OCR1A) doesnt conflict with NRFLite library. I will use this code.

and sorry for my tough code.

Since the servo library is not clearly confirmed to conflict with NRFLite library, it may not be necessary to include it in readme file. But if there is someone similar to me, I hope my case can be referenced.

dparson55, 2022-05-14:
Ok good luck with the rest of the project.

closed 2022-05-14 by dparson55