dparson55/NRFLite #78: No se envían los datos

kikesandi, 2022-06-01:
Im mac user: macOS Monterey 12.4
Mac Pro Late 2013
3.5 FHz Intel Xeon E5 de 6 nucleos
64 GB Ram
amd FirePro D300 2Gb

Arduino IDE 1.8.15

***************
I am using examples from the library, specifically the SIMPLE RX and SIMPLE TX, and the data is not being sent.
The "CHANNELSCANNER" sketch works fine... it shows me on the monitor all the channels and their respective usage... as expected.
At the moment of installing the SIMPL-TX the errors begin. The monitor shows:
17:56:10.056 -> Sending 61847 ms...Failed

Installing "Simple-RX" on the serial monitor shows nothing... absolutely nothing.

Because of the above, I decided to install TEXT-TX, and I get:
17:59:24.836 -> 250KBPS bitrate
17:59:24.937 ->   Starting sync
17:59:25.009 ->   Retrying in 4 seconds
17:59:29.055 ->   Retrying in 4 seconds
17:59:33.082 ->   Retrying in 4 seconds
17:59:37.118 ->   Retrying in 4 seconds
17:59:41.164 ->   Retrying in 4 seconds
17:59:45.218 ->   Retrying in 4 seconds
17:59:49.274 ->   Retrying in 4 seconds
17:59:53.296 ->   Retrying in 4 seconds
17:59:57.358 ->   Retrying in 4 seconds
18:00:01.397 ->   Retrying in 4 seconds
18:00:05.429 ->   Retrying in 4 seconds

And if i run "Test_RX" i get:
18:00:31.820 -> 250KBPS bitrate
18:00:31.920 ->   Starting sync

I made my own PCB and the wiring is correct, and the sketches are intact as the examples are shown.

Lastly... yesterday I uptated the Operative Sistem of my mac and the "RF24" library also stop working... So... I asume the error may be this.

dparson55, 2022-06-01:
Sorry I don't have a mac to help troubleshoot any OS issues.  If you can get the Arduino IDE to successfully compile code and upload it to your microcontrollers, here is some code to help with the debugging.  You will need to use SPI and set the CE and CSN pins to the pins you are using, and if all goes well `printDetails` will output all of the important registers NRFLite configures in the nRF24L01+ chip.  Can you try uploading them and post all of the serial output from each?

### TX Code
```
#include "SPI.h"
#include "NRFLite.h"

NRFLite _radio(Serial); // Note we pass in a Serial object so printDetails will work
uint8_t _data;

void setup()
{
    _radio.init(1, 9, 10); // Set radio to Id = 1, along with the CE and CSN pins

    _radio.printDetails(); // Prints nRF24L01+ registers
    Serial.println();
}

void loop()
{
    _data++;
    Serial.print("Sending ");
    Serial.println(_data);

    _radio.send(0, &_data, sizeof(_data)); // Send data to the radio with Id = 0
    delay(1000);
}
```

### RX Code
```
#include "SPI.h"
#include "NRFLite.h"

NRFLite _radio(Serial); // Note we pass in a Serial object so printDetails will work
uint8_t _data;

void setup()
{
    Serial.begin(115200);
    _radio.init(0, 9, 10); // Set radio to Id = 0, along with its CE and CSN pins

    _radio.printDetails(); // Prints nRF24L01+ registers 
    Serial.println();
}

void loop()
{
    while (_radio.hasData())
    {
        _radio.readData(&_data);
        Serial.println(_data);
    }
}
```

kikesandi, 2022-06-01:
On TX i got nothing.

On RX i got this:
CONFIG 00001011
EN_AA 00111111
EN_RXADDR 00000011
SETUP_AW 00000011
SETUP_RETR 00011111
RF_CH 01100100
RF_SETUP 00001110
STATUS 00001110
OBSERVE_TX 00001111
RX_PW_P0 00000000
RX_PW_P1 00000000
FIFO_STATUS 00010001
DYNPD 00000011
FEATURE 00000111
TX_ADDR 1,2,3,4,0
RX_ADDR_P0 1,2,3,4,0
RX_ADDR_P1 1,2,3,4,0

dparson55, 2022-06-02:
RX output looks good, this confirms the code is compiling correctly and uploading successfully to your microcontroller, and the microcontroller is successfully communicating with the nRF24L01+.

For TX I forgot to include the `Serial.begin(115200);`, just go ahead and add that similar to how it is in the RX code.  You'll want to see the same output as RX except for 1 register, we expect RX_ADDR_P1 to be 1,2,3,4,1.  This is the register that NRFLite uses for the radio id where the last digit will be the radio id.

kikesandi, 2022-06-02:
You're right. I added the Serial.begin() and got these results:
EN_AA 00111111
EN_RXADDR 00000011
SETUP_AW 00000011
SETUP_RETR 00011111
RF_CH 01100100
RF_SETUP 00001110
STATUS 00001110
OBSERVE_TX 00001111
RX_PW_P0 00000000
RX_PW_P1 00000000
FIFO_STATUS 00010001
DYNPD 00000011
FEATURE 00000111
TX_ADDR 1,2,3,4,0
RX_ADDR_P0 1,2,3,4,0
RX_ADDR_P1 1,2,3,4,1

Sending 1
Sending 2
Sending 3
Sending 4
Sending 5
Sending 6
Sending 7
Sending 8
Sending 9
Sending 10

Today in the afternoon I will reinstall the example sketches that are included in the library (simple rx and simple tx), and I will use printDetails(), for a better report.
Tkanks for your time.

kikesandi, 2022-06-02:
Update: 

Just uploades the Basic_TX.ino on a Arduino nano iot
And uploaded the Basic_RX.ino on a Arduino nano (regular).

On the Serial monitor or the TX, i got Success message 

on the serial monitor of the RX, I got nothing.

¿Am I putting the printDetails in the wrong place?

This is the code:

`/*

Demonstrates simple RX and TX operation.
Any of the Basic_TX examples can be used as a transmitter.
Please read through 'NRFLite.h' for a description of all the methods available in the library.

Radio    Arduino
CE    -> 9
CSN   -> 10 (Hardware SPI SS)
MOSI  -> 11 (Hardware SPI MOSI)
MISO  -> 12 (Hardware SPI MISO)
SCK   -> 13 (Hardware SPI SCK)
IRQ   -> No connection
VCC   -> No more than 3.6 volts
GND   -> GND

*/

#include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
    Serial.begin(115200);

    // By default, 'init' configures the radio to use a 2MBPS bitrate on channel 100 (channels 0-125 are valid).
    // Both the RX and TX radios must have the same bitrate and channel to communicate with each other.
    // You can run the 'ChannelScanner' example to help select the best channel for your environment.
    // You can assign a different bitrate and channel as shown below.
    //   _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE2MBPS, 100) // THE DEFAULT
    //   _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE1MBPS, 75)
    //   _radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS, 0)
    
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }
     _radio.printDetails();
     Serial.println();
}

void loop()
{
    while (_radio.hasData())
    {
        _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.

        String msg = "Radio ";
        msg += _radioData.FromRadioId;
        msg += ", ";
        msg += _radioData.OnTimeMillis;
        msg += " ms, ";
        msg += _radioData.FailedTxCount;
        msg += " Failed TX";

        Serial.println(msg);
    }
}`

dparson55, 2022-06-03:
In order for `printDetails` to work, the creation of the `radio` object needs to include a `Serial` object.

```
NRFLite _radio(Serial); // Note we pass in a Serial object so printDetails will work
```

kikesandi, 2022-06-03:
Hello, I changed the module for another one that has a 10 micro farad capacitor. I installed the code as you put it in the previous post and the data I get from "_radio.printDetails()" corresponds to the setup() and not to the loop(). To test this I commented out the last part and no information appeared on the serial monitor.

``` #include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio(Serial); // Note we pass in a Serial object so printDetails will work.
RadioPacket _radioData;
uint32_t _lastPrintDetailsMillis = 0;

void setup()
{
    Serial.begin(115200);
    
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }

    // This is the part I commented out
    //_radio.printDetails();
    //Serial.println();

}

void loop()
{
    while (_radio.hasData())
    {
        _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.

        String msg = "Radio ";
        msg += _radioData.FromRadioId;
        msg += ", ";
        msg += _radioData.OnTimeMillis;
        msg += " ms, ";
        msg += _radioData.FailedTxCount;
        msg += " Failed TX";

        Serial.println("msg");
    }

    // Show nRF24 registers once every 4 seconds.

    // This part show nothing on the Serial Monitor

    if (millis() - _lastPrintDetailsMillis > 3999)
    {
      _lastPrintDetailsMillis = millis();
      _radio.printDetails();
      Serial.println();
    }
}```

dparson55, 2022-06-03:
Almost perfect on your code formatting...last fix to do is put the 3 backtick marks on a new line, not on the same line as your code.  That will format the code perfectly and not show the additional backticks in the code block.

The behavior you're describing sounds like either the microcontroller is locking up in the loop or the serial output is failing in the loop.  What is happening on the TX side, is it reporting successes or failures?  That will help indicate if the RX side microcontroller is locking up.

Try adding some more debug information to help with the troubleshooting, for example:

```
#include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 0;       // Our radio's id.  The transmitter will send to this id.
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio(Serial); // Note we pass in a Serial object so printDetails will work.
RadioPacket _radioData;
uint32_t _lastPrintDetailsMillis = 0;

void setup()
{
    Serial.begin(115200);
    
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        Serial.println("Cannot communicate with radio");
        while (1); // Wait here forever.
    }

    _radio.printDetails();
    Serial.println();
}

void loop()
{
    while (_radio.hasData())
    {
        Serial.println("Before readData");
        _radio.readData(&_radioData); // Note how '&' must be placed in front of the variable name.

        String msg = "Radio ";
        msg += _radioData.FromRadioId;
        msg += ", ";
        msg += _radioData.OnTimeMillis;
        msg += " ms, ";
        msg += _radioData.FailedTxCount;
        msg += " Failed TX";

        Serial.println(msg);
    }

    // Show nRF24 registers once every 4 seconds.
    if (millis() - _lastPrintDetailsMillis > 3999)
    {
        _lastPrintDetailsMillis = millis();

        Serial.print("OnTime ");
        Serial.println(_lastPrintDetailsMillis);

        _radio.printDetails();
        Serial.println();
    }
}
```

dparson55, 2022-06-16:
@kikesandi any updates?  It's been a couple of weeks with no response so I'd like to close this issue if your troubleshooting has stopped.

dparson55, 2022-06-21:
Closing due to lack of response.

closed 2022-06-21 by dparson55