dparson55/NRFLite #80: How to set the Output power?

vkvark, 2022-06-29:
Hi,

I'm aware that the library will always set the output power to the maximum (0 dBm) level. This is not always convenient. For example, in my project the module will be located on a headband, so I'd like to set the output power to -12 dBm to avoid negative impact on health caused by the 2.4G radiation. Can you please add an output power setting to the library?

dparson55, 2022-06-29:
Please see #12 for why the output power isn't exposed and how the library controls it.  If you'd still like to use this library instead of a more full featured library then you can just edit your local copy and change the RF_SETUP register.

https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf
[image]

vkvark, 2022-06-30:
Well, I guess one additional setting wouldn't make the library too complex. Anyway, thanks for the clarification.

closed 2022-06-30 by dparson55

richardHnewmark, 2023-03-31:
Hi, I'm using two 1.2V AA batteries to power a 3.3V Arduino Pro / NRF24l01 (with step-up converter  to get 3.3V) and 1 in 6 transmissions fail. Powering the setup directly from the computer I get 100% successful transmissions. So I'm guessing that the default max power from the NRF24l01 is exceeding the current output capabilities of the step-up converter.
I'm really keen on powering the NRF24l01/Arduino on just two rechargeable AA batteries, and thought I could try reducing the transmission  power of the NRF24l01 to see if that makes a difference.... so could I add my voice to a function that does this!
At the moment I'm too much of a novice to "edit your local copy and change the RF_SETUP register."
But thanks for a great library... I too am an advocate of simplicity..!

vkvark, 2023-03-31:
> Hi, I'm using two 1.2V AA batteries to power a 3.3V Arduino Pro / NRF24l01 (with step-up converter to get 3.3V) and 1 in 6 transmissions fail. Powering the setup directly from the computer I get [...]

Which step-up converter are you using? Did you try to add an additional capacitor to the converter's output?

richardHnewmark, 2023-03-31:
https://www.ebay.co.uk/itm/253164825504   is the step-up converter I'm using.  I have a 10uF capacitor across the grd/Vcc of the nrf24l01, but that didn't make any improvement.

vkvark, 2023-03-31:
Try a 33 to 47 uF ceramic capacitor.