dparson55/NRFLite #82: hasData() should either return actual dataLength or an error.

MikeysLab, 2023-12-16:
If a packet is too large, hasData() should return the true length, or an error. This will allow the recieving program to better handle the issue.

dparson55, 2023-12-18:
The nRF24L01 chip doesn't support receiving of packets larger than 32 bytes so the designers of the chip detailed in the datasheet that if the chip reports a packet length > 32 bytes (in the R_RX_PL_WID the register), its RX FIFO buffer should be flushed and the packet should be ignored. So I don't think it's possible to return a packet's true length.

I'm not sure how to throw an error with the Arduino framework, haven't really seen that. I could add a callback for when this flushing occurs, potentially something like `OnInvalidPacketReceived` but I'm not sure if it's really all that helpful and might just lead to more confusion and complication.

Instead of changing the library I added some additional comments to the library's documentation in NRFLite.h, hopefully that will help. I've already noted the limitation of the maximum packet size in the examples.

dparson55, 2023-12-18:
The nRF24L01 chip doesn't support receiving of packets larger than 32 bytes so the designers of the chip detailed in the datasheet that if the chip reports a packet length > 32 bytes (in the R_RX_PL_WID the register), its RX FIFO buffer should be flushed and the packet should be ignored. So I don't think it's possible to return a packet's true length.

I'm not sure how to throw an error with the Arduino framework, haven't really seen that. I could add a callback for when this flushing occurs, potentially something like `OnInvalidPacketReceived` but I'm not sure if it's really all that helpful and might just lead to more confusion and complication.

Instead of changing the library I added some additional comments to the library's documentation in NRFLite.h and the examples, and hopefully that will help.

dparson55, 2023-12-28:
Been a week without a response so hoping the changes I made were satisfactory.

closed 2023-12-28 by dparson55