dparson55/NRFLite #83: mixing addAckData() and send() results in a lot of received packets

mvoss96, 2024-02-05:
Hi, 
for one of my current projects i want to mix mix hardware  based two-way communication and acknolement payoads, however i found a weird behaviour that i currently don't understand.
My basic idea is: i have two Devices A and B. Both can send messages to each other. But B will also include an acknowledgement payload. 
I have no problem just having the radios talk to each other without ack payloads.
My Problem is that when i try to integrate ackpayloads i recieve the payloads 20-30 times after another. Also the payloads get detected as normal messages and not as ack payloads.

i have created a simple code example that shows it:

Code A

```
#include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 1;
const static uint8_t DESTINATION_RADIO_ID = 5;
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket {
  uint8_t test = 0;
};

uint8_t counter = 0;
NRFLite _radio;
uint8_t buff[32];

void setup() {
  Serial.begin(115200);

  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS, 111)) {
    Serial.println("Cannot communicate with radio");
    while (1)
      ;  // Wait here forever.
  }
  Serial.println("Start A");
}

void printbuffer(uint8_t* buff, size_t l) {
  for (int i = 0; i < l; i++) {
    Serial.print(buff[i]);
    Serial.print(" ");
  }
  Serial.println();
}

void send() {
  RadioPacket pck;
  pck.test = counter++;
  bool res = _radio.send(DESTINATION_RADIO_ID, &pck, sizeof(pck));
  Serial.print("Send: ");
  Serial.println(res ? "Success" : "Failed");

  while (_radio.hasAckData())  // Look to see if the receiver provided the ACK data.
  {
    uint8_t ackLength = _radio.hasAckData();
    _radio.readData(&buff);
    Serial.print("ACK DATA: ");
    printbuffer(buff, ackLength);
  }
}

void receive() {
  uint8_t packetSize = _radio.hasData();
  if (packetSize > 0) {
    _radio.readData(&buff);
    Serial.print("RECV: ");
    printbuffer(buff, packetSize);
  }
}

void loop() {
  static long timer = millis();

  // Send
  if (millis() - timer > 5000) {
    timer = millis();
    send();
  } else {
    receive();
  }
}
```

Code B
```
#include "SPI.h"
#include "NRFLite.h"

const static uint8_t RADIO_ID = 5;
const static uint8_t DESTINATION_RADIO_ID = 1;
const static uint8_t PIN_RADIO_CE = 9;
const static uint8_t PIN_RADIO_CSN = 10;

struct RadioPacket {
  uint8_t test = 1;
};

struct AckPacket {
  uint8_t a = 11;
  uint8_t b = 12;
};

uint8_t counter = 0;
NRFLite _radio;
uint8_t buff[32];

void setup() {
  Serial.begin(115200);

  if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN, NRFLite::BITRATE250KBPS, 111)) {
    Serial.println("Cannot communicate with radio");
    while (1)
      ;  // Wait here forever.
  }
  Serial.println("Start B");
}

void printbuffer(uint8_t* buff, size_t l) {
  for (int i = 0; i < l; i++) {
    Serial.print(buff[i]);
    Serial.print(" ");
  }
  Serial.println();
}

void send() {
  RadioPacket pck;
  pck.test = counter++;
  bool res = _radio.send(DESTINATION_RADIO_ID, &pck, sizeof(pck));
  Serial.print("Send: ");
  Serial.println(res ? "Success" : "Failed");

  while (_radio.hasAckData())  // Look to see if the receiver provided the ACK data.
  {
    uint8_t ackLength = _radio.hasAckData();
    _radio.readData(&buff);
    Serial.print("ACK DATA: ");
    printbuffer(buff, ackLength);
  }
}

void receive() {
  uint8_t packetSize = _radio.hasData();
  if (packetSize > 0) {
    _radio.readData(&buff);
    Serial.print("RECV: ");
    printbuffer(buff, packetSize);
    AckPacket ack;
    _radio.addAckData((void*)&ack, 2, true);
  }
}

void loop() {
  static long timer = millis();

  // Send
  if (millis() - timer > 5000) {
    timer = millis();
    send();
  } else {
    receive();
  }
}
```

this results in output like:
Send: Success
RECV: 4 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12 
RECV: 11 12

Am i missing some behavior of the nrf chip?
thanks for your great library.

dparson55, 2024-02-06:
Thanks for the code, I was able to recreate the problem and the issue is trying to mix and match hardware and software-based two-way communcation. You can use either type, but not both at the same time. I never tried mixing them like you are doing and you can see it doesn't work, as you discovered hasData and hasAckData are not reported correctly. This leads to the bug where `RECV: 11 12` is repeated many times...it is because the `buff` variable is 32 bytes in length and the Serial output is being repeated 32 times (the radio didn't actually send the packet 32 times).

When you're using ACK packets, that's hardware-based communcation, so carefully read [lines 7-11 from the hardware-based example](https://github.com/dparson55/NRFLite/blob/master/examples/TwoWayCom_HardwareBased_RX/TwoWayCom_HardwareBased_RX.ino#L7) to understand the limitations, particular this line: "the receiver cannot initiate two-way communication, it must be started by the transmitter".

So if you need both device A and B to initiate communication, you'll need to avoid using ACK data packets. This will be terribly slow compared to using ACK data packets but will hopefully work for your situation.

mvoss96, 2024-02-06:
Thanks for your answer. 
My Goal was to have a device (lamp) that could be polled for its status but would also report special events (like being switched by an external switch). 

I ported my example to the rf24 library and there the mixing of hardware and software based two way communication works when i load the ack payload after a send.

Maybe something similar could also be done within nrflite?

Example code:

```
#include <RF24.h>

RF24 radio(9, 10);  // CE, CSN pins
const byte address[6] = "00001";
char dataToSend[10] = "Hello";
char receivedData[10] = "";  // Buffer to hold incoming packets
uint8_t counter = 0;

void addAckPayload() {
  radio.writeAckPayload(1, &counter, sizeof(counter));  // Pre-load the acknowledgment data
}

void send() {
  radio.stopListening();  // Stop listening to start sending

  bool result = radio.write(&dataToSend, sizeof(dataToSend));
  Serial.print("Sending data: ");
  Serial.print(dataToSend);
  if (result) {
    Serial.print(" -> Acknowledgment received");
    if (radio.isAckPayloadAvailable()) {
      uint8_t ackPayload[32];
      uint8_t payloadSize = radio.getDynamicPayloadSize();
      radio.read(&ackPayload, payloadSize);
      Serial.print(" Acknowledgment payload: ");
      for (int i = 0; i < payloadSize; i++) {
        Serial.print(ackPayload[i]);
        Serial.print(" ");
      }
    }
  } else {
    Serial.print(" -> Acknowledgment not received");
  }
  Serial.println();
  radio.startListening();  // Start listening after sending
  addAckPayload();
}

void receive() {
  if (radio.available()) {
    memset(receivedData, 0, sizeof(receivedData));  // Clear previous data
    radio.read(&receivedData, sizeof(receivedData));
    Serial.print("Data received: ");
    Serial.println(receivedData);
    counter++;
  }
}

void setup() {
  Serial.begin(115200);
  if (!radio.begin()) {
    Serial.println(F("Radio hardware is not responding!!"));
    while (1) {}  // Hold in an infinite loop if radio initialization fails
  }

  radio.openWritingPipe(address);
  radio.openReadingPipe(1, address);
  radio.setPALevel(RF24_PA_LOW);
  radio.enableDynamicPayloads();
  radio.enableAckPayload();  // Enable acknowledgment payloads
  addAckPayload();
  counter++;
  radio.startListening();
  Serial.println("Transmitter A is ready");
}

void loop() {
  static unsigned long timer = millis();

  // Send data every 5 seconds
  if (millis() - timer > 5000) {
    timer = millis();
    send();
  }

  receive();  // Continuously check for received data
}
```

dparson55, 2024-02-07:
That example uses ACK data packets so it cannot be done like that with NRFLite, you have to use normal data packets and manually switch the radios into TX and RX mode as needed. So if you want to use NRFLite, you'll need to use the [software-based two way communcation example](https://github.com/dparson55/NRFLite/blob/master/examples/TwoWayCom_SoftwareBased_TX/TwoWayCom_SoftwareBased_TX.ino) as a starting point. If you have a specific question I'll be happy to help.

Just for fun I went ahead and implemented a little of what you described: I made a `lamp` and a `lamp montior`. The lamp has a switch connected to it so it can be turned on and off. If the switch is flipped, it will send the new state to the lamp monitor. Independently the lamp monitor will ask the lamp for its state once every 30 seconds.

[image]

In the serial output of the lamp monitor on the left, at 16:12:10 it was started, then 30 seconds later it asked the lamp for its state and the lamp replied back that it was on, then 7 seconds later I switched the lamp off and it sent the new off state to the lamp monitor at 16:12:47.
[image]

These are the two types of data packets used in this example. The lamp monitor sends the `GetLampStatePacket` packet to the lamp every 30 seconds. This packet includes its radio id so the lamp knows who it should send a `LampStatePacket` packet to.
```
// An Arduino can send us this packet to ask for the lamp's state.
// If we receive one of these packets, we'll send a LampStatePacket back to it.
struct GetLampStatePacket
{
    uint8_t FromRadioId;
};

// This is the packet that we will send to report the lamp's state.
struct LampStatePacket
{
    uint8_t FromRadioId;
    uint8_t LampState;
};
```

mvoss96, 2024-02-09:
Thanks again for your response. I will look into it.

closed 2024-02-09 by mvoss96