dparson55/NRFLite #85: (NRF24L01+) ("NRFLite.h") (SplitDataAcrossMultiplePackets_RX) read register 0x09 directly(RPD)

gordon4liu, 2024-04-11:
Hi,
 Bracuse "NRFLite.h: don't supply the RPD (Rx power detect), I want to read the register 0x09 directly.
    Serial.println(_radio.readRegister(0x09));  //  I only insert the code of but it's not work. 
Could you give ne any suggestion to resolve the issue??

===my code===
void loop()
{
    while (_radio.hasData())
    {
      // Serial.println(_radio.readRegister(CD));
      Serial.println(_radio.readRegister(0x09));

        RadioPacket packet;
        _radio.readData(&packet);

        if (packet.PacketNumber == 0)
        {
.
.
========

===Target==   RPD function  ===
bool goodSignal = radio.testRPD();
if(radio.available()){
Serial.println(goodSignal ? "Strong signal > 64dBm : "Weak signal < 64dBm");
radio.read(0,0); 
=======

dparson55, 2024-04-11:
You can use [scanChannel](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L34) to measure the strength of signals on a channel. The [ChannelScanner](https://github.com/dparson55/NRFLite/blob/master/examples/ChannelScanner/ChannelScanner.ino) example shows its usage.

I made `readRegister` and many other functions private to keep the library as simple to use as possible. If you ever want to change this though, you can fork this repo and customize the library as you see fit.

dparson55, 2024-04-14:
Closing due to lack of response.

closed 2024-04-14 by dparson55