dparson55/NRFLite #87: Cannot compile a sketch after ESP32 core update

vkvark, 2024-07-24:
Hi there.

I cannot compile any sketch using your lib after the recent ESP32 core update without errors. 
There are multiple compilation errors in the log like follows:
`c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:399:35: warning: 'B01011111' is deprecated: use 0b01011111 instead [-Wdeprecated-declarations]
  399 |         writeRegister(SETUP_RETR, B01011111); // 0101 = 1500 uS between retries, 1111 = 15 retries

c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src/nRF24Lite.h:114: warning: "NOP" redefined
  114 | #define NOP           0xFF
      | 
In file included from C:\Users\kvark\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/Arduino.h:36,
                 from C:\Users\kvark\AppData\Local\Temp\arduino\sketches\3D79A5B17F61DA8D3F1156B320C9A9CB\sketch\Basic_RX.ino.cpp:1:
C:\Users\kvark\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal.h:70: note: this is the location of the previous definition
   70 | #define NOP()         asm volatile("nop")
      | 
In file included from c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.h:5,
                 from c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:1:
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\nRF24Lite.h:114: warning: "NOP" redefined
  114 | #define NOP           0xFF
      | 
In file included from C:\Users\kvark\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/Arduino.h:36,
                 from c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.h:4:
C:\Users\kvark\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.2\cores\esp32/esp32-hal.h:70: note: this is the location of the previous definition
   70 | #define NOP()         asm volatile("nop")
      | 
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp: In member function 'uint8_t NRFLite::twoPinTransfer(uint8_t)':
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:622:20: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  622 |         *_momi_DDR |= _momi_MASK;                         // Change MOMI to be an OUTPUT pin.
      |         ~~~~~~~~~~~^~~~~~~~~~~~~
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:624:40: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  624 |         if (data & 0x80) { *_momi_PORT |=  _momi_MASK; }  // Set MOMI HIGH if bit position 7 of the byte we are sending is 1.
      |                            ~~~~~~~~~~~~^~~~~~~~~~~~~~
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:626:20: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  626 |         *_sck_PORT |= _sck_MASK;  // Set SCK HIGH to transfer the bit to the radio.  CSN will remain LOW while the capacitor begins charging.
      |         ~~~~~~~~~~~^~~~~~~~~~~~
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:627:20: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  627 |         *_sck_PORT &= ~_sck_MASK; // Set SCK LOW.  CSN will have remained LOW due to the capacitor.
      |         ~~~~~~~~~~~^~~~~~~~~~~~~
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:629:21: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  629 |         *_momi_PORT &= ~_momi_MASK; // Set MOMI LOW.
      |         ~~~~~~~~~~~~^~~~~~~~~~~~~~
c:\Users\kvark\OneDrive\Arduino\libraries\NRFLite\src\NRFLite.cpp:630:20: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  630 |         *_momi_DDR &= ~_momi_MASK;  // Change MOMI back to an INPUT.  Since we previously ensured it was LOW, its PULLUP resistor will never
`

dparson55, 2024-07-24:
I installed the latest Arduino IDE (2.3.2) and tried the latest ESP32 core (3.0.3) and they are compiling the Basic_RX example okay. Can you provide some more information on how to recreate the problem?

[image]

dparson55, 2024-07-24:
Updates complete and are included in NRFLite 3.0.5

vkvark, 2024-07-25:
Many thanks for the prompt follow-up!

closed 2024-07-25 by dparson55