dparson55/NRFLite #91: Compilation error when using ATtiny24/44/84 board

Jimbolaya64, 2025-02-16:
1. Compilation error:

```
In file included from /home/jim/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/io.h:99:0,
                 from /home/jim/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/pgmspace.h:90,
                 from /home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/Arduino.h:28,
                 from /home/jim/Sketches/libraries/NRFLite/src/NRFLite.h:4,
                 from /home/jim/Sketches/libraries/NRFLite/src/NRFLite.cpp:1:
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h: In member function 'void SPISettings::init_AlwaysInline(uint32_t, uint8_t, uint8_t)':
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:146:16: error: 'SPE' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                ^
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:146:16: note: suggested alternative: 'SPL'
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:146:27: error: 'MSTR' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                           ^
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:146:27: note: suggested alternative: 'PSTR'
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:146:65: error: 'DORD' was not declared in this scope
     spcr = _BV(SPE) | _BV(MSTR) | ((bitOrder == LSBFIRST) ? _BV(DORD) : 0) |
                                                                 ^
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:146:65: note: suggested alternative: 'DDRB'
In file included from /home/jim/Sketches/libraries/NRFLite/src/NRFLite.cpp:15:0:
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h: In static member function 'static void SPIClass::beginTransaction(SPISettings)':
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:203:5: error: 'SPCR' was not declared in this scope
     SPCR = settings.spcr;
     ^~~~
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:203:5: note: suggested alternative: 'EECR'
     SPCR = settings.spcr;
     ^~~~
     EECR
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:204:5: error: 'SPSR' was not declared in this scope
     SPSR = settings.spsr;
     ^~~~
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:204:5: note: suggested alternative: 'ACSR'
     SPSR = settings.spsr;
     ^~~~
     ACSR
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h: In static member function 'static uint8_t SPIClass::transfer(uint8_t)':
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:209:5: error: 'SPDR' was not declared in this scope
     SPDR = data;
     ^~~~
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:209:5: note: suggested alternative: 'EEDR'
     SPDR = data;
     ^~~~
     EEDR
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:217:14: error: 'SPSR' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait
              ^~~~
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:217:14: note: suggested alternative: 'ACSR'
     while (!(SPSR & _BV(SPIF))) ; // wait
              ^~~~
              ACSR
In file included from /home/jim/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/io.h:99:0,
                 from /home/jim/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/avr/include/avr/pgmspace.h:90,
                 from /home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/Arduino.h:28,
                 from /home/jim/Sketches/libraries/NRFLite/src/NRFLite.h:4,
                 from /home/jim/Sketches/libraries/NRFLite/src/NRFLite.cpp:1:
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:217:25: error: 'SPIF' was not declared in this scope
     while (!(SPSR & _BV(SPIF))) ; // wait
                         ^
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:217:25: note: suggested alternative: 'WDIF'
In file included from /home/jim/Sketches/libraries/NRFLite/src/NRFLite.cpp:15:0:
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h: In static member function 'static uint16_t SPIClass::transfer16(uint16_t)':
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:223:11: error: 'SPCR' was not declared in this scope
     if (!(SPCR & _BV(DORD))) {
           ^~~~
/home/jim/.arduino15/packages/arduino/hardware/avr/1.8.6/libraries/SPI/src/SPI.h:223:11: note: suggested alternative: 'EECR'
     if (!(SPCR & _BV(DORD))) {
           ^~~~
           EECR
[... 236 lines cut ...]
exit status 1

Compilation error: exit status 1
```

2. Example code:

```
#include "NRFLite.h"

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

```

3. I'm using version 1.0.2 of the David Mellis library, the "ATtiny24/44/84" board and version 2.3.4 of the Arduion IDE. I have more complex code I'm trying to compile. But this simple bit of code reproduces the error.

4. Switching the board to "LOLIN(WEMOS) D1 mini (clone)" or "Arduino Nano" results in it compiling with no error.

5. Any suggestions? Is it me? Thank you.

dparson55, 2025-02-16:
I see what you mean, unfortunately the `David Mellis Attiny library` doesn't work with the Arduino 2.x IDE and it looks like there are no plans to get it supported. So I downloaded the latest version of the older IDE, Arduino 1.8.19, and it is still working ok. I'll make a note of this requirement in the NRFLite ATtiny examples and the readme.

[image]

Jimbolaya64, 2025-02-17:
That’s unfortunate. But, thanks for the quick reply and updating the relevant sketches.

dparson55, 2025-02-18:
Sure thing, gl with the project.

closed 2025-02-18 by dparson55