dparson55/NRFLite #92: Compile Error with DigiSpark using Micronucleus

AZGLi, 2025-02-24:
1. Error: In file included from D:\Documents\Arduino\libraries\NRFLite\examples\Basic_TX_ATtiny85\Basic_TX_ATtiny85.ino:20:0:
D:\Documents\Arduino\libraries\NRFLite\src/NRFLite.h:78:5: error: 'constexpr' does not name a type
     constexpr static uint8_t ADDRESS_PREFIX[4] = { 1, 2, 3, 4 }; // 1st 4 bytes of addresses, 5th byte will be RadioId.
     ^
D:\Documents\Arduino\libraries\NRFLite\src/NRFLite.h:78:5: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
exit status 1
Error compiling for board Digispark (16mhz - No USB)

2. Code: 
3. /*

Demonstrates simple TX operation with an ATtiny85.  Note in this example the same pin is used for CE and CSN.
This slows the communication between the microcontroller and radio, but it frees up one pin.

Any of the Basic_RX examples can be used as a receiver.

Radio    ATtiny85
CE    -> Physical Pin 2, Arduino 3         
CSN   -> Physical Pin 2, Arduino 3         
MOSI  -> Physical Pin 6, Arduino 1 (Hardware USI_DO)  
MISO  -> Physical Pin 5, Arduino 0 (Hardware USI_DI)  
SCK   -> Physical Pin 7, Arduino 2 (Hardware USI_SCK)  
IRQ   -> No connection
VCC   -> No more than 3.6 volts
GND   -> GND

*/

#include "NRFLite.h"

const static uint8_t RADIO_ID = 4;
const static uint8_t DESTINATION_RADIO_ID = 0;
const static uint8_t PIN_RADIO_CE = 3;
const static uint8_t PIN_RADIO_CSN = 3;

struct RadioPacket // Any packet up to 32 bytes can be sent.
{
    uint8_t FromRadioId;
    uint32_t OnTimeMillis;
    uint32_t FailedTxCount;
};

NRFLite _radio;
RadioPacket _radioData;

void setup()
{
    if (!_radio.init(RADIO_ID, PIN_RADIO_CE, PIN_RADIO_CSN))
    {
        while (1); // Cannot communicate with radio.
    }

    _radioData.FromRadioId = RADIO_ID;
}

void loop()
{
    _radioData.OnTimeMillis = millis();

    if (!_radio.send(DESTINATION_RADIO_ID, &_radioData, sizeof(_radioData)))
    {
        _radioData.FailedTxCount++;
    }

    delay(1000);
}

3. I am using Arduino IDE version 1.8.19 with the Digistump library 1.6.7. This board library uses Micronucleus to compile. I have updated the complier to version 2.6 but the behavior also exists with 2.0a4. I suspect it's a Micronucleus issue. 
4. I am trying to use the NRFLITE library with the Digispark which uses the ATTiny85. I was hoping that I could use this board with the 4-wire connection to get two more available pins. I can change boards but I wanted to see if I could get this working. Anything you can provide to assist will be welcome.

AZGLi, 2025-02-24:
Thanks for the reply. I'll try that path. I was using the files here: 

https://github.com/digistump/arduino-boards-index/blob/master/package_digistump_index.json

AZGLi, 2025-02-24:
Switching to the ATTinyCore resolved the issue. Thank you for the assistance. I wasn't able to get that board set installed at first but found a tutorial I hadn't seen.

dparson55, 2025-02-25:
Good to hear and good luck getting the radio to work. Hopefully it will even work with NRFLite's 2 pin control method.

dparson55, 2025-02-25:
1. You get 256 addresses ... each number in the address is a byte which can be 0-255. [This comment](https://github.com/dparson55/NRFLite/issues/54#issuecomment-747687341) goes into more detail.
2. Yes (defaults [here](https://github.com/dparson55/NRFLite/blob/master/src/NRFLite.h#L36))
3. Yes (defaults same line as above)