dparson55/NRFLite #93: RX buffers need to be flushed if R_RX_PL_WID returns 0

TMRh20, 2025-05-04:
1. Describe the bug:
The RX buffers need to be flushed if R_RX_PL_WID returns 0
Else radio reads will have no effect, and FIFO status will continue to indicate available packets.

2. Provide a short program that demonstrates the bug. DO NOT post your entire program, instead create a smaller version with no extraneous code: it should only include the logic necessary to reproduce the issue.

Affects pretty much all code using Dynamic Payloads. The issue can be extremely difficult to reproduce, making it very difficult to debug. Essentially high traffic conditions are more conducive to it happening. It appears to affect all RF24 radios & many libraries. Use any RX code example that utilizes Dynamic Payload Lengths.

We discovered in the RF24 library this register can return 0. After that, a radio read has no effect, the RX FIFOs need to be flushed.

Per the datasheet: 
> Note: Always check if the packet width reported is 32 bytes or shorter when using the
R_RX_PL_WID command. If its width is longer than 32 bytes then the packet contains errors
and must be discarded. Discard the packet by using the Flush_RX command.

Well it turns out the same condition applies when it returns 0.

I noticed your code accesses the R_RX_PL_WID register here:
https://github.com/dparson55/NRFLite/blob/394e843ebbd4cb53bc12f605aaa8ec19f445aa68/src/NRFLite.cpp#L352
and
https://github.com/dparson55/NRFLite/blob/394e843ebbd4cb53bc12f605aaa8ec19f445aa68/src/NRFLite.cpp#L177

I also note that the line
https://github.com/dparson55/NRFLite/blob/394e843ebbd4cb53bc12f605aaa8ec19f445aa68/src/NRFLite.cpp#L460
clears the RX buffers, so transmitting with this library may resolve the receiving errors.

Just thought I would share the info from one lib maintainer to another.

Changes:
https://github.com/dparson55/NRFLite/blob/394e843ebbd4cb53bc12f605aaa8ec19f445aa68/src/NRFLite.cpp#L355
to
```cpp
    if (dataLength > 32 || !dataLength)
```

dparson55, 2025-05-04:
Thanks, this is very kind of you to share! I read through the discussions I could find from your libraries and incorporated the suggested change and released a new version. That's a really tricky situation.

Unfortunately I couldn't recreate the issue with my setup and can't purposefully load a 0 byte packet into the radio for transmission (all my loops need a length > 0), but if it somehow happens NRFLite will now handle the situation, so thanks!

TMRh20, 2025-05-04:
That was quick! 
Its about as common as the >32 bytes payload issue, doesn't pop up that often, but when it does its a pain!

closed 2025-05-04 by TMRh20

TMRh20, 2025-05-09:
Just FYI I've opened an issue with Nordic:
https://devzone.nordicsemi.com/f/nordic-q-a/121237/nrf24l01-radio-r_rx_pl_wid-returns-0

dparson55, 2025-05-09:
Thanks, I'll keep an eye on this ... will be interesting to see if they have some insights.